slic3r/Slic3r #2911: Suggestion: freely choosable seam position.

Sungod3000, 2015-06-14:
Some time ago I decided that I liked it more to have a visible seam somewhere in the model instead of random layer changing marks all over the print. Of course the better the retraction the less visible the seam but I guess there will always be a little mark upon layer change.

Therefore it would be great to be able to move the seam around the model, so you could place it on the backside or the inside of holes/arches to hide it better.

lonelymyp, 2015-06-15:
join.
sometimes you want to hide the seam, for example if you are printing a bust of a man.
but there is a seam or on the nose or ear.
It looks like a scar.

DrLex0, 2016-07-31:
I also vote for this. I printed the Marvin model with ‘aligned,’ and it looks nice up until the point where the shape consists of pure circles. Slic3r made the unfortunate choice of placing the seam right on his forehead. Maybe there could also be a hybrid mode which uses the ‘nearest’ algorithm if there is a good corner to hide the seam, and a user-defined default otherwise.
[image]

lahure-c, 2016-11-23:
This would also allow to put the seam where it's easier to sand it off.

Cura has a setting for this:  Z seam alignment = Back
in that case the seam is always at the highest Y coordinate. 
The user places the model so that its back it is in the Y direction and
the seam is in the back.