slic3r/Slic3r #2912: Print center

edegelder, 2015-06-14:
Hi, 

I'm new here, so I'm not sure whether this is the right place for asking my question or not. 

I just upgraded my slic3r from 1.1.7 to 1.2.7 and now I can't find the option to change the print center anymore. Now the print center is always at the center of the print bed, while I would like to be able to vary this position. Any idea how I could change the print center?

Excuse me if this is a very silly question as I might be looking in the wrong section. Any help is appreciated. Thank you in advance!

alexrj, 2015-06-14:
Since bed shape can be customized in Slic3r 1.2.x there's no need to specify the center manually anymore. If you want to turn off the autocentering you can just disable it from the application preferences dialog.

closed 2015-06-14 by alexrj

a4jp-com, 2015-06-14:
You can change the origin in the bed settings area if you need to.

Printer Settings tab > General > Size and coordinates > Bed shape > Set

CoDanny, 2015-06-22:
When calling it from the command line using a ini file, it seems that print_center is ignored and the default (100,100) is used.

How can I specify the print center for square and circular beds in 1.2.x when using slic3r from the command line using an ini file?

For reference, I used to set print_center = 0,0 for circulars and print_center = x/2,y/2 for rectangular in 1.1.7

hzmangel, 2015-07-15:
I have met the same print center problem in command line mode. I have a rectangular bed with size 500x500. If generating gcode via GUI, the model will be placed in center of bed, which is (250, 250). However, the default (100, 100) is used if using command line.

I know I can pass `--print-center` argument in command line to set the center, but I can't find the way to setup it in configuration file. Can anyone help me on this?

Thanks

a4jp-com, 2015-07-15:
[image]

Printer settings > Size and coordinates > Bed shape: Set

hzmangel, 2015-07-15:
@a4jp-com Thanks for the explanation. I know the setting of GUI, but the problem I am facing is from command line interface. The print center is set to 100,100 by default, and can't be changed via configuration file.

The bed size I am using is 500x500, in the GUI slice result, the print center is (250, 250), which is set to middle automatically. However, if using perl ./slice.pl, the print center is set to (100, 100). Then the same STL file and configuration file generates different gcode.

lordofhyphens, 2015-07-15:
@hzmangel It's definitely a "hard" problem; although there is a
script-based workaround: offset all XY coordinates by 150,150. I agree this
needs to be addressed, however, working out the center of an oddly-shaped
bed (which can happen) is not trivial.

Being able to set the center coordinate as a variable would probably work.

istergiou, 2016-11-29:
command option --print-center=90,150  works in MacOS binaries of Slic3r 1.2.9
Option print_center = 90,150 in slic3r.ini file is not working. In a bed_shape = 0x0,185x0,185x290,0x290 it is using print center of 100,100.

jed-frey, 2017-10-05:
Just commenting that this is on the first page of google when searching for this problem.

Exporting from the GUI centers. Exporting from cli does not. Just added --print-center=150,150 to my make file for now.

lordofhyphens, 2017-10-05:
On the gui it is relative to the print bed center.

lordofhyphens, 2017-10-05:
@jed-frey from what I can tell with 1.3.0-dev if you don't specify a print-center it calculates the center of the bed shape from the points it reads in bed_shape. There is no `print_center` configuration option defined in libslic3r; it's a CLI configuration override only.