slic3r/Slic3r #2913: 1.2.8-dev 1st layer and 2nd layer around the wrong way. Also the top and bottom fills not calculated on my A4JP logo model.

a4jp-com, 2015-06-14:
My first layer is .3 so it should be bigger and second .21 so smaller but the layers are opposite for some reason. Can you flip these the right way around please?

http://a4jp.com/3d-files/bugs/2015-06-15th-12-12-am-for-github-config.ini
http://a4jp.com/3d-files/bugs/cube20bmm.stl

[image]
[image]
[image]

------

I have 3 layers set at the top and bottom but they aren't working. The 1st bottom layer is there but other 2 aren't and all the top layers are missing a fill pattern.

http://a4jp.com/3d-files/bugs/2015-06-15th-12-12-am-for-github-config.ini
http://a4jp.com/3d-files/bugs/a4jp.stl

[image]

kefir-, 2015-06-14:
You should file separate issues for separate problems. 

Don't know about the second issue, but what's wrong with the first and second layers of the box? What do you mean by "bigger" and "smaller"? From your pictures it looks like the layer heights are correct. It doesn't look like you've specified any extrusion widths, so a 0.3mm layer height will result in thinner extrusions than a 0.21mm layer. It looks right to me.

a4jp-com, 2015-06-14:
0.3mm is bigger so the bottom layer should have less lines. 0.3mm is bigger than 0.2mm. I'm sure this is a mistake in the program.

a4jp-com, 2015-06-14:
Hi kefir, I just fixed this post and made it one layer issue. I thought putting both together would be useful but will make another post for objects not having top or bottom layers.

kefir-, 2015-06-14:
0.3mm is the layer height. You're extruding a cylinder of plastic from your printer's hotend, and if you squeeze that to thinner layers, the extrusion will be wider. That results in fewer lines in your second layer. Conversely, when you print a thicker layer it'll be squeezed less, and you'll have thinner extrusions, like in your first layer. 

If you want wider (bigger?) extrusions and fewer lines *and* a thicker first layer, you'll need to change your first layer extrusion width under advanced settings.

a4jp-com, 2015-06-14:
I understand that but my setting is default in the Advanced section. I'm not choosing to make it fatter and flatter as the instruction say. If I want it to do that I know I can set it there but it is at the default of 0.

a4jp-com, 2015-06-14:
[image]
[image]

I think something is wrong with the calculation.

a4jp-com, 2015-06-14:
[image]

I couldn't delete the 0.21 number from the field.

a4jp-com, 2015-06-14:
There are also 5 layers instead of 3 like I set.

a4jp-com, 2015-06-14:
I just changed the bottom layer count to 2 but there was a 4th and 5th layer.

a4jp-com, 2015-06-14:
[image]

I made a mistake pointing at the top 3 but the bottom 3 is what I was meant to be pointing at.

alexrj, 2015-06-15:
I don't see the problem. A thicker layer (in Z) will have thinner extrudates (in XY), so more lines. It's normal.

a4jp-com, 2015-06-15:
If 3 layers are set the extra layers are part of a bug, aren't they?

a4jp-com, 2015-06-15:
[image]

a4jp-com, 2015-06-17:
Everything looks okay now in 1.2.9-dev ^^

[image]

closed 2015-06-17 by a4jp-com