slic3r/Slic3r #2916: Not possible to choose the XY spacing between support and object

bitouns, 2015-06-15:
In slic3r the spacing between support and object have a default value. But with some prints you can need to change this default value to get a better result .It also depend of the material you are using : supports with PLA don't need the same spacing as supports with ABS.
If it's a default value may be it's easy to just add a little option in slic3r support interface to chose this value.
What do you think ? Is it possible ?

lonelymyp, 2015-06-15:
It's an old problem. 
In my case, support stick to the perimeter, especially in the corners, where excessive extrusion, due to bowden and acceleration.

bitouns, 2015-06-17:
Yes so if you could choose the distance between support and object you would not have this problem.

dropinsight, 2015-06-19:
Also make the z-distance to the support variable, not either soluble 0mm or detachable 0.2mm.

lonelymyp, 2015-06-19:
It variable, you just have not tried ;)

[image]

bitouns, 2015-06-19:
Yes it possible for Z but not for XY

alexrj, 2015-06-19:
I dislike starting from solutions without having discussed concrete test cases. I usually look at a test case, then I try to identify the problem, then I try to figure possible solutions. Let's do it this way, and let's not skip steps.

"XY spacing" means little to nothing actually.  Support material is NOT supposed to have horizontal contact. There should be NO contact in XY. We only have vertical contact (Z). The XY distance between object and any projected support material should be large enough so that there's NO contact, but large enough so that the upper support material layer can be supported. This should be algorithmically determined, and I really don't see why this algorithm would change for each print or why user should decide to use a different value.

lonelymyp, 2015-06-19:
large nozzles is a problem. Support sticks at all angles.
but even with small nozzles is not so good ...

moment ago specially printed the worst-case scenario:
[image]

I want to quickly print, so I have to use some acceleration and bowden
Therefore, in the corners there are thickening. they adhere to the support.
I almost always use a knife after printing if the item has been with support, If only I could increase the XY distance between object and support material +0.2 mm, everything would be fine :)

an alternative option is to change the logic of adding support, let support goes around the corners of large radius.
[image]

dropinsight, 2015-06-19:
Ups I feel a bit stupid now. Yeah you are right. I always thought it is only a drop down list, because of the arrow next to it. Thanks for the info.

dropinsight, 2015-06-20:
I haven't tried the volcano of e3d (http://e3d-online.com/Volcano/Volcano-Starter-Pack-3mm) yet, but  I could imagine that the XY-distance could be a bit tight for using a 0.8mm nozzle (or even bigger). It would be then really useful to change such a setting, also changing the XY-distance of the interface layers would be an interesting option. I like using a model testing different settings to see what works the best. I can understand this would involve a lot more work for you guys, but playing around with settings is part of the fun working with 3d printers. Also it would be cool to have different layer heights for dual extruders. So that I could for instance print with a 0.3mm nozzle at 0.1mm layer height and with my 0.8mm nozzle the support (soluble PVA) in 0.4mm layer height. This could decrease the print time.

Revenger72, 2015-06-24:
I join the request. I also need the ability to change the distance from the object to the support.

CodeChief, 2016-01-10:
@alexrj You say "I really don't see why this algorithm would change for each print or why user should decide to use a different value". Okay then there's a problem with the algorithm. Should be many times the nozzle width, at least 2. That won't hurt the perfect printers, but it will help all the average printers and new users out a lot! Please add the setting and increase the default.

lordofhyphens, 2016-02-15:
From looking at the code, appears to be defined as one support extrusion width for XY (at least for the interfaces).

lordofhyphens, 2016-03-17:
#1539

jfpion, 2016-04-06:
the algorithm is working well if you want more clearance from the support all you have to do is to increase the width of the support in the advanced tab. the point is why is it not documented ?

lordofhyphens, 2016-04-06:
That has some side effects @jfpion. I am working on a tweak to be able to
increase the separation distance, but I have been quite busy lately and
testing the change is not easy for me (automatic tests that is) as I am
unfamiliar with how that is done.

Also the documentation is old and in need of pull requests.

lonelymyp, 2016-04-06:
This is madness.
I'm more than a year suffer tearing stuck support.
I could not even think that the extrusion width parameter actually refers not to the width.

[image]

lordofhyphens, 2016-04-06:
It is both actually. In the code @alexjr used the single width for the xy
separation. I have it on my list of things to change but once you make it
tweakable like that other things can break :(

jfpion, 2016-04-06:
it refers to the width but has also an impact on the distance from the part

may be it could be simple to use a multiplier factor (from 1 to 4 could be enough) to be added in the algorithm used to place the supports. this way this could be a minimal change in the code. 

just my 2 cents

lonelymyp, 2016-04-06:
increases the width from 0.4 to 0.7 
at first glance there is nothing terrible, but the support does not stick

jfpion, 2016-04-06:
not stick to what ? in a good way as easier to detach, or a bad way as not stick to the bed and fall down ?

lordofhyphens, 2016-04-06:
My plan was to just add a box for a constant (in mm). Just have to do it. :)

lordofhyphens, 2016-04-09:
Added a new branch to my fork. The config option is there, but apparently I
was hallucinating when I saw changing clip_with_object() in
SupportMaterial.pm as the way to set this up. @alexjr does this sound
correct to you?
On Apr 6, 2016 3:22 PM, "Joe Lenox" <lordofhyphens@gmail.com> wrote:

> My plan was to just add a box for a constant (in mm). Just have to do it.
> :)
>

jfpion, 2016-04-15:
I've made some test setting the support width to 0.9 and printing 0.4 with 0.4 nozzle. the difference is stunning. 
and side effect (good one from my point of view, it's bloking the support generation for small holes like screw hole to be stuck by support . for 2 mm hole if you specify 0.9 clearance there no room for support.

and it is great as it solve an old  request  (dont generate support for small hole)
i'm waiting for the next version as I don't kown how to install your fork on my computer

thank you a lot

CodeChief, 2016-04-15:
That's exactly the two major headaches I have. Well done with the solution! Please make a build available for Windows install ASAP. I can't print anything with support until that is available.

lordofhyphens, 2016-04-15:
The main reason for why it is not in this repo (and instead is in my fork)
is that I don't want to merge my own PRs. Better that @alexrj gets a set of
eyes on it first.

I will see about getting an unsupported Windows build going.

lordofhyphens, 2016-04-15:
@CodeChief you can fake an increase in separation by increasing the extrusion width, as that's the default for "how close supports are to the model in X/Y".

If you can set up (see wiki instructions) and build from source my branch (see my pull request further up), I'd appreciate feedback.

@jfpion it's very similar to doing a build otherwise. If you're a Windows user then I can't currently help you as I have no idea how @alexrj makes his executable packages.

jfpion, 2016-04-15:
no real need to hurry and break something! there is a work around, not perfect, but working
but it look like killing 2 bird with one stone, 

thank you

jfpion, 2016-04-15:
> @jfpion it's very similar to doing a build otherwise. If you're a Windows user then I can't currently help you as I have no idea how @alexrj makes his executable packages. 

nor do I ..sadly I know enough to understand the logic of the pb but not enough to contribute other by giving idea .

many thank to the ones who know giving some of their time to the others, never said enough

lordofhyphens, 2016-06-27:
I'm going to close and lock this, please use the PR thread or #1539 for discussion. 
@jfpion the wiki instructions for performing a build worked first time with no fiddling for me on Windows 10 with Strawberry Perl 5.22

closed 2016-06-27 by lordofhyphens