slic3r/Slic3r #2917: Feature request: Enforce support for not only first layers

Zuborg, 2015-06-16:
Existing feature works fine with bottom of models. But if model has overhangs above enforced level, only one support/interface layer generated. Generally this means narrow overhangs will be broken before its reached the main model body. It will be very nice if enforced layers at top of support columns can be computed about the bottom of overhangs same as for model's bottom.

alexrj, 2015-06-16:
Can you provide a full test case please?

Zuborg, 2015-06-17:
This part was broken after begin of the print middle overhang
[image]
Config: https://drive.google.com/file/d/0BzAa6v5hnj3fcVQ3M2xHMzJzYms/view?usp=sharing
STL: https://drive.google.com/file/d/0BzAa6v5hnj3feGVjWUNKTFE0dTA/view?usp=sharing