slic3r/Slic3r #2918: Slic3r 1.2.8 prints gaps between infill edges and perimeter, then does gapfill

hyperair, 2015-06-16:
The shape's a simple cylinder, and it's doing one of the top solid layers (but not the topmost). The topmost solid layer appears to be done correctly (possibly due to a different extrusion width).

[image]

Test STL: https://gist.github.com/hyperair/3e658c7079345526f3c3

hyperair, 2015-06-16:
G-code in question: https://gist.github.com/62c2f0124f3892e380c2

a4jp-com, 2015-06-16:
https://github.com/alexrj/Slic3r/issues/2913

My point is kind of the same. The infill isn't touching the perimeter.

alexrj, 2015-06-16:
@hyperair, actually that's not gap fill. That's internal infill.  Look at layer 5.20mm:

[image]

Then let's go down one layer:

[image]

As you see, that's normal internal infill which is built below the upper part.
We could say that it's very narrow and Slic3r should merge it with the adjacent solid infill, but I'd say that this behavior does not create problems as it's not visible from the outside and it's also consistent with what Slic3r normally does for any shape.

hyperair, 2015-06-17:
Oh yeah you're right. Sorry for the noise.

closed 2015-06-17 by hyperair