slic3r/Slic3r #2919: 1.2.8-dev settings missing

a4jp-com, 2015-06-16:
All the saved settings are missing from the plater view.

[image]

alexrj, 2015-06-16:
What are your Wx and wxWidgets versions?  You can read them by launching Slic3r with the `--debug` option.

a4jp-com, 2015-06-17:
I'll check. Thank you for the reply.

alexrj, 2015-06-17:
Can you try to update wxWidgets? You need to issue the following command: `cpanm -v Alien::wxWidgets`.

a4jp-com, 2015-06-17:
Done.

[image]

alexrj, 2015-06-17:
Hm, that screenshot shows that wxWidgets wasn't actually updated to 3.0.2. Can you now do `cpanm -v --reinstall Wx` ?

a4jp-com, 2015-06-17:
Doing it just now ^^

[image]

a4jp-com, 2015-06-17:
The options are back now just a few icons are messing but everything is good ^^ Thank you soooo much.

[image]

a4jp-com, 2015-06-17:
Just the filament icons are missing. Maybe this is very easy to fix now.

xoan, 2015-06-17:
Filament dropdown menu options has no iconos but square color you can configure un The proper Filament Settings tab.

a4jp-com, 2015-06-17:
I see. I changed the color and the icon changed. So I guess that is all good but none of the actual preview colors changed. What are the colors for? Just the menu icons? Or should the model change color as well?

xoan, 2015-06-17:
Just for easy identification.

On Wed, Jun 17, 2015 at 7:03 PM, a4jp-com <notifications@github.com> wrote:
>
> I see. I changed the color and the icon changed. So I guess that is all good but none of the actual preview colors changed. What are the colors for? Just the menu icons? Or should the model change color as well?

a4jp-com, 2015-06-17:
Okay. So I can close this as everything works now ^^

Just one last thing, if the actual 3D Preview color changed this would be very cool.

Or the top filament default should also have the color square icon and not the actual filament icon as it is a bit confusing.

All my colors are white so it just looks like the icons are missing. 

Even a light #ccc 1px border would be nice.

a4jp-com, 2015-06-17:
@alexrj thanks for fixing this.

alexrj, 2015-06-17:
Yes, I'll probably add a 1px border. But I don't know what would be the appropriate color for the default preset, so I think it's better to stick with a generic icon.
Using that color for the 3D preview is definitely in my plans.

closed 2015-06-17 by alexrj

a4jp-com, 2015-06-17:
Okay ^^