slic3r/Slic3r #2920: [1.2.8] Different extrusion width in some first solid layer portions

xoan, 2015-06-16:
First of all, sorry for the weird title (I have no real idea on how to abstract this issue).

STL file: [`3Dponics-Drip-Nozzle-3_4-inch-3-holes.STL`](https://github.com/3dponics/Drip-3Dponics/blob/master/Drip%20Nozzles/Drip%20Nozzle%20(3_4%20inch%2C%203%20holes)/3Dponics-Drip-Nozzle-3_4-inch-3-holes.STL)
Configuration file: [`FoldaRap - 1.2.8.ini`](https://gist.github.com/xoan/314f57d2cd5255d42d44)

[image]
[image]

The issue appears also in the 2.25, 2.75 and 3.75mm layers but here in 1.75mm (7th layer) is very noticeable.

If it is needed, I can paste the `.gcode` file too.

xoan, 2015-06-16:
Same issue here:

STL file: [`3Dponics-Conduit.STL`](https://github.com/3dponics/Drip-3Dponics/blob/master/Conduit/3Dponics-Conduit.STL)
Configuration file: (same as above)

[image]

Right side seems ti be correct, while left side seems to be over extrudate (the part of the conduits near the middle one)

alexrj, 2015-06-16:
@xoan, regarding the first model… it does look ugly, but those are solid shells above overhangs. The inner large hole is not regular since it appears to be threaded, so each layer has a thin overhang, which generates a solid shell with bridging flow:

[image]

The toolpaths might be made more elegant (I can't figure how they should be), but I think they're not strictly wrong either - i.e. causing a badly printed object.

alexrj, 2015-06-16:
Also, see #240

xoan, 2015-06-16:
Yep, it is a threaded hole, so it has some areas with top/bottom solid shells. My first picture shows one of this areas with a top solid shell, and a portion of this with the first solid layer (I have configured 2 top/bottom solid layers). Here is with the strange extrusion hilighted:

[image]

And this is the same layer for the first test case sliced in Slic3r-1.2.7:

[image]

I know that the first solid layer inherit bridge parameters, but some of this layers are different than others first solid layers in the same part (as shown in the second test case.)

xoan, 2015-06-16:
I've searching and this is maybe related to #2899 and therefore it is fixed (even it is not a bug) in 38a9e32a287c4a601e256ebd96533f68c0c8dcc0

I can't compile current dev here, son I can't confirm that :\

xoan, 2015-06-17:
Yep, it is fixed in 1.2.9-experimental.

[image]

Thanks

closed 2015-06-17 by xoan