slic3r/Slic3r #2921: [1.2.8] single wall object not sliced with X-Y compensation Enabled

simone97, 2015-06-16:
I notice this bug since version 1.2.6 as in that version, the slicing process never ends.
With the new version, the slicing process ends but gives me only the skirt lines.
If I set X-Y compensation to 0 the file is correctly sliced.

STL (0.5mm-thin-wall.stl):
http://www.thingiverse.com/download:17273
Screen (I made a screen of RepetierHost but the gocde was generated with Slic3r):
[image]

simone97, 2015-06-16:
; generated by Slic3r 1.2.8 on 2015-06-17 at 00:26:15

; external perimeters extrusion width = 0.30mm
; perimeters extrusion width = 0.34mm
; infill extrusion width = 0.34mm
; solid infill extrusion width = 0.34mm
; top infill extrusion width = 0.34mm

M107
M107 ; Turn off fan
M190 S60 ; Activate bed and wait
G28 ; home all axis
G90; Set absolute coordinates
M82 ; Extruder in absolute mode
G1 Z5 F5000 ; lift nozzle
M109 T0 S220 ; Activate hotend and wait
G28 Z0 ; home Z axis
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
G1 Z0.250 F18000.000
G1 E-2.00000 F2400.00000
G92 E0
G1 Z0.350 F18000.000
G1 X85.099 Y85.241 F18000.000
G1 Z0.250 F18000.000
G1 E2.00000 F2400.00000
M204 S1250
G1 X85.241 Y85.099 E2.00357 F5819.352
G1 X86.000 Y84.462 E2.02122
G1 X86.164 Y84.347 E2.02479
G1 X87.022 Y83.852 E2.04243
G1 X87.204 Y83.767 E2.04600
G1 X88.135 Y83.428 E2.06366
G1 X88.329 Y83.376 E2.06723
G1 X89.305 Y83.204 E2.08488
G1 X90.004 Y83.165 E2.09735
G1 X109.996 Y83.165 E2.45339
G1 X110.491 Y83.186 E2.46222
G1 X110.695 Y83.204 E2.46587
G1 X111.671 Y83.376 E2.48352
G1 X111.865 Y83.428 E2.48708
G1 X112.796 Y83.767 E2.50474
G1 X112.978 Y83.852 E2.50831
G1 X113.836 Y84.347 E2.52595
G1 X114.000 Y84.462 E2.52952
G1 X114.759 Y85.099 E2.54717
G1 X114.901 Y85.241 E2.55074
G1 X115.538 Y86.000 E2.56839
G1 X115.653 Y86.164 E2.57196
G1 X116.148 Y87.022 E2.58961
G1 X116.233 Y87.204 E2.59318
G1 X116.572 Y88.135 E2.61083
G1 X116.624 Y88.329 E2.61440
G1 X116.796 Y89.305 E2.63205
G1 X116.835 Y90.004 E2.64453
G1 X116.835 Y109.996 E3.00056
G1 X116.814 Y110.491 E3.00939
G1 X116.796 Y110.695 E3.01304
G1 X116.624 Y111.671 E3.03069
G1 X116.572 Y111.865 E3.03426
G1 X116.233 Y112.796 E3.05191
G1 X116.148 Y112.978 E3.05548
G1 X115.653 Y113.836 E3.07313
G1 X115.538 Y114.000 E3.07670
G1 X114.901 Y114.759 E3.09435
G1 X114.759 Y114.901 E3.09792
G1 X114.000 Y115.538 E3.11557
G1 X113.836 Y115.653 E3.11914
[... 434 lines cut ...]
; extra_perimeters = 1
; fill_angle = 45
; fill_density = 30%
; fill_pattern = rectilinear
; gap_fill_speed = 5
; infill_every_layers = 1
; infill_extruder = 1
; infill_extrusion_width = 0
; infill_overlap = 1%
; infill_speed = 250
; overhangs = 1
; perimeter_extruder = 1
; perimeter_extrusion_width = 0
; perimeter_speed = 50
; perimeters = 3
; small_perimeter_speed = 10
; solid_infill_below_area = 70
; solid_infill_every_layers = 0
; solid_infill_extruder = 1
; solid_infill_extrusion_width = 0
; solid_infill_speed = 100
; thin_walls = 1
; top_infill_extrusion_width = 0
; top_solid_infill_speed = 50
; top_solid_layers = 3

peter1960, 2015-06-16:
I have seen this behavior via Repetier/MatterControl as well where it never completes but works ok with slic3r standalone.

simone97, 2015-06-16:
I've alse tried to export the gcode directly from slic3r and gives the same result.

alexrj, 2015-06-17:
It's not a bug.
Despite the name says 0.5mm, your test object has 0.25mm walls. If you apply a size compensation of -0.1mm (for each side) you'll trim it down to 0.05mm walls, which is filtered out by Slic3r because it's quite thin to be extruded with a 0.3mm nozzle.

closed 2015-06-20 by alexrj