slic3r/Slic3r #2925: No individual FILAMENT settings for parts

cocoknight, 2015-06-18:
I got some new types of filament yesterday and wanted to try out different print parameters to get the best print results. I added my test part several times and created different filament profiles. I activated "complete individual objects" and tried to apply for each part an individual filament setting. Unfortunately Slic3r does not let me do that. I'm now asking myself if this behaviour is desired or if it is a bug. Would be great if it worked for parameter studies.
[image]
Slic3r 1.2.9, Windows 7 x64

Regards
Stefan

pyrophreek, 2015-06-18:
This would also be useful for setting up different print settings for different layers when you are doing layer selective printing.  

This goes back to previous posts discussing the possibility of applying presets to the different parts or modifier meshes.

alexrj, 2015-06-18:
The filament options are difficult to vary across a single print. For example, changing temperature while going back and forth between multiple objects will likely require pauses, but pauses require a parking point and a priming strategy. Not easy.

harriv, 2015-06-18:
Going back and forth between multiple objects doesn't happen when using sequential printing?

alexrj, 2015-06-19:
@harriv, using per-object settings doesn't imply sequential printing...

cocoknight, 2015-06-19:
But activating this option could enable using different filament settings.

2015 10:08:

NimrodQuimbus, 2015-06-27:
I have found that Slic3r 1.2.9 is now only loading the the Filament settings for Extruder1 and ignoring Extruder2. 

Print settings>Multiple extruders always uses filament settings for extruder 1 no matter what.