slic3r/Slic3r #2928: Export STL creates corrupted STL file (1.2.9 Windows 32 and 64 bit)

Rysium, 2015-06-19:
In 1.2.1 and before the export STL function produced STL file that could be opened. The problem started with version 1.2.2 and continues to the latest version 1.2.9.  When model is exported to STL the file created is somehow corrupted.  If I try to open it with slic3r again I'm getting message "This STL file couldn't be read because it's empty" -  the file is not empty, it has many, many bytes :-)

To repeat the problem try to load Garden Toad (http://www.thingiverse.com/thing:35595) and just export STL without any changes. Saved STLl will refuse to open.

Slic3r reports 517 error which are fiexd but I don't think it is the source of the problem. The same toad "repaired" with version 1.1.7 and exported to STL opens OK with version 1.2.9, but then exporting to STL from version 1.2.9 creates corrupted file which cannot be opened anymore.

The other issue with this toad is that slic3r crashes when I try to cut 5mm from the bottom.  Some numbers for this cut work, some don't.

[image]

Ryszard

alexrj, 2015-06-20:
Thank you. I fixed this. Can you please try downloading 1.2.9a from http://dl.slic3r.org/download ?

The error when cutting the model is caused by the model not being manifold, and Slic3r not being able to do miracles for repairing it. :)

Rysium, 2015-06-21:
Thank you.  I installed 1.2.9 (86) Stable and tested on my old laptop with Wondows XT.  The STL Export writes file OK.  THANK YOU.

Cutting the toad between 4.7 and 5.9 causes the program to crash.  However cut at 4.6 and lower or 6.0 and higher is OK.  I understand that the cut may cause the model to be non-manifold.   
Would it be possible to not crash but just not cut and show relevant error message?

Thank you so much for your development.  You saved my day :-)

Ryszard

Rysium, 2015-06-21:
Thank you so much for super fast response.  I downloaded 1.2.9a and saving
the toad was OK.  Would it be possible to not crash the program but show
some message instead when cut produces non-manifold model?  I don't know
how hard it would be especially to "undo" the cut as the crash is happening
while building already cut model.

Thank you again,
Ryszard

closed 2016-03-19 by alexrj