slic3r/Slic3r #2930: Contact Z-distance is double what is supposed to be

dropinsight, 2015-06-20:
I have created a model to test support materials and noticed that Slic3r 1.2.9 has an issue when creating a 'Contact Z-distance' that is not equal to zero. For instance when the 'Contact Z-distance' is set to 0.2mm the distance is actually 0.4mm. This happens nearly for all the features I have created in that model. Here are some images to show what I am talking about (this is going to be a cylinder where the axis of the cylinder is parallel to the print bed):

Layer Height 3.2mm
[image]

Layer Height 3.4mm
[image]

Layer Height 3.6mm
[image]

Layer Height 3.8mm
[image]

alexrj, 2015-06-21:
You haven't provided the full information required by the guidelines. However please keep in mind that when extruding in free air you'll get a round extrusion whose diameter is more or less equal to the nozzle diameter. So if you want a 0.2mm airgap distance between the object and support, you need to raise the extruder by 0.2mm + *nozzle_diameter*.

dropinsight, 2015-06-21:
Frist of all thank you for all your work. I love Slic3r and is still my number one slicer. I understand where you are going with this. This should help when using detachable support. I use PVA as a support and just want more spacing between the object and the support. I know the 0mm setting is intended for support. But please see issue 2927 about why soluble support is a bit to tight. Also I using this model to test how to detach and dissolve PVA the best. I attached the model with which i am testing currently support. Maybe it could also help with your development.
Sorry about the incomplete report. Here are some more information:
[image]

Here is the model:
https://www.youmagine.com/designs/support-model-test

alexrj, 2015-06-21:
So this is a duplicate of #2927? Can we close this?

dropinsight, 2015-06-21:
It is not exactly a dupilcate. Both issue are occured using the same model but not the same settings. Issue 2927 is related to when the Contact Z-distance is set to soluble 0mm. It fuses the support material with the build material on the dome. This issue here is related to Z-distance when it is not set 0mm, so for instance 0.2mm. I understood your explanation, your approach is not very intuivtive when someone uses 0.2mm as a offset.

alexrj, 2015-06-21:
It's not "my" approach. It's physical reality. What do you think Slic3r should do if you ask it to leave a 0.2mm air gap? I only know this solution...

dropinsight, 2015-06-21:
Well the issue I have with the Z-distance is how you approach the solbule and the detachable. If I choose soluble so 0mm z-distance, Slic3r prints right on the print material, so there is no noozle diameter included in this z-distance, but when I change that it to something else other than 0mm it includes the nozzle diameter. This means either I choose 0mm or another offset unequal to zero but with a added nozzle diameter. It is more intuitive to explain how you would approach printing detachables so z-offset + noozle diameter, so people set it to 0.2mm + 0.4mm = 0.6mm, rather than Slic3r adding the nozzle diameter himself. Because at it is right now, it either go 0mm or noozle diameter + a Z-offset and nothing in between.

alexrj, 2015-06-21:
I disagree.  An option whose purpose is to set the *distance* between support and object MUST consider the actual round shape of the extrusion. Otherwise it would do the wrong thing.

dropinsight, 2015-06-21:
You may see it that way, but it does not help in finding the right settings. You are basically saying it must be this way, because the theory tells me it must work this way. I tried a lot of prints and I find the results do not match the theory. The cylinder I printed is just not round and he could be better if I just could work with different Z-distances. But well it is your slicer. Thank you for taking time in answering this issue. Please do still look at issue 2927.

alexrj, 2015-06-21:
I still don't understand what's the actual problem you're reporting...

dropinsight, 2015-06-29:
Sorry for the late reply but I was very busy last week. 

When I put 0mm in the Z-distance, so soluble I get this:
[image]

Lets say for instance that soluble is a too tight fit for the build material or that soluble takes too long to dissolve, so I want to make one layer height of space between the build and support material, so 0.2mm. I would think now that I would get this result:
[image]

But actually I get this result:
[image]

This seems to be the correct behaviour according to your explanation, but this is not the behaviour a user would expect. What I am trying to so say is that the number that are being typed into Slic3r do not correlate with the outcome. I understand your explanation and why you are doing it, but I would rather advise people to set a 0.4mm distance if they want a good detachable support instead of that people type 0.2mm and get a gap of 0.4mm.
For instance Simplify 3D does it the way I describe here. 

If for instance I type 0.1mm in Z-distance, I would like to see this result:
[image]

sarf2k4, 2015-10-17:
Hi, I also want to point out about the z contact distance or airgap, I've been fiddling with it using 20mm cube with raft. with first layer of 0.1, layer height 0.2, the raft should be 0.3 each. 

So the raft finished at 0.7, based on what you explained "not a bug" it should be start laying plastic on 1.1 instead of 0.9. But my case here, it starts laying the plastic on the raft at 1.3mm instead of 1.1mm, that would be a total of 0.6mm increment instead of 0.4mm increment from the thread starter said.

Even though I put in 0.1, the plastic will get laid at 1.2mm and tried using 0.001mm on z contact distance, now it goes to 1.1mm

I am using 1.2.9, Is there anything that I'm missing causing the slic3r somewhat tripled the gap?

markwal, 2015-10-17:
@wooddragon48, that is all that is in the dropdown, but you can type a different value into the box.

rokk0, 2016-06-21:
Have same issue. I only need nozzle to be (layer height + 0.1 mm) over the support layer, but not (nozzle diameter + 0.1 mm). Its ridiculous, when you printing 0.1 layer with 0.4 nozzle and getting 0.5 mm gap instead of 0.2 mm.

markwal, 2016-06-21:
@rokk0, but how is it accomplished? If you put the nozzle at 0.2mm above the previous layer and extrude a 0.4mm diameter bead, it will touch, in fact it will more than touch, it'll fuse which isn't much like a contact distance.

ruomserg, 2017-04-05:
I did some research on this issue with 0.3mm nozzle/0.2mm default level height and can report the following:

- Any positive contact distance is pretty much unusable with current support material generation logic. Having contact_distance>0 activates bridge mode for the first supported model layer and resets extrusion width to nozzle diameter (as it extrudes to thin air) and starts fan max-speed (if enabled in filament settings).
- Extruding filament with fan on and layer height greater then nozzle diameter gives no chance for it to bind to the support structures. I observe heavy perimeter distortion caused by extruded filament been dragged over solid interface layers. If printing in ABS - supported layers also tend to deform and delaminate a few layers later.
- Zero contact distance provides good filament bonds, but makes almost impossible to break support structures from the model. I ended up making interface layer sparse and thin (0.5mm distance, 0.2-0.25 width). Printed that way, support structures break off between interface layers. But this method leaves at least one interface layer fused to the model and requires post-print processing (OK for ABS, almost impossible for PLA)
- I've tried to change current support material logic by editing "sub contact_distance" in SupportMaterial.pm by using current layer height as a base instead of nozzle diameter.
- So far I can report repeatable good results with layer_height+0.1mm contact distance. Extruded filament is nicely bound to solid interface layers, but the bond is weaker then between normal layers. Using a sharp knife, I can pop support layer off leaving smooth model surface.
- Doing some math (assuming extrusion cross-section shape is the same and the area (w-h)*h+Phi*(h/2)^2 constant) I've found that you can get about 62% normal contact area (w) between top interface layer and a model with a layer height approximately 1.2w/(4-Phi). It seems be possible to get even less contact area but at the expense of the extrusion lines not contacting their sides.

Overall, I recommend to change sub contact_distance to return $layer_height + $extra; instead of  $nozzle_diameter + $extra; Anyone who desire to extrude in thin air still could do this just by increasing contact_distance to appropriate value.

lordofhyphens, 2017-04-05:
The bridging flow over support interface is intentional, as without a layer
below it to squish against it is impossible to define the layer height.

lordofhyphens, 2017-04-05:
There is a PR out there to do just that, give it some printing experiments
;)

lordofhyphens, 2017-04-05:
https://github.com/alexrj/Slic3r/pull/3379 #3376

ehanseng, 2017-12-27:
A short explenation, I hope be clear. 
https://docs.google.com/presentation/d/1j9IupPthcPUwxF2mBdEuGB4oQgTDVU5HaXW9igvktAA/edit?usp=sharing