slic3r/Slic3r #2931: [Feature request] set (retract) speed to less than 1.0

nicksears, 2015-06-21:
I have an extruder with a really high e calibration 45000 because of a 19:1 gear down ratio AND 1.411 pitch thread. Since 0.074mm of filament travel translates to an entire rotation of the motor,  speed needs to be around 0.1mm/s.

I currently post process to make all the 1mm/s (F60; retract/in retract) into F6, etc (0.1mm/s). I previously ignored retractions (because they wouldn't work, they were trying to move too far)

Mikk36, 2015-06-21:
Reduce microstepping on the stepper drivers?

nicksears, 2015-06-21:
Good thinking, but this would decrease torque/resolution and the actual
calibration number, but the speed of the retraction would still be the same.

Basically, since the diameter is so big (10x) extrusion/retraction needs to
be MUCH slower (100x).

The optimal solution is to allow a decimal value for the retraction (or
any?) speed, but for now I've found a work around: decrease the diameter
artificially by 10x, and the calibration by 100x, since this will give the
same steps/mL calculation. Alternatively by decreasing the diameter by any
number and calibration by that numbers square.

Testing today and trying the actual amount of retraction needed, I'm
thinking ~10uL=0.01mL=0.044mm. This means 4.4mm for the "1.75mm filament
equivalent", which seems reasonable.

Reduce microstepping on the stepper drivers?

nicksears, 2015-06-21:
Note this is not version or is dependant but I'm using win8.1x64 and 1.2.9x64

alexrj, 2015-12-19:
As of current master, decimals are allowed in the retract_speed option.

closed 2015-12-19 by alexrj