slic3r/Slic3r #2932: Weird arrangment when adding more than 4 copies of a parts

gege2b, 2015-06-21:
Hi there
It's a bug that I've see since 1.2.7 exp, but maybe was here before
Win7 x64 with the latest 1.2.9a x64

when adding more copies of some STL (using sur "+" button), the first 4th copies autoarranged correctly, but 5th+ are just superposed with a little offset on X & Y (see screen)

[image]

Clicking "Autoarrange" does nothing

a STL where the problem occurs : https://dl.dropboxusercontent.com/u/34934879/24h%2B.STL
Here is the config : https://dl.dropboxusercontent.com/u/34934879/copies_config.ini
Gcode is irrelevant

Complementary info :
Right now, I used to slice this : https://www.youmagine.com/designs/toothpaste-squeezer
As you can see, the object isn't correctly oriented, so I rotated him. And added some (up to 12) copies and the problem didn't occurs at all, they was all arranged correctly, even if out of printing area.
Tried without rotating, and the problem was back again after the 4th copy

Tried with the default config and the first linked STL, problem still occurs but after the 9th copy
Seems to be some math related to bed size etc...

cheers

a4jp-com, 2015-06-21:
I had no problem adding models 1cm by 1cm blocks or just pushing the plus button.

[image]

I had this problem with your model from the 10th model, just using the plus button. I can drag and drop parts into the correct position though.

[image]

gege2b, 2015-06-22:
yeah, I forgot to say that drag&drop of the superposed objects works anyway

My guess is that it depends of : 
- the size of the object
- the size of the bed
- maybe the rotation
so the 1cm cube you used is maybe just OK, mathematically speaking ^_^

alexrj, 2017-03-16:
Thank you, this behavior was recently fixed.

closed 2017-03-16 by alexrj

a4jp-com, 2017-03-18:
Nice ^^