slic3r/Slic3r #2935: segfault

twistoneup, 2015-06-21:
i've been having issues with slic3r for some unknown reason; first it ran fine, then it caused issues, and i didn't change anything on the system afaik.

i'm getting this:
slic3r[5042]: segfault at 1 ip 00007f110f2aedb8 sp 00007fff3bd14050 error 6 in libXrender.so.1.3.0[7f110f2aa000+9000]

thanks
twistoneup

alexrj, 2015-06-21:
This report is very incomplete. Why haven't you read the guidelines?

twistoneup, 2015-06-22:
On 06/21/2015 02:15 PM, Alessandro Ranellucci wrote:

> This report is very incomplete. Why haven't you read the guidelines?

because i'm new to github, that's why.

i'll read the guidelines and supply more info, if you would like.

alexrj, 2015-06-22:
Please do, otherwise no action can be taken...

twistoneup, 2015-07-08:
Descriptive title
	Slic3r segfault linux ubuntu

Steps to trigger the problem that are specific, and repeatable
	launched from launch bar icon that runs (Application in Terminal):
		gksudo /root/slic3r.sh
	/root/slic3r.sh looks like this:
		cd /opt/Slic3r/bin
		./slic3r
		sleep 15
		cd /root

What happens when you follow the steps:
	the program launches, i see the Slic3r layout and menu appear, then disappear.
	one of two things then happens - either:
	(1)	segfault (i haven't been able to recreate this, so no segfault info...  sorry!)
	-or-
	(2) slic3r: Fatal IO error 11 (Resource temporarily unavailable) on X server :0.0.

What you expected to happen instead.
	i expected the Slic3r layout to remain visible and to use Slic3r.
	
Include the exact text of any error messages if applicable (or upload screenshots).
		root@YGH:/opt/Slic3r/bin# ./slic3r
		slic3r: Fatal IO error 11 (Resource temporarily unavailable) on X server :0.0.
		root@YGH:/opt/Slic3r/bin#

Slic3r version:
		/opt/Slic3r/bin/slic3r --help | more
	shows:
		Slic3r 1.1.7 is a STL-to-GCODE translator for RepRap 3D printers
		written by Alessandro Ranellucci <aar@cpan.org> - http://slic3r.org/

Did this work in a previous version? If so, also provide the version that it worked in.
	the *same version* used to work fine.  perhaps ubuntu maint caused this issue to arise; i'm not certain.

OS version
	Unbuntu 14.04.1

please lmk if this is incomplete or you need more info.
thanks!
TwistOneUp