slic3r/Slic3r #2936: 1.2.9. crashes in spiral vase mode that was OK in 1.1.7

EldRick, 2015-06-21:
I have a simple hex "vase" that crashes S3r 1.2.9 when Exporting G-code if Spiral Vase is selected.
The .stl file can be found at www.clearlyadjustable.com/Temp/171hex.stl 
generated .gcode from S3r 1.7.1 is in the same directory.

alexrj, 2015-06-21:
I can't make that model crash with default settings. Can you please attach yours?
*File -> Export config…*

EldRick, 2015-06-21:
I’ve uploaded a copy of the config.ini and also crashreport.txt from the failure to www.clearlyadjustable.com/Temp/ <http://www.clearlyadjustable.com/Temp/>

Best,
Rick

alexrj, 2015-06-21:
This is not strictly related to spiral vase.  I'm copying the stack trace here:

```
Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libperl.dylib                 	0x000e71cc Perl_pp_entersub + 1564
1   libperl.dylib                 	0x00074bcd Perl_call_sv + 509
2   libperl.dylib                 	0x000c9fcc S_invoke_exception_hook + 492
3   libperl.dylib                 	0x000cbe33 Perl_vwarner + 243
4   libperl.dylib                 	0x000cbeac Perl_warner + 44
5   libperl.dylib                 	0x000f0a76 Perl_sv_free + 182
6   libperl.dylib                 	0x000a911d Perl_cv_undef + 701
7   libperl.dylib                 	0x000f0f47 Perl_sv_clear + 1111
8   libperl.dylib                 	0x000f0985 Perl_sv_free2 + 85
9   libperl.dylib                 	0x0011a411 Perl_free_tmps + 97
10  libperl.dylib                 	0x000781c6 perl_destruct + 182
11  threads.bundle                	0x05a56fc4 S_ithread_clear + 196
12  threads.bundle                	0x05a5a9ba XS_threads_detach + 458
13  libperl.dylib                 	0x000e74a6 Perl_pp_entersub + 2294
14  libperl.dylib                 	0x000e5af6 Perl_runops_standard + 22
15  libperl.dylib                 	0x00074ca5 Perl_call_sv + 725
16  Wx.bundle                     	0x0746fe67 wxPliEventCallback::Handler(wxEvent&) + 1111
17  libwx_baseu-3.0.0.0.0.dylib   	0x07f27d9f wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const + 95
18  libwx_baseu-3.0.0.0.0.dylib   	0x0806f7a8 wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) + 104
19  libwx_baseu-3.0.0.0.0.dylib   	0x0806ff98 wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 88
20  libwx_baseu-3.0.0.0.0.dylib   	0x08070904 wxEvtHandler::TryHereOnly(wxEvent&) + 52
21  libwx_baseu-3.0.0.0.0.dylib   	0x080709ac wxEvtHandler::ProcessEventLocally(wxEvent&) + 60
22  libwx_baseu-3.0.0.0.0.dylib   	0x08070a21 wxEvtHandler::ProcessEvent(wxEvent&) + 81
23  libwx_baseu-3.0.0.0.0.dylib   	0x08071daf wxEvtHandler::ProcessPendingEvents() + 319
24  libwx_baseu-3.0.0.0.0.dylib   	0x07f28f20 wxAppConsoleBase::ProcessPendingEvents() + 144
25  libwx_baseu-3.0.0.0.0.dylib   	0x08039abb wxCFEventLoop::CommonModeObserverCallBack(__CFRunLoopObserver*, int) + 155
26  com.apple.CoreFoundation      	0x93d1c81e __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 30
27  com.apple.CoreFoundation      	0x93d1c760 __CFRunLoopDoObservers + 400
28  com.apple.CoreFoundation      	0x93d0e0c2 __CFRunLoopRun + 946
29  com.apple.CoreFoundation      	0x93d0daa6 CFRunLoopRunSpecific + 390
30  com.apple.CoreFoundation      	0x93d0d90b CFRunLoopRunInMode + 123
31  com.apple.HIToolbox           	0x921918f8 RunCurrentEventLoopInMode + 262
32  com.apple.HIToolbox           	0x92191503 ReceiveNextEventCommon + 192
33  com.apple.HIToolbox           	0x9219142c _BlockUntilNextEventMatchingListInModeWithFilter + 99
34  com.apple.AppKit              	0x92dfd721 _DPSNextEvent + 742
35  com.apple.AppKit              	0x92dfcdc5 -[NSApplication nextEventMatchingMask:untilDate:inMode:dequeue:] + 350
36  com.apple.AppKit              	0x92df177c -[NSApplication run] + 907
37  libwx_osx_cocoau_core-3.0.0.0.0.dylib	0x079bc685 wxGUIEventLoop::OSXDoRun() + 277
38  libwx_baseu-3.0.0.0.0.dylib   	0x080397c6 wxCFEventLoop::DoRun() + 38
39  libwx_baseu-3.0.0.0.0.dylib   	0x07f689ba wxEventLoopBase::Run() + 74
40  libwx_baseu-3.0.0.0.0.dylib   	0x07f28a3f wxAppConsoleBase::MainLoop() + 95
41  Wx.bundle                     	0x07375be2 wxPliApp::MainLoop() + 50
42  Wx.bundle                     	0x0731a8b1 XS_Wx__App_MainLoop(interpreter*, cv*) + 129
43  libperl.dylib                 	0x000e74a6 Perl_pp_entersub + 2294
44  libperl.dylib                 	0x000e5af6 Perl_runops_standard + 22
45  libperl.dylib                 	0x0007542e perl_run + 366
46  slic3r                        	0x00006d2b 0x1000 + 23851
47  slic3r                        	0x000045a5 0x1000 + 13733
```

EldRick, 2015-06-22:
Nevertheless, it works with 1.1.7, and crashes with 1.2.9 - same settings.

lordofhyphens, 2015-06-22:
@EldRick, @alexrj was copying it for his own information, not disputing
that there is a bug/regression.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

EldRick, 2015-06-22:
Thanks - I did misunderstand.

alexrj, 2017-03-18:
Newer builds are much more stable. This is a general stability issue (wxWidgets / threads / OS linked libraries etc.) so I assume we can safely close it.

closed 2017-03-18 by alexrj