slic3r/Slic3r #2937: Enable support features with modifier meshes

phizz166, 2015-06-21:
So I was trying to print this part the other day

http://www.thingiverse.com/thing:34343

With proper settings and good filament cooling, there is only a single part of the model that needs support: directly underneath his left hand. Everything else prints fine thanks to steep angles or bridging. However, for some reason I couldn't find a threshold angle for the support that would get only his hand. Setting it to e.g. 1 degree did nothing, even though the hand is fully overhung (even a bit negative). In fact, no angles below 13 degrees generated any support on the model, while 14 degrees created support in many different places (under the chest and head, etc). I don't want this.

That may be an issue in itself, but it's not what I'm interested in. I realized that I could just create a modifier mesh that covered a volume from the model's base plane up to the hand, and turn on "generate support" for that area. Problem solved...?

...but you can't alter support settings with modifier meshes, it seems. Can this functionality be added? It would be a good step on the way to something like Simplify3D's manual support definition, which is just awesome.

lordofhyphens, 2016-04-18:
Duplicate of #2570

lordofhyphens, 2016-12-06:
Closing as it is a dupe.

closed 2016-12-06 by lordofhyphens