slic3r/Slic3r #2938: Mysterious crash in xcb

darrenfreeman, 2015-06-22:
I am running slic3r from slic3r-linux-x86_64-1-2-1-experimental.tar.gz which I downloaded on 13 Nov 2014 and which contains a slic3r binary having datestamp 9 Nov 2014.

Yes I should be running it from git. Yes I should at least be running a newer version. But you know.. every time I update slic3r, something I need breaks completely :P

So I've been happily slic1ng away and then now I tried to run slic3r from the console and I get the following output. The main application window does appear, but then it aborts.

The Pango warning appears every time and I ignore it. The other stuff is new. It's possible that I've upgraded some packages since the last time it worked. I don't expect you to fix this on an old experimental version, I just think the bug report may be useful, since it's probably not a change in slic3r but in some other part of the system.

--------8<--------8<--------8<--------8<--------

(slic3r:8087): Pango-WARNING **: /usr/lib/pango/1.6.0/modules/pango-basic-fc.so: cannot open shared object file: No such file or directory
[xcb] Unknown request in queue while dequeuing
[xcb] Most likely this is a multi-threaded client and XInitThreads has not been called
[xcb] Aborting, sorry about that.
slic3r: xcb_io.c:179: dequeue_pending_request: Assertion `!xcb_xlib_unknown_req_in_deq' failed.

darrenfreeman, 2015-06-22:
Although if you have any idea what is causing this, please let me know :) I'm about to start guessing because I have no idea what this means. May end up breaking my system in the process.

darrenfreeman, 2015-06-22:
http://flavio.tordini.org/forums/topic/segfault-on-linux

(might be relevant)

darrenfreeman, 2015-06-22:
and:

http://forums.reprap.org/read.php?263,224469

lordofhyphens, 2015-06-22:
Well at least one of your problems is evident by the error message - the
version of pango slic3r is linked against is not on your system. Odds are
it was replaced and slic3r is being picky about the naming. Still, I can't
see anything that leads me to believe that this is a slic3r issue.

If you have to run 1.2.1 (check out 1.2.9a and if something you need broke,
make a post about it if it is not already a GH issue; remember to read the
guidelines),  compile that version from source against your current libs.
You would need to clone the repo from github and checkout the 1.2.1 tag.
Then follow the usual building instructions. As I think you are using a
fedora - based distro (from the names in the paths you provided) you could
see if there is a package for slic3r and adapt the source for that to the
version you need. It has the advantage of not breaking your distro. If you
are using a Debian based distro like Ubuntu or mint you can copy the debian
directory from my fork into a clean copy of the 1.2.1 source, run quilt -a
and build a Debian package.

Failing that, you could make a symlink in your slic3r directory to your
current version of pango and name it what slic3r is looking for. This might
have side effects or not work if pan's ABI changed from one version to
another.

lordofhyphens, 2015-06-22:
And yes you probably broke it on your system itself. So not much point to
posting an issue against an experimental version that is specific to your
distro config and we won't be able to reproduce. If it still does it with
1.2.9 then there might be something to this.

darrenfreeman, 2015-06-22:
I figured that was almost certainly the case, but since I found others having the same error message in the distant past, I thought it worth mentioning.

I have unpacked 1.2.9-stable and it appears to be working. Haven't used it yet, but I do note the same Pango error.

(slic3r:21441): Pango-WARNING **: /usr/lib/pango/1.6.0/modules/pango-basic-fc.so: cannot open shared object file: No such file or directory

I am running an up-to-date openSUSE 13.2 , so your binaries don't quite work on SUSE. I have 1.8.0 installed under an otherwise identical path as above.

darrenfreeman, 2015-06-22:
Looks to be working. And excellent work on the gcode view and the 3D preview :)

closed 2015-06-22 by darrenfreeman