slic3r/Slic3r #2941: Collapse in Preview.(1.2.9 Windows)

fuding, 2015-06-22:
Collapse in Preview.(1.2.9 Windows)

ZombieHunterHead_v4_90k.stl
http://www.thingiverse.com/thing:69709/#files

[image]

My Print.ini
------------------------------------------------
avoid_crossing_perimeters = 1
bottom_solid_layers = 3
bridge_acceleration = 0
bridge_flow_ratio = 1
bridge_speed = 80
brim_width = 0
complete_objects = 0
default_acceleration = 1000
dont_support_bridges = 1
external_fill_pattern = rectilinear
external_perimeter_extrusion_width = 0
external_perimeter_speed = 30
external_perimeters_first = 0
extra_perimeters = 1
extruder_clearance_height = 20
extruder_clearance_radius = 20
extrusion_width = 0
fill_angle = 45
fill_density = 25%
fill_pattern = honeycomb
first_layer_acceleration = 0
first_layer_extrusion_width = 170%
first_layer_height = 0.2
first_layer_speed = 25
gap_fill_speed = 15
gcode_comments = 0
infill_acceleration = 1000
infill_every_layers = 1
infill_extruder = 1
infill_extrusion_width = 0
infill_first = 0
infill_only_where_needed = 1
infill_overlap = 15%
infill_speed = 60
interface_shells = 0
layer_height = 0.2
max_print_speed = 80
max_volumetric_speed = 0
min_skirt_length = 0
notes = 
only_retract_when_crossing_perimeters = 1
ooze_prevention = 0
output_filename_format = [input_filename_base].gcode
overhangs = 1
perimeter_acceleration = 1000
perimeter_extruder = 1
perimeter_extrusion_width = 0
perimeter_speed = 40
perimeters = 4
post_process = 
raft_layers = 0
resolution = 0
seam_position = aligned
skirt_distance = 3
skirt_height = 1
skirts = 2
small_perimeter_speed = 20
solid_infill_below_area = 70
solid_infill_every_layers = 0
solid_infill_extruder = 1
solid_infill_extrusion_width = 0
solid_infill_speed = 40
spiral_vase = 0
standby_temperature_delta = -5
support_material = 0
support_material_angle = 0
support_material_contact_distance = 0.2
support_material_enforce_layers = 0
support_material_extruder = 1
support_material_extrusion_width = 0
support_material_interface_extruder = 1
support_material_interface_layers = 1
support_material_interface_spacing = 2
support_material_interface_speed = 100%
support_material_pattern = rectilinear
support_material_spacing = 3
support_material_speed = 50
support_material_threshold = 0
thin_walls = 1
threads = 4
top_infill_extrusion_width = 0
top_solid_infill_speed = 20
top_solid_layers = 3
travel_speed = 80
xy_size_compensation = 0
------------------------------------------------

My Printer.ini
------------------------------------------------
bed_shape = 116.856x12.2821,114.932x24.4296,111.749x36.3095,107.342x47.7916,101.758x58.75,95.0595x69.0648,87.3195x78.6228,78.6228x87.3195,69.0648x95.0595,58.75x101.758,47.7916x107.342,36.3095x111.749,24.4296x114.932,12.2821x116.856,0x117.5,-12.2821x116.856,-24.4296x114.932,-36.3095x111.749,-47.7916x107.342,-58.75x101.758,-69.0648x95.0595,-78.6228x87.3195,-87.3195x78.6228,-95.0595x69.0648,-101.758x58.75,-107.342x47.7916,-111.749x36.3095,-114.932x24.4296,-116.856x12.2821,-117.5x0,-116.856x-12.2821,-114.932x-24.4296,-111.749x-36.3095,-107.342x-47.7916,-101.758x-58.75,-95.0595x-69.0648,-87.3195x-78.6228,-78.6228x-87.3195,-69.0648x-95.0595,-58.75x-101.758,-47.7916x-107.342,-36.3095x-111.749,-24.4296x-114.932,-12.2821x-116.856,0x-117.5,12.2821x-116.856,24.4296x-114.932,36.3095x-111.749,47.7916x-107.342,58.75x-101.758,69.0648x-95.0595,78.6228x-87.3195,87.3195x-78.6228,95.0595x-69.0648,101.758x-58.75,107.342x-47.7916,111.749x-36.3095,114.932x-24.4296,116.856x-12.2821,117.5x0
before_layer_gcode = 
end_gcode = G28 ; Home to top 3 endstops when done\nM104 S0 ; turn off temperature\nM190 S0 ; \nM84     ; 
extruder_offset = 0x0
gcode_flavor = reprap
layer_gcode = 
nozzle_diameter = 0.3
octoprint_apikey = 
octoprint_host = 
pressure_advance = 0
retract_before_travel = 2
retract_layer_change = 1
retract_length = 3
retract_length_toolchange = 10
retract_lift = 0
retract_restart_extra = 0
retract_restart_extra_toolchange = 0
retract_speed = 100
start_gcode = M302 ; Allow cold extrusion to avoid USB errors\nM220 S100 ; Set feedmultiply back to normal (100%)\nM104 S[first_layer_temperature]  ; Extruder (no wait)\nM106 S50 ; 50% fan to prevent hotend jam\nM190 S[first_layer_bed_temperature] ; Bed (wait)\nM109 S[first_layer_temperature]; Extruder (wait)\nG28 ; Home to top 3 endstops\nG4 P1000\nG29 Z+0
toolchange_gcode = 
use_firmware_retraction = 0
use_relative_e_distances = 0
use_volumetric_e = 0
vibration_limit = 0
wipe = 1
z_offset = 0
------------------------------------------------

If you use the default settings, Preview display normal.

lordofhyphens, 2017-04-03:
Looks like it's crashing after the infill generates; the perimeters are fine though.

Confirmed in 90623c4

lordofhyphens, 2017-04-03:
Raised exception:
````
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted (core dumped)
````

Backtrace:
````
#0  0x00007f22f243b067 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
#1  0x00007f22f243c448 in __GI_abort () at abort.c:89
#2  0x00007f22f06aeb3d in __gnu_cxx::__verbose_terminate_handler() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007f22f06acbb6 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007f22f06acc01 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007f22f06ace19 in __cxa_throw () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007f22f07027c2 in std::__throw_bad_alloc() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007f22f0bb4da1 in std::vector<Slic3r::Point, std::allocator<Slic3r::Point> >::vector(std::vector<Slic3r::Point, std::allocator<Slic3r::Point> > const&) () from /home/lenox/test/Slic3r/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/Slic3r/XS/XS.so
#8  0x00007f22f0c8e225 in Slic3r::FillRectilinear::_fill_single_direction(Slic3r::ExPolygon, std::pair<float, Slic3r::Point> const&, long, std::vector<Slic3r::Polyline, std::allocator<Slic3r::Polyline> >*) ()
   from /home/lenox/test/Slic3r/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/Slic3r/XS/XS.so
#9  0x00007f22f0c8e8b8 in Slic3r::FillRectilinear::_fill_surface_single(unsigned int, std::pair<float, Slic3r::Point> const&, Slic3r::ExPolygon&, std::vector<Slic3r::Polyline, std::allocator<Slic3r::Polyline> >*) ()
   from /home/lenox/test/Slic3r/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/Slic3r/XS/XS.so
#10 0x00007f22f0cbcb76 in Slic3r::Fill::fill_surface(Slic3r::Surface const&) ()
   from /home/lenox/test/Slic3r/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/Slic3r/XS/XS.so
#11 0x00007f22f0c246a7 in Slic3r::LayerRegion::make_fill() ()
   from /home/lenox/test/Slic3r/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/Slic3r/XS/XS.so
#12 0x00007f22f0c1c624 in Slic3r::Layer::make_fills() ()
   from /home/lenox/test/Slic3r/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/Slic3r/XS/XS.so
#13 0x00007f22f0cb6c03 in void Slic3r::_parallelize_do<Slic3r::Layer*>(std::queue<Slic3r::Layer*, std::deque<Slic3r::Layer*, std::allocator<Slic3r::Layer*> > >*, boost::mutex*, boost::function<void (Slic3r::Layer*)>) ()
   from /home/lenox/test/Slic3r/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/Slic3r/XS/XS.so
#14 0x00007f22f0cb6b0d in boost::detail::thread_data<boost::_bi::bind_t<void, void (*)(std::queue<Slic3r::Layer*, std::deque<Slic3r::Layer*, std::allocator<Slic3r::Layer*> > >*, boost::mutex*, boost::function<void (Slic3r::Layer*)>), boost::_bi::list3<boost::_bi::value<std::queue<Slic3r::Layer*, std::deque<Slic3r::Layer*, std::allocator<Slic3r::Layer*> > >*>, boost::_bi::value<boost::mutex*>, boost::_bi::value<boost::function<void (Slic3r::Layer*)> > > > >::run() () from /home/lenox/test/Slic3r/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/Slic3r/XS/XS.so
#15 0x00007f22f0cd2725 in thread_proxy () from /home/lenox/test/Slic3r/local-lib/lib/perl5/x86_64-linux-thread-multi/auto/Slic3r/XS/XS.so
#16 0x00007f22f2ef5064 in start_thread (arg=0x7f22d0ff9700) at pthread_create.c:309
#17 0x00007f22f24ee62d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
````