slic3r/Slic3r #2943: Slic3r 1.2.9 crash when changing infill characteristics on multi part assembly

michaelr123, 2015-06-22:
So I've been playing with the last few versions of slic3r working with multi-part assemblies and I've found an issue with a part that we print at work (1 base piece with 4 sub parts inside it that change extruders and also varies density and pattern of the infill). 

The issues arises when you have a value other than 0 for "perimeters" in the "advanced" tab under "print settings". If you assemble the parts within the base part (double click on part, select "load parts") and then procceed to change TWO of the pieces infill densities or patterns it will pull a warning:

Argument "\x{31}\x{30}..." isn't numeric in numeric eq (==) at /home/ekuzmenko/Downloads/Slic3r/lib/std/Slic3r/Layer.pm line 53.

This occurs on windows and linux, I had my co-worker try it on his linux box and it pulls the exact same warning. It seems to be that if you make any changes to infill on the pieces in the subassembly windows it will cause this warning. I tried just having two pieces on the bed and varying their infill types/densities and it didn't cause the problem. 

This may have something to do with how the perimeters of the two differerent parts meet, but I'm not totally sure about that.

lordofhyphens, 2015-06-22:
Could you post a copy of your slic3r settings and some part that you can
get the same issue on (that you can share)?

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

michaelr123, 2015-06-26:
Sorry I got busy, here are the stl's and the slic3r settings used: 

```
# generated by Slic3r 1.2.9 on Fri Jun 26 08:11:18 2015
avoid_crossing_perimeters = 0
bed_shape = 0x0,1000x0,1000x1000,0x1000
bed_temperature = 0
before_layer_gcode = 
bottom_solid_layers = 3
bridge_acceleration = 0
bridge_fan_speed = 100
bridge_flow_ratio = 1
bridge_speed = 10
brim_width = 0
complete_objects = 0
cooling = 1
default_acceleration = 0
disable_fan_first_layers = 3
dont_support_bridges = 0
duplicate_distance = 6
end_gcode = T0\nG28 X0 Y0 F2500\nM104 S0 T0 ; turn off heaters\nM104 S0 T1\nM84 ; disable motors\nT0\nM300\nM300
external_fill_pattern = rectilinear
external_perimeter_extrusion_width = 0
external_perimeter_speed = 50%
external_perimeters_first = 0
extra_perimeters = 1
extruder_clearance_height = 25
extruder_clearance_radius = 70
extruder_offset = 0x0,0x0
extrusion_axis = E
extrusion_multiplier = 1,1
extrusion_width = 0.7
fan_always_on = 1
fan_below_layer_time = 60
filament_colour = #FFFFFF;#FFFFFF
filament_diameter = 2.85,2.85
fill_angle = 45
fill_density = 0%
fill_pattern = honeycomb
first_layer_acceleration = 0
first_layer_bed_temperature = 0
first_layer_extrusion_width = 0
first_layer_height = 110%
first_layer_speed = 50%
first_layer_temperature = 215,215
gap_fill_speed = 20
gcode_arcs = 0
gcode_comments = 0
gcode_flavor = reprap
infill_acceleration = 0
infill_every_layers = 1
infill_extruder = 1
infill_extrusion_width = 0
infill_first = 0
infill_only_where_needed = 0
infill_overlap = 20%
infill_speed = 50
interface_shells = 0
layer_gcode = \n; layer : [layer_num]\n; [layer_z]
layer_height = 0.36
max_fan_speed = 100
max_print_speed = 80
max_volumetric_speed = 0
min_fan_speed = 65
min_print_speed = 20
min_skirt_length = 0
notes = 
nozzle_diameter = 0.6,0.6
octoprint_apikey = 
octoprint_host = 
only_retract_when_crossing_perimeters = 0
ooze_prevention = 0
output_filename_format = [input_filename_base].gcode
overhangs = 1
perimeter_acceleration = 0
perimeter_extruder = 1
perimeter_extrusion_width = 0
perimeter_speed = 50
perimeters = 1
post_process = 
pressure_advance = 0
raft_layers = 0
resolution = 0
retract_before_travel = 2,2
retract_layer_change = 1,1
retract_length = 2,2
retract_length_toolchange = 10,5
retract_lift = 0.25,0.25
retract_restart_extra = 0,0
retract_restart_extra_toolchange = 0,0
retract_speed = 10,10
seam_position = nearest
skirt_distance = 20
skirt_height = 1
skirts = 2
slowdown_below_layer_time = 20
small_perimeter_speed = 30
solid_infill_below_area = 70
solid_infill_every_layers = 0
solid_infill_extruder = 1
solid_infill_extrusion_width = 0
solid_infill_speed = 60
spiral_vase = 0
standby_temperature_delta = -5
start_gcode = T0\nG28\nG1 Z10 F2500\nG1 X400 F2500\nG1 Y400 F2500\n\n
support_material = 0
support_material_angle = 0
support_material_contact_distance = 0.2
support_material_enforce_layers = 10
support_material_extruder = 1
support_material_extrusion_width = 0
support_material_interface_extruder = 1
support_material_interface_layers = 3
support_material_interface_spacing = 1.5
support_material_interface_speed = 100%
support_material_pattern = honeycomb
support_material_spacing = 2.5
support_material_speed = 60
support_material_threshold = 45
temperature = 215,215
thin_walls = 1
threads = 6
toolchange_gcode = T[next_extruder]\nG92 E0\nG0 E11\nG92 E0\nT[previous_extruder]\nG92 E0\nG0 E-10\nG92 E0\n
top_infill_extrusion_width = 0
top_solid_infill_speed = 50
top_solid_layers = 0
travel_speed = 130
use_firmware_retraction = 0
use_relative_e_distances = 0
use_volumetric_e = 0
vibration_limit = 0
wipe = 1,1
xy_size_compensation = 0
z_offset = 0
```
here's a link to the stl's. let me know if they dont work 
https://www.dropbox.com/sh/648h4191n12in8n/AADIEI4ErJSUBURyg9iViYeOa?dl=0

bbriggstkd, 2015-10-31:
Has this been resolved.  I get crashes too on both the 32 and 64 bit versions.

zsserg, 2017-03-06:
I'm getting the same error while trying to apply modifying mesh to a part. I'm adding a second STL as a modifier mesh, click green plus icon to add settings, add Infill %, set it to "100%" and click OK. At this point I'm getting the same error message & application hangs.

lordofhyphens, 2017-03-28:
@zsserg can you add a test case for this (with the STLs you are trying to add) and the initial print settings?

I just ran a test with [the bunny](http://www.thingiverse.com/download:37337) and a 10mm slab modifier mesh. Works. 

[image]
[image]

You can export AMF files with the modifier meshes intact, so if you have a test case where you are having problems (http://dl.slic3r.org/dev) please set up the modifier mesh w/o the options and then export it.

If you turn off background processing it probably won't hang so you can do the export of the AMF with the problematic combination.
[bunny-100fill-rect.zip](https://github.com/alexrj/Slic3r/files/876487/bunny-100fill-rect.zip)

I'll reopen if you provide a test case.

closed 2017-03-28 by lordofhyphens