slic3r/Slic3r #2945: With autospeed on and either brim or skirt (or both) present, the first layer is too fast.

hacker, 2015-06-22:
I have tried the autospeed feature (in the 1.2.9 Slic3r) and it turns out that it prints the first layer at a higher speed if either brim or skirt or both are enabled. I reproduced it with the 0.5mm-thin-wall STL.

Here are some fragments of G-code generated. Courtesy of `grep -e Z0.24 -e Z0.48 -e F 0.5mm-thin-wall.gcode`. Layer height is 0.24.

This one is without skirt and brim:

```gcode
G1 Z0.240 F7800.000
G1 E-2.00000 F1200.00000
G1 Z0.315 F7800.000
G1 X96.376 Y89.989 F7800.000
G1 Z0.24 F7800.000
G1 E2.00000 F1200.00000
G1 X96.395 Y89.794 E2.00349 F1044.685
G1 Z0.480 F7800.000
G1 X96.392 Y89.805 E3.55976 F1044.683
```

The first reasonable F (1044.685) appears after Z0.24

This one with a skirt:

```gcode
G1 Z0.240 F7800.000
G1 E-2.00000 F1200.00000
G1 Z0.315 F7800.000
G1 X92.167 Y84.821 F7800.000
G1 Z0.240 F7800.000
G1 E2.00000 F1200.00000
G1 X92.321 Y84.667 E2.00388 F2564.144
G1 X118.256 Y82.604 E2.47560 F2564.144
G1 E2.25530 F1200.00000
G1 Z0.315 F7800.000
G1 X96.376 Y89.989 F7800.000
G1 Z0.240 F7800.000
G1 E2.00000 F1200.00000
G1 X96.395 Y89.794 E2.00349 F2564.144
G1 Z0.480 F7800.000
G1 X96.392 Y89.805 E3.55976 F1044.683
```

The first 1044 rate is at Z0.48, with a nice F2564 before that. The same and brim:

```gcode
G1 Z0.240 F7800.000
G1 E-2.00000 F1200.00000
G1 Z0.315 F7800.000
G1 X90.752 Y83.406 F7800.000
G1 Z0.240 F7800.000
G1 E2.00000 F1200.00000
G1 X90.906 Y83.252 E2.00388 F4800.000
G1 X118.431 Y80.612 E2.50665 F4800.000
G1 E2.47886 F1200.00000
G1 Z0.315 F7800.000
G1 X92.211 Y84.865 F7800.000
G1 Z0.240 F7800.000
G1 E2.00000 F1200.00000
G1 X92.554 Y84.521 E2.00865 F4800.000
G1 X118.518 Y82.690 E2.47954 F4800.000
...
G1 Z0.480 F7800.000
G1 X96.392 Y89.805 E66.64692 F1044.683
```

Nothing less that 4800 before Z0.48.

I don't know if it's a bug or a feature, I'm relatively new to the whole thing, but it does look like a bug to me? Alternatively, I'm doing something wrong to trigger this behaviour. The configuration (from .gcode tail) looks like this:

```
; avoid_crossing_perimeters = 1
; bed_shape = 0x0,215x0,215x200,0x200
; bed_temperature = 0
; before_layer_gcode =
; bridge_acceleration = 0
; bridge_fan_speed = 100
; brim_width = 8
; complete_objects = 0
; cooling = 1
; default_acceleration = 0
; disable_fan_first_layers = 0
; duplicate_distance = 6
; end_gcode = <killed off to shorten line>
; extruder_clearance_height = 20
; extruder_clearance_radius = 20
; extruder_offset = 0x0
; extrusion_axis = E
; extrusion_multiplier = 0.947
; fan_always_on = 1
; fan_below_layer_time = 60
; filament_colour = #FFFFFF
; filament_diameter = 1.75
; first_layer_acceleration = 0
; first_layer_bed_temperature = 0
; first_layer_extrusion_width = 100%
; first_layer_speed = 30%
; first_layer_temperature = 210
; gcode_arcs = 0
; gcode_comments = 0
; gcode_flavor = reprap
; infill_acceleration = 0
; infill_first = 0
; layer_gcode =
; max_fan_speed = 100
; max_print_speed = 80
; max_volumetric_speed = 0
; min_fan_speed = 100
; min_print_speed = 10
; min_skirt_length = 0
; notes =
; nozzle_diameter = 0.4
; only_retract_when_crossing_perimeters = 1
; ooze_prevention = 0
; output_filename_format = [input_filename_base].gcode
; perimeter_acceleration = 0
; post_process =
; pressure_advance = 0
; resolution = 0
; retract_before_travel = 1.5
; retract_layer_change = 0
; retract_length = 2
; retract_length_toolchange = 0
; retract_lift = 0.075
; retract_restart_extra = 0
; retract_restart_extra_toolchange = 0
; retract_speed = 20
; skirt_distance = 6
; skirt_height = 1
; skirts = 1
; slowdown_below_layer_time = 5
; spiral_vase = 0
; standby_temperature_delta = -5
; start_gcode = <killed of to shorten line>
; temperature = 210
; threads = 5
; toolchange_gcode =
; travel_speed = 130
; use_firmware_retraction = 0
; use_relative_e_distances = 0
; use_volumetric_e = 0
; vibration_limit = 0
; wipe = 0
; z_offset = 0
; dont_support_bridges = 1
; extrusion_width = 0
; first_layer_height = 0.24
; infill_only_where_needed = 0
; interface_shells = 0
; layer_height = 0.24
; raft_layers = 0
; seam_position = aligned
; support_material = 0
; support_material_angle = 0
; support_material_contact_distance = 0.2
; support_material_enforce_layers = 0
; support_material_extruder = 1
; support_material_extrusion_width = 0
; support_material_interface_extruder = 1
; support_material_interface_layers = 0
; support_material_interface_spacing = 0
; support_material_interface_speed = 100%
; support_material_pattern = rectilinear
; support_material_spacing = 2.5
; support_material_speed = 0
; support_material_threshold = 0
; xy_size_compensation = 0
; bottom_solid_layers = 3
; bridge_flow_ratio = 1
; bridge_speed = 0
; external_fill_pattern = rectilinear
; external_perimeter_extrusion_width = 0
; external_perimeter_speed = 0
; external_perimeters_first = 1
; extra_perimeters = 1
; fill_angle = 45
; fill_density = 5%
; fill_pattern = line
; gap_fill_speed = 0
; infill_every_layers = 2
; infill_extruder = 1
; infill_extrusion_width = 0
; infill_overlap = 15%
; infill_speed = 0
; overhangs = 1
; perimeter_extruder = 1
; perimeter_extrusion_width = 0
; perimeter_speed = 0
; perimeters = 3
; small_perimeter_speed = 0
; solid_infill_below_area = 70
; solid_infill_every_layers = 0
; solid_infill_extruder = 1
; solid_infill_extrusion_width = 0
; solid_infill_speed = 0
; thin_walls = 1
; top_infill_extrusion_width = 0
; top_solid_infill_speed = 0
; top_solid_layers = 3
```

xoan, 2015-06-22:
I think it is not related to either brim or skirt, but only to first layer speed when it is expressed as a percentage.

#2810 [(comment)](https://github.com/alexrj/Slic3r/issues/2810#issuecomment-112601365)

hacker, 2015-06-22:
Thanks for the reference. I'm still processing the information, but I wouldn't dismiss skirt/brim link completely as there's clearly a correlation. Also this got my attention — `I would make it optional and default to calculating it as the maximum one given by the combination of Max Print Speed and the thinnest extrusion width used in the print`. Maybe this is how  skirt/brim influence it?

But, as I said, I'm still processing the information. The most reasonable thing to do is to look into code, but last time I tried to make it work I lost my fight with perl, cpan and macports and I don't want to look at the code that I can't sensibly change :) Need to try again… If I get time…

luisonoff, 2015-07-18:
I also experienced this today while trying autospeed. First layer was much faster than my 30mm/s setting for first layer. Any news or ideas?

Praesmeodymium, 2015-07-26:
I find Autospeed currently ignores % for first layer, but honors hard numbers like 30mms.

First layer set to 45mms with autospeed (ignore the title I put 30mms in incorrectly)
http://pastebin.com/ii2ygeMr

First layer set to 50% with autospeed and a later layer to show its about the same speed as the first layer

http://pastebin.com/eNa2tFMY

alexrj, 2017-03-04:
Thank you for the bug report. I fixed this. First layer speed was not applied over autospeed when set as percent.

closed 2017-03-04 by alexrj