slic3r/Slic3r #2948: Artifacts and inconsistent layers

alexrj, 2015-06-23:
Reported by @napter:

I found a similar issue where all the layers are not consistent. I'm not sure if it's the same issue, this is a weird one. Slice the model and it prints with the issues in the top right picture - missing parts of layers and extra blobs in the GCode. Then rotate the model, as you can see the orientation in the bottom set is rotated 180 degrees, and the issues go away and it slices and prints perfectly. [config file](https://drive.google.com/file/d/0B57Du2WjL1iqX0Q0MHJBYlA5RDQ/view?usp=sharing) and [STL file](https://drive.google.com/file/d/0B57Du2WjL1iqejVwSzNaMmFSdVk/view?usp=sharing)
[image]

DanielJoyce, 2016-03-17:
Shouldn't there be yellow top fill on top of the pink areas too?

lordofhyphens, 2016-03-17:
Yellow is always perimeters.

alexrj, 2016-12-21:
This issue is still present as of current master and can be reproduced with default settings.

alexrj, 2016-12-21:
Interesting. A few edges are missing from the Voronoi diagram:

[image]

alexrj, 2016-12-21:
Simpler model that reproduces the issue: [fence_cut.stl.zip](https://github.com/alexrj/Slic3r/files/667067/fence_cut.stl.zip)
SVG file of the Voronoi diagram of last layer (default settings): [voronoi.svg.zip](https://github.com/alexrj/Slic3r/files/667070/voronoi.svg.zip)

alexrj, 2016-12-21:
Reported to the upstream maintainers: https://svn.boost.org/trac/boost/ticket/12707
I'm afraid Boost Voronoi is unmaintained. I wish @asydorchuk could take a look at those two pending bugs...

lordofhyphens, 2016-12-21:
I can look at its usage and perhaps converting to a different library? My quick search turned up a couple other OS voronoi tesselation implementations: http://math.lbl.gov/voro++/ is one and it's BSD-licensed.

There's been some significant changes as well to the Boost build system since 1.60 IIRC, at least I had problems getting it built on Win32, I think that either the syntax changed or that building with mingw broke at some point. I will have to look at it again though to be sure.

alexrj, 2016-12-21:
That Voro++ looks even less maintained :) and it's 3D so I'm not sure it fits our environment.

I actually like Boost Voronoi because it's numerically robust, fast, nicely coded and documented, and uses our native types thanks to traits. And so far I had never found a bug :-(

bubnikv, 2020-05-27:
The simplest polygon with nice rounded numbers to evoke such a boost::polygon::voronoi issue is the following, which I distilled from the example provided by @alranel 

        {       0, 10000000},
        {  700000,        1}, // it has to be 1, higher number, zero or -1 work.
        {  700000,  9000000},
        { 9100000,  9000000},
        { 9100000,        0},
        {10000000, 10000000}

There is not much in the documentation about that, but the voronoi library does not ensure topological correctness due to rounding of the Voronoi circle events, see
https://www.youtube.com/watch?v=80kDK0VwBP4

bubnikv, 2020-05-28:
I have created another issue with the minimal test case.
https://github.com/boostorg/polygon/issues/43

Interestingly enough, the library seems to be pretty stable, the only other real issue I was able to find on the Internet is https://svn.boost.org/trac10/ticket/12067
There may be no Voronoi edges produced for the 1st and last sweep line position, which is not really an issue for Slic3r (maybe for the path planner, not for the gap fill).
Other issues reported are due to intersections in the input data.

asydorchuk, 2020-05-28:
Hi, I am going to have a look at this issue this weekend, somehow missed the initial reference back in 2016.