slic3r/Slic3r #2949: v. 1.2.9 "Pillar" Support Type Generates Extrusions Floating In Space

rf1k-mjh11, 2015-06-23:
I noticed this issue on v.1.2.7, too, but nevert reported it.
a) When using pillars as the support pattern, Slic3r will occasionally create the inverted pyramid portions of the support without the pillar - so the pyramid(s) seem(s) to float in space and cannot be printed.
[image]
b) The new algorithm for the pyramid portion regularily creates extrusions floating in space.  Every time the inverted, stepped pyramid gets wider, a portion of the extrusion has nothing underneath.
[image]

The previous stable version, v.1.1.7, used a different approach.
[image]
Here, in 1.1.7, the inverted pyramids are represented by outlines.  The newer versions use a stepped raft-type approach.

mjh11

rf1k-mjh11, 2015-06-23:
Update:  Sorry, the free-floating support structures also show up in version 1.1.7 using the exact same STL-file.  The picture above was from an almost identical part - unfortunately only almost.
[image]

mjh11

lordofhyphens, 2016-08-21:
I think what is going on here w/r/t the stepping is that the overhang is calculated and then the raft pattern is put in place. 

The floating support islands are interesting; could you check in 1.3.0 ([win32 development build](https://dl.bintray.com/lordofhyphens/Slic3r/:slic3r-add-libglut-dll--566bbbd.zip)) and see if the problem still exists (and provide a STL file and config where you see this problem)?

rf1k-mjh11, 2016-08-21:
Joseph/Alex,

I downloaded the 1.3.0 development version 
(slic3r-master--566bbbd.zip).  I added a part I've had issues with in 
the past.  The results look much better (coincidence or not?), but there 
are still a few support islands floating in the air.
The part I loaded is the active cooling fan for my Pico hot end.
Look at the layers starting from 15.60 to about 20.50.  Look in the area 
where the air output is, near the top.  There are several small pieces 
of support that start in the middle of the air.

I guess the lines may change slightly, depending on the location you 
move the object to.

The Command window in the background spit out a few error messages, too:
*
**Caught exception at option bed_size.
Caught exception at option duplicate.
Caught exception at option duplicate_grid.
Caught exception at option g0.
Caught exception at option print_center.
Caught exception at option randomize_start.
Caught exception at option rotate.
Caught exception at option scale.
Caught exception at option solid_fill_pattern.
Caught exception at option start_perimeters_at_concave_points.
Caught exception at option start_perimeters_at_non_overhang.
Caught exception at option solid_fill_pattern.*

I tried two completely different iterations of my fan funnel.  Both 
showed at least one floating support island, too.
I don't know if all the error messages came from the first part I loaded 
(the STL supplied), or if they cropped up with subsequent part trials.

Here is my Config-bundle and the respective STL.

I hope this is of use to you,

rf1k-mjh11

# generated by Slic3r 1.3.0-dev on Sun Aug 21 19:40:53 2016

[filament:Mndl_PLA_200_D5_L0,12]
bed_temperature = 42
bridge_fan_speed = 100
cooling = 1
disable_fan_first_layers = 3
extrusion_multiplier = 0.96
fan_always_on = 1
fan_below_layer_time = 60
filament_diameter = 2.95,7
first_layer_bed_temperature = 54
first_layer_temperature = 222
max_fan_speed = 100
min_fan_speed = 40
min_print_speed = 9
slowdown_below_layer_time = 20
temperature = 212

[filament:Mndl_PLA_200_D5_L0,2]
bed_temperature = 42
bridge_fan_speed = 100
cooling = 1
disable_fan_first_layers = 3
extrusion_multiplier = 0.96
fan_always_on = 1
fan_below_layer_time = 60
filament_diameter = 2.95,7
first_layer_bed_temperature = 54
first_layer_temperature = 222
[... 2351 lines cut ...]
solid_infill_extrusion_width = 0.6
solid_infill_speed = 27
spiral_vase = 0
standby_temperature_delta = -5
support_material = 0
support_material_angle = 20
support_material_contact_distance = 0.2
support_material_enforce_layers = 0
support_material_extruder = 1
support_material_extrusion_width = 0.265
support_material_interface_extruder = 1
support_material_interface_layers = 2
support_material_interface_spacing = 0.4
support_material_interface_speed = 100%
support_material_pattern = rectilinear
support_material_spacing = 3
support_material_speed = 36
support_material_threshold = 35
thin_walls = 1
threads = 3
top_infill_extrusion_width = 0.6
top_solid_infill_speed = 23
top_solid_layers = 3
travel_speed = 150
xy_size_compens

lordofhyphens, 2016-08-21:
You can drag/drop zip files ;)

The exceptions can be safely ignored with regards to configuration options.
They just mean that there are configuration options that are not understood
by Slic3r  (old options that got renamed or changed). You can remove them
if you like to get rid of the warnings.

lordofhyphens, 2016-08-21:
File->Export Config will give you a single ini file to upload.

lordofhyphens, 2016-08-21:
@rf1k-mjh11 Looks like you uploaded a config bundle, which is everything you have, not the config that you're seeing an issue with.

lordofhyphens, 2016-08-26:
@rf1k-mjh11 Attachments won't let you upload files, you have to use the Github web page.

lordofhyphens, 2016-08-27:
While messing around with another issue, I managed to reproduce this issue on another model in such a way that shows what's going on. 

[image]

The pillars generation logic is making a "pillar" top that covers the "unsupported areas" but when the intersection of the areas needed to be supported with the actual support structures generated is made, those areas are actually unsupported. 

Reproduction steps (on 1.3.0-dev):
Edit lib/Slic3r/Print/SupportMaterial.pm, line 25 to read:
`use constant PILLAR_SPACING => 25;` (10->25). 

STL/Configuration:
[2949.zip](https://github.com/alexrj/Slic3r/files/440783/2949.zip)

lordofhyphens, 2016-08-27:
[image]

This is the result of a hack that uses the pillars as they are without any clipping. One solution that I can think of is to generate an outer shell of the contact area (as a whole) and take the intersection. 

I'm trying to figure out a way to structure the boolean operations so that we still get the benefits of clipping while avoiding the bug.

lordofhyphens, 2016-08-27:
The way that Slic3r generates pillar supports is that it fills in the volume of "all areas that need support" and then takes the intersection of that area and what got produced for the grid. This is what is causing the bug: parts of the pillars are being removed and Slic3r assumes that those islands got covered.

rf1k-mjh11, 2016-09-06:
Sorry folks.  I messed up with the config bundle.

So here is the config setup I just used to slice:
[Slic3r_config_bundle_Pillar_Support_Error.ini.txt](https://github.com/alexrj/Slic3r/files/457415/Slic3r_config_bundle_Pillar_Support_Error.ini.txt)

And here's an STL where the issue occurs (here it starts at layer height 15.60):
[40mmFanFunnel_v8ba_.zip](https://github.com/alexrj/Slic3r/files/457398/40mmFanFunnel_v8ba_.zip)
Here's a screen shot of the condition (close-up and overview):
[image]
[image]

Version used: 1.3.0-dev

mjh11

imarin2, 2017-03-18:
Hi!

I have just come across a similar situation using today's git code:

```
commit 76f60471afb5a5fc378065d8266b59cc71937ddc
Author: Alessandro Ranellucci <aar@cpan.org>
Date:   Sat Mar 18 19:51:07 2017 +0100

    Moved typedefs to FillRectilinear.cpp

```

The STL (V7_HEAD.stl):
http://www.thingiverse.com/download:3456504

The Configuration:

[slic3r_latest2_slic3rbugreport.ini.zip](https://github.com/alexrj/Slic3r/files/852812/slic3r_latest2_slic3rbugreport.ini.zip)

Description:
Three cols of pillars are created mid print on thin air.

The issue in a picture:
[image]

I would like to thank you for fixing the other issue I opened a time ago (part top collapsing). Now it is closed and I did not want to reopen it just to thank you. So, a big thank you here.

lordofhyphens, 2017-03-19:
You can post in an issue without reopening it.​

lordofhyphens, 2017-03-19:
@imarin2 some of the syntax for support generation changed recently. 0 is no longer `auto`, use a % value (we default it to 60% now).

imarin2, 2017-03-19:
@lordofhyphens 

Posted there. Thanks. I did not know.

That support generation parameter that changed, is it the threshold?

```
support_material = 1
support_material_angle = 0
support_material_contact_distance = 0.2
support_material_enforce_layers = 0
support_material_extruder = 1
support_material_extrusion_width = 0
support_material_interface_extruder = 1
support_material_interface_layers = 3
support_material_interface_spacing = 0
support_material_interface_speed = 100%
support_material_pattern = pillars
support_material_spacing = 2.5
support_material_speed = 60
support_material_threshold = 60%
```
There I use 60%.

lordofhyphens, 2017-03-19:
Yeah it was threshold. Just wanted to be sure.