slic3r/Slic3r #2950: Can we get tool-specific tool change gcode?

nicksears, 2015-06-23:
I've posted an issue about this before (issue 2675) where I requested to be able to modularly invoke custom gcode before/after a certain number of layer/tool/print changes.

While I'd still love to see this level of sophistication, and we've recently seen before/after layer change sections added, we really need the ability to have custom g-code before/after switching to a specific tool.

I have different hot swappable tools and I need to calibrate them each time I switch to one. I've come up with two solutions: 1. Calculate the equivalent filament diameter (calculate cross sectional diameter*calibration and make them equal, back calculate filament diameter - or if they're close, just fudge the ratio of the CSA with the extrusion multiplier) 2. Use a post-processing program (my favorite is "find and replace" "fnr.exe" (https://findandreplace.codeplex.com/) to replace each T1 statement with T1;
M92 EXXX;

I realize some firmwares have different extruder calibrations (repetier) but marlin (mine at least) does not, and I'd rather not reflash the firmware when I decide I want to use T1 in the T0 spot and vice versa.

Where can I go about  helping make these changes? I've done some coding, so I'd love to try and help. Very little python so far, but I can try and have some friends who can.

Nick

phizz166, 2015-06-25:
First of all, +1 to this. I could really use different G-codes on switching from one tool to the next, as I have two different purge/wipe zones at either side of the bed, for both extruders, and right now the code doesn't let me intelligently select the proper side according to which tool is becoming active. 

Ultimately the best thing would be to have proper conditional statements in the toolchange/layer change G-code -- if [next_extruder] == 0 { } else { } ... but having at least a T0 -> T1 mode and T1 -> T0 would be a good start.

Second, to just address your specific problem -- my machine also has two different step-rates for the two extruders (one is geared 3mm, the other direct-drive 1.75mm). I've solved this by running  https://github.com/MagoKimbra/MarlinKimbra, which works very well.

pyrophreek, 2015-06-25:
I agree with this as well, and I would suggest the tool change custom gcode be moved to the tool section of the printer settings, rather than in the custom gcode section.  This way however many tools you have, you can configure the before and after code for each one independently and intelligently.

simonkuehling, 2015-12-11:
Nice to find this issue, i was about writing a similar request.

Extruder-specific toolchange GCODE would be a marvel for those that do their own toolchange logic rather than letting slic3r do it. My example case is the Nozzle Purging/Cleaning system of our HT500 3D printer (like seen here: https://www.youtube.com/watch?v=G4HWzoBPA7Y)
For several material combinations it would be best to use different spots per extruder on the silicone lip to avoid mixing in leftovers from the other nozzle while wiping. kind of the same as having different purging areas like @phizz166.
Also for materials that need different purging parameters or wiping moves, this could be useful to further improve reliability and of course print quality resulting from this.

lordofhyphens, 2016-07-04:
Should be solvable with the conditional gcode syntax being worked on, right?

simonkuehling, 2016-08-22:
@lordofhyphens - a conditional syntax is all that we need here, i agree!

nicksears, 2016-08-22:
placeholder values would work, but it could get messy really quick. We
already have too many custom g-code sections for most people (most people
only use start and end). Why not make it configurable?

I'll also add here, but I'm about to make a separate post, but we really
need a few other similar things, too: support for multiple fans. Marlin
supports it now with "M106 P2 Sxxx", etc and we need slic3r to actually
obey the filament cooling profile for those fans
Also, we really need per-section post processing scripts; scripts
associated with the printer or filament settings, not just the print
settings.

hexxter, 2019-03-31:
a solution for tool changer is a good feature. Have anyone found a solution?

lordofhyphens, 2019-04-01:
@hexxter we have conditional gcode now.

hexxter, 2019-04-01:
@lordofhyphens is a docu avalible?

simonkuehling, 2019-04-01:
@hexxter, have a look at https://manual.slic3r.org/advanced/conditional-gcode for a documentation on conditional gcode syntax!