slic3r/Slic3r #2951: Support interface layers: purpose of zig-zag-perimeter?

jonaskuehling, 2015-06-24:
Hi Alex,

when configuring support with interface layers, the topmost interface layer will be generated with this zig-zag perimeter and I don't really see the purpose for it... For a while I though it should help supporting model perimeters especially on sparse support, in order to help like small hole perimeters of the supported model not to fall down between the support pattern spacing for example.

Now I realized that the zig-zag only appears with at least one interface layer set, and furthermore those zig-zag patterns are never really aligned below model perimeters, but just represent the support area perimeter.

If you could give me a little clue about the general purpose of those zig-zag lines, that would be very much appreciated ;-)

Currently I feel, that especially with soluble support (z distance = 0) the zig-zag doesn't provide a significant advantage over pure solid interface layers just without zig-zag. I'd even say it produces avoidable inhomogeneity on supported bottom model surfaces. I'm curious about the intended zig-zag benefits - otherwise I may suggest deactivating these at least for soluble support...?

lordofhyphens, 2016-09-20:
I think it's just a quick way to cover the same area/width with low amounts of material.

Martronic-SA, 2017-06-05:
I think a checkbox for desactivating those zigzag would be welcome, especially with soluble support.
Those zig zag are more difficult to print and tend to decrease supported part quality.

samibaskir, 2018-01-29:
Agreed, when you use soluble support, it is best to align the interface layer pattern with the model boundary