slic3r/Slic3r #2953: OpenGL.so not found on line 157 in 1.2.9 on Ubuntu 14.04 LTS 64bit (wroks in 1.1.7)

bmjbmj, 2015-06-24:
Hello!

I downloaded 1.2.9 and upgraded from 1.1.7 oddly enough 1.2.9 do not find 'OpenGL.so' even that the path is correct and the file exist? I have double and triple checked that 'OpenGL.so' has the right access-rights and and even tried to run as sudo. I rolled back to 1.1.7 and it works. Maybe there is an error in the error message, and some thing else is wrong???  I'm using Ubuntu 14.04 LTS 64bit.
DUMP--->
bmj@bmj-X202E:~/Slic3r/bin$ ./slic3r --gui
Can't load '/home/bmj/Slic3r/lib/vrt/9e109cfe461eb9a764d58156ba7bc844/OpenGL.so' for module OpenGL: libglut.so.3: kan inte öppna delad objektfil: Filen eller katalogen finns inte at <content>/DynaLoader.pm line 157.
 at /home/bmj/Slic3r/lib/std/Slic3r/GUI/Plater/2DToolpaths.pm line 120
Compilation failed in require at /home/bmj/Slic3r/lib/std/Slic3r/GUI/Plater/2DToolpaths.pm line 120.
BEGIN failed--compilation aborted at /home/bmj/Slic3r/lib/std/Slic3r/GUI/Plater/2DToolpaths.pm line 120.
Compilation failed in require at /home/bmj/Slic3r/lib/std/Slic3r/GUI.pm line 16.
BEGIN failed--compilation aborted at /home/bmj/Slic3r/lib/std/Slic3r/GUI.pm line 16.
Compilation failed in require at (eval 71) line 1.

lordofhyphens, 2015-06-24:
> katalogen finns inte at /DynaLoader.pm line 157.

Ah, another Ubuntu user. You should probably delete the libopengl that
comes with Slic3r, it's been known to have problems.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

xoan, 2015-06-24:
Actually libGL, libGLU and libglut.

#2535

alexrj, 2015-06-24:
But they were removed as of 1.2.9...

t-paul, 2015-06-25:
@bmjbmj try installing the `freeglut3` package from your distribution.

bmjbmj, 2015-06-25:
Thank you all!

I tried  t-paul:s solution first and it worked.

I like the 3D view and the new option for a connection to OctoPrint in 1.2.9!

lordofhyphens, 2015-06-25:
I will add freeglut3 to my package reqs then

alexrj, 2015-06-25:
Good!  I updated the http://slic3r.org/download page with freeglut.

closed 2015-06-25 by alexrj