slic3r/Slic3r #2955: 1.2.9 command line can't find paths with non ASCII characters

afolgado, 2015-06-27:
Hi, I've come across this Issue:

When running the following command to slice a RepetierHost composition:

`slic3r.exe --load "C:\Users\Andrés\AppData\Local\RepetierHost\slic3r.ini" --print-center 93,100 -o "C:\Users\Andrés\AppData\Local\RepetierHost\composition.gcode" "C:\Users\Andrés\AppData\Local\RepetierHost\composition.amf"`

Output is:

`Cannot find specified configuration file (C:\Users\AndrÚs\AppData\Local\RepetierHost\slic3r.ini).`

It looks like the accented "é" character is not supported by Slic3r. This did not happen in the previous stable version.

alexrj, 2017-03-26:
This was fixed long time ago, but since we ported many critical parts to C++ I'm leaving this open until we have a chance to test again on Windows (open file, load config, preset names, CLI processing).

lordofhyphens, 2017-03-26:
We can set up some scripts and run them on the appveyor server.

lordofhyphens, 2017-03-28:
My test using powershell on Windows 10 just failed :(

lordofhyphens, 2017-03-28:
There's the input and output. 
` perl .\slic3r.pl  --load "C:\dev\Andrés\config.ini"`
`Cannot find specified configuration file (C:\dev\AndrΘs\config.ini).`

alexrj, 2017-04-04:
Wrapping all file paths in `Slic3r::encode_path()` did solve the issue in the past. However:

* I applied many changes and `Slic3r::encode_path()` is not always called.
* Since most of the IO happens from C++ now we should handle this entirely on the C++ layer. Maybe Boost provides something useful.

lordofhyphens, 2017-04-15:
I've been digging around:
* http://boost.2283326.n4.nabble.com/boost-filesystem-path-as-utf-8-td4320098.html
* http://www.boost.org/doc/libs/1_51_0/libs/locale/doc/html/default_encoding_under_windows.html

Both seem relevant.

Epok13, 2017-04-24:
This bug with non-ASCII characters in path also happens in the latest 1.3.0 version (2017.04.21) when using the UI to add a STL file. I'm using Windows 10 64-bits on the latest insider build, French locale.

lordofhyphens, 2017-04-24:
Yeah, I have been looking into this, it has a common source

lordofhyphens, 2017-05-05:
Still a problem on Windows.
````
PS C:\Users\lenox\slic3r> perl .\slic3r.pl --load "C:\test\Andrés\config.ini"
Cannot find specified configuration file (C:\test\AndrΘs\config.ini).
````

lordofhyphens, 2017-05-18:
@afolgado can you check with http://dl.slic3r.org/dev/win/branches/slic3r-utf8.2017.05.18.1066.31115e0.64bit.zip

alexrj, 2017-05-19:
This was fixed now. It can be tested with the latest dev build available at http://dl.slic3r.org/dev/win

closed 2017-05-19 by alexrj

afolgado, 2017-05-23:
Hi, sorry I can't test it right now, but I trust it's fixed.

Thank you very much.

Epok13, 2017-05-24:
For my concerns, it's fixed with that build, thanks!

alexrj, 2017-05-24:
Thank you for providing feedback!

PR #3963 (2017-05-19): Use UTF-8 in libslic3r. Future-proof portable solution for non-ASCII files
Should fix #2955. I haven't tested this on Windows yet.

http://utf8everywhere.org