slic3r/Slic3r #2957: Slic3r 1.2.9a stable: dual extruder: brim created with wrong material despite correct setting and preview

R3D3-yellow, 2015-06-27:
In my dual extruder setup I have the following issue:
Despite defining the second extruder as providing both the support material and the brim/skirt, I find that the brim is created with the first extruder. I have traced this down to the G-Code that Slic3r 1.2.9a creates - 1.1.7 is not affected. Note that nifty new 3D preview in 1.2.9a purports that the same material is used for brim and support...

NimrodQuimbus, 2015-06-28:
I am having same sort of problem.

In Print Settings> Multiple Extruders> Extruders. 
All extruder settings seem to be ignored, and the default extruder is always used.

MartinMajewski, 2015-07-02:
Same here... But this problem seams so disappear for a while when you restart slic3r completely.

alexrj, 2015-07-02:
3D preview doesn't color extrusions by extruder.

R3D3-yellow, 2015-07-02:
Thanks- however why were 2 colours shown then? And in any case the extruder used for support and brim should have been the same, judging by the tooltip that appears over the support extruder field... or is that not correct?

lordofhyphens, 2015-07-02:
Second color is to show support and brim, contrasted with the model itself.

NimrodQuimbus, 2015-07-02:
in Print Settings> Multiple Extruders> Extruders. 
Slicing is always done using filament settings for extruder 1 no matter which extruder is shown/selected......

R3D3-yellow, 2015-07-03:
@lordofhyphens - thanks, didn't know that and it's useful. However the 3D preview happened to (inadvertently, I now gather) show what was in fact produced as G-code, namely support with T1 and brim with T0... despite the support (and implicitly brim) being defined as Ext2 = T1...

simonkuehling, 2015-07-24:
@alexrj, @R3D3-yellow, i confirm this bug for skirt as well: like the brim the skirt will currently always be printed with Ext.1, although *Support material/raft/skirt extruder* is set to Ext.2...

alexrj, 2016-03-19:
The tooltip for the _Support material extruder_ option says that it's used for skirt and raft as well, and no mention about brim.  In the code, skirt and brim are intentionally tied to the extruder used for the perimeters of the first object. Skirt needs a general redesign (there are other pending issues here). Printing brim with the first object extruder actually makes sense. What do you think?

simonkuehling, 2016-03-19:
There is one special case where brim should not be printed with the first object extruder: In case of using a raft with a separate support material (either for leveling out an uneven print surface or for more consistent bottom surface finish of soluble support prints), it would be best to always use the raft-extruder (=support extruder in this case) for brim as well.

Other than that your suggested logic is fine i think...

nicksears, 2016-06-20:
@alexrj I agree the logic makes sense - always use the perimeter extruder for the skirt/brim  - I assume this is to wipe it off as it goes to use it, helping ensure a clean perimeter)but why have the ability to override it if it won't actually let you? Also, I have an example where it does NOT use the perimeter extruder for the skirt (but the perimeter is extruder 2) 

Dictating the support/raft/skirt should work authoritatively, eve if it's not optimal. I've resorted to designing in a skirt for now...

alexrj, 2017-03-13:
Done. We use the perimeter extruder unless there's a raft: in that case we use the support material extruder. A new unit test guards this behavior.

closed 2017-03-13 by alexrj