slic3r/Slic3r #2958: Change to existing layer height override not having an effect

darrenfreeman, 2015-06-29:
Slicer 1.2.9.

Load an STL. (This was 10 mm high.) Right click the object -> settings. Enter a layer height of 0, for Z from 5 to 10 mm.

Go to layer view to verify that it is now 5 mm high. Not correct - there is a layer at 5.2 mm.

Go back to settings, and change the lower Z value to 4.8. Go back to layer view, no change.

Nothing that I did from this point seemed to have an effect. Change the layer height for that range to 0.2 mm (same as all the others), didn't cause layer view to update.

Delete the object, load it again, enter layer height of 0 from 4.8 mm to 10 mm. Check in layer view - last layer is now 5.0 mm as intended.

I'm guessing that once the layer range is created, there is no trigger for re slicing the object if the parameters are changed.

darrenfreeman, 2015-06-29:
I think that changing seam position from "random" to "nearest" or "aligned" doesn't trigger a slice either. But changing the XY size compensation does - so I'm guessing the intention is to make this happen all the time whenever a parameter is changed.

alexrj, 2015-07-04:
@darrenfreeman, Slic3r is smarter. It doesn't reslice everything, but it only recalculates the affected geometry. If you change Seam Position, nothing needs to be recalculated. If you change XY Size Compensation, it will affect perimeters (and subsequent steps), so Slic3r will keep the layer slices but will recalculate perimeters etc.

There was a bug in the range-based layer heights dialog that didn't trigger reslicing upon subsequent changes to the ranges. I fixed that now.

Regarding the layer at Z = 5.2mm, please note that according to your Layer Height and First Layer Height settings, an integer number of layers might not fit into your object height, so Slic3r needs to round height to the nearest multiple.
In other words, you can't do a 20mm object with 0.3mm layers. Slic3r will make a 20.1mm object instead.

alexrj, 2015-07-04:
(Maybe an option for generating thinner layers automatically would be useful...)

closed 2016-03-18 by alexrj