slic3r/Slic3r #2959: Possible memory issue

Psoc04, 2015-06-29:
Had just started using Slic3r 1.2.9 today and had rendered a part in the preview page and was constantly adjusting the support angle and scale of the part, after slicing it and exporting it for printrun/pronterface windows was saying low memory, checked and slic3r was using 11.3GB or RAM.

lordofhyphens, 2015-06-29:
Please review the contribution guidelines and update your post; your report
is missing information and as such is extremely unlikely to have something
done about it.

hakimio, 2015-07-02:
I can confirm that the more times you slice the object in the slic3r interface, the more memory it eats. Simply try moving, scaling or rotating the object to trigger automatic reslicing and see how the memory usage grows each time.

closed 2017-03-20 by alexrj