slic3r/Slic3r #2960: Gapfill problem with thin walls

teppix, 2015-06-29:
I think that the gapfill feature may be producing too wide extrusions, and needs to be configurable.

These tests were done in 1.2.9.

This model shows the problem really well.
https://dl.dropboxusercontent.com/u/448992/github/idler.stl
Slic3r profiles are linked below the images.

Note that the inner walls of the object is slanted in some places, which causes the gap fill to vary along the height.

Here is the result when using automatic width for perimeters:
[image]
Slic3r profile: https://dl.dropboxusercontent.com/u/448992/github/slicer_gapfill_issue/idler_a.ini

The measurements are pretty accurate, apart from the obvious bulge on the middle of the part. This bulge coincides with the layers where the gap-fill is made. The g-code shows perfect sides, but I believe that the gap fill perimeters apart.

I can work around the problem by manually tweaking extrusion width for perimeters, but that seems to introduce other inaccuracies in the model like the perimeters becoming inaccurate. (i have calibrated the extrusion multiplier, so I don't think that's the problem.

This is the result with manually set 0.5 mm perimeters.
[image]
Slic3r profile: https://dl.dropboxusercontent.com/u/448992/github/slicer_gapfill_issue/idler_b.ini

You can clearly see that the thickness of the rim is affected (guess this is another issue though).

I'll add more information if needed, but I think this is enough to be able to reproduce the problem.

a4jp-com, 2015-06-30:
My models were thick at the bottom too and I just thought it was a printer problem. Glad to hear it might be software related.

teppix, 2015-06-30:
That's interresting a4jp-com. I'd like to know if someone else is seeing the same issue, and also if you could check the g-code preview if there are any gap fills near the area where the problem occurs.

Just adding a clarifying illustration

[image]

Have also tried printing the part again with same settings as in the first image, except with 0% infill. The bulge is gone, so this proves that the gap fill is what's causing the problem.

Also, note that I'm using a 0.5mm nozzle, so maybe you won't see the same problems that I do at this scale if you have a smaller nozzle.

Perhaps the nozzle size is what triggers the problem?

hakimio, 2015-06-30:
I see the same problem with 0.4 nozzle. I don't think it has anything to do with nozzle diameter. Some things to note: 
* [extrusion width cannot be smaller than nozzle diameter * 1.05](http://manual.slic3r.org/advanced/flow-math) . So in your case minimum is 0.525.
* There is [an old issue](https://github.com/alexrj/Slic3r/issues/2599) dealing with pretty much the same issue
* Also you might want to take a look a [this one](https://github.com/alexrj/Slic3r/issues/2812) too.

I have 0.4mm nozzle and I have set external and internal perimeter extrusion width to 0.45mm which gets rid of the issue.

teppix, 2015-06-30:
hakimio, thanks for the comment. I will check the links more in detail. I did make some effort of finding the issue before, but apparently i may have failed :)

Regarding the extrusion width in relation to the nozzle diameter: You may be right regarding the 1.05 factor, but that's not really relevant to the issue.

1st print used automatic extrusion widths for everything.

2nd print used manual extrusion widths for perimeters (internal and external), but automatic for everything else. The width were tweaked so that the gap fill wouldn't be so significant.

So the problem I'm focusing on here is only present when in the 1st print, where I'm using automatic setting for extrusion widths. The only point I'm trying to make is that I can eliminate the problem by either manually setting extrusion width, or by setting infill to 0%.

teppix, 2015-07-01:
Here's almost definitely proof of the problem:
[image]

I have been taking a peek at the code, and if I understand it correctly, it has two different widths of gap fill.
https://github.com/alexrj/Slic3r/blob/master/lib/Slic3r/Layer/PerimeterGenerator.pm#L286-287

For small fills it always uses 100% of perimeter width for the gap fill, even down to gaps which is 0.1 times the perimeter width.

I'm not sure if I'm reading it correctly, but I think the picture speaks for itself in any case. the infill is pretty wide if you compare to the gap it's supposed to fill.

Will try to experiment a bit with the code to see if I can make it work better.

teppix, 2015-07-02:
https://github.com/teppix/Slic3r/commit/0a41715512c37bf4b32b8fa2a1be59d55bfcad9a
This update makes this issue disappear for me. The values are pretty arbitrary, and could probably be refined even more, but so far I'm getting pretty good results.

I'm not sure what $pspacing is referring to though, so it would be very welcome if someone with a bit more knowledge about this code could comment on this,

Also note that this issue doesn't seem to appear as much when using larger layer heights, but it becomes very apparent at 0.2mm layers. Also, like I mentioned, I'm using 0.5mm nozzle, and I'm not really sure what happens if I use a different width.

Also not sure how much this commit affects slicing performance. I haven't noticed any significant difference anyhow.

I'll be making more test prints to compare, but I'd be happy if more people could try this out and comment!

hakimio, 2015-07-02:
@teppix you should make a push request to get some feedback from @alexrj .

teppix, 2015-07-02:
@hakimio ok, thanks, I'll do that!

alexrj, 2015-07-03:
Thank you @teppix. I decided to put the error on the abundance side, since people usually complain about gaps more than they complain for overextrusion. But I agree that needs to be improved.
The more correct and more efficient solution will be to have the `medial_axis()` logic return the **actual** gap width; then we can assign it to the extrusion instead of trying fixed tiers. That way there would be no error at all, and variable-width gaps would be handled nicely.

Working on Slic3r I've learned one thing. Whenever tuning some hard-coded fudge-factor seems to fix some issue, it means that a better and more robust logic is needed. :)

teppix, 2015-07-05:
@alexrj it's nice to hear that this is an actual issue and not just my imagination :)

I have been tweaking the values a bit more, but I realize of course that this is not the most elegant solution. Would love to see a fix that does a correct calculation. Solving it the "right" way is beyond me at the moment but if you want my "hack" as a temporary solution, the current values which seems to work fine mostly, are available in this branch: https://github.com/teppix/Slic3r/tree/gapfix

I have seen some comments about 1.2.9 producing inferior results to earlier versions, I think this issue is one of the main reasons. Another one is the default settings, which I don't agree with 100%, but I suppose that's also in part a personal preference. This is a bit off topic here, but if tweaking the default settings is up for discussion I'd love to chime in, since I believe that a lot can be done to ease the transition for new users (migrating from previous versions or other competing slicers).

joecarpita, 2015-07-07:
@teppix @alexrj 

I just wanted to chime in and add my experience to the discussion. I think I'm experiencing the same problem? Over the weekend I updated from 1.1.7 to 1.2.9 and using the same settings I'm now getting some odd smudging as a result.

[image]

• a was sliced using 1.1.7. The settings worked great, everything was nice and smooth. 
• b was sliced using 1.2.9 using the exact same config file as I had used in 1.1.7. Nothing changed on the printer, or in the settings.
• c was sliced using 1.2.9 with detect thin walls turned off -- I thought it might have something to do with the adjustment to thin wall logic made that is mentioned on the changelog for Slic3r 1.2.9, so I turned off "detect thin walls."

This is a print of this part:  http://www.thingiverse.com/thing:903411

teppix, 2015-07-08:
@joecarpita 

i think it's very likely that the problem in picture B is the same.

In picture C you get other errors, simply becaulse slic3r doesn't check for perimeter collisions, so you may get too much extrusions there also, but that's expected. The assymetry is probably caused by pressure buildup in the extruder along the extrusion path. Part of the pressure problem might go away by increasing temperature and/or decreasing speed (assuming you are printing at lower temperatures now, i.e. 190-200 C).

a4jp-com, 2015-07-08:
It looks like the program is slowing the feed down too early and then when it starts up again maybe that amount is missing from the feed calculation. I could be wrong though. I have a slightly similar problem when the first blob of PLA is always a bit too thick on the start of the skirt but as it is the skirt it doesn't really matter. This problem would probably be really bad if you're trying to print a house with windows though. Do you have any other models that have the same type of problem?

joecarpita, 2015-07-10:
@teppix I'm printing at 210 deg. in PLA, 30mm/s in all those photos. To make sure is was something in Slic3r 1.2.9, I downloaded the part from Thingiverse again, and downloaded Slic3r 1.1.7 again, sliced, printed, came out perfect (like in photo A).

@a4jp-com I haven't experienced this with any other models, yet. I just updated to slic3r 1.2.9 this weekend and have been trying to troubleshoot this particular print. I'll try another thing wall with a logo on it and report back.

Hexman64, 2015-07-14:
I had too-much-material issues with gaps, too.

To investigate, I created a simple wall with 2.12mm width and set the extrusion width to 0.7mm:
[image]

This should give one perimeter and a gap fill which is 0.72mm wide. So the extrusion speed of the gap should only be slightly higher than for the perimeters.
I got:
; generated by Slic3r 1.2.9 on 2015-07-14 at 20:27:00
; external perimeters extrusion width = 0.70mm
; perimeters extrusion width = 0.70mm
...
G1 Z2.010 F15000.000  ;(somewhere in the middle)
G1 E1.60000 F1620.00000 ;(retraction)
G1 X80.350 Y120.710 E2.78121 F747.168
G1 X80.350 Y119.290 E2.81524
G1 X129.650 Y119.290 E3.99645
G1 X129.650 Y120.635 E4.02868
G1 X129.217 Y120.460 F15000.000
G1 X129.325 Y120.000 F15000.000
**G1 X80.675 Y120.000 E6.44835 F600.000  ;(gap fill with high E value)**
G1 E4.84835 F1620.00000

**while Slic3r 1.1.7 gives:**

G1 Z2.010 F15000.000
G1 E1.60000 F1620.000
G1 X75.350 Y100.710 E2.79290 F747.126
G1 X75.350 Y99.290 E2.82726
G1 X124.650 Y99.290 E4.02016
G1 X124.650 Y100.635 E4.05271
G1 X124.217 Y100.460 F15000.000
G1 X124.321 Y100.000 F15000.000
**G1 X75.679 Y100.000 E5.84512 F600.000  ;(a difference, even though I would expect E4.xx)**
G1 F1620.000 E4.24512

Shouldn't the E-Value be even less than in 1.1.7? I would have expected about the same extrusion amount for the gap-fill-line as for the perimeters.

teppix, 2015-07-14:
This looks like the same problem: https://www.youtube.com/watch?v=ZUgwaFuD5Us

oysteinkrog, 2015-07-17:
This bug is IMO a showstopper for having as 1.2.9.

teppix, 2015-07-17:
@oysteinkrog I agree completely. I think this problem creates a bad reputation for Slic3r.

On another note, Travis (author of the video in my previous comment) confirmed via IRC that changing from 0.2mm to 0.4 mm layer height fixes the problem, just like I described in an earlier comment. I think the video reflects the frustration many are feeling when trying to make their prints work with 1.2.9. I ceartainly had pretty much the same reaction myself before I realized what was causing the bad prints.

@alexrj what's your opinion on this? Do you think that there is a fix in sight? Otherwise think marking 1.2.9 as experimental is the best option, like @oysteinkrog mentioned, or you'll likely see a lot of people migrating to other slicers (especially first time users).

Hexman64, 2015-07-17:
I repeat myself 'cause it might be helpful for both, users and alexrj:
1.2.7 gives good (or at least much better) prints and has layer-preview built in,
1.2.8 and 1.2.9 have too much extrusion on gap-fills. 1.2.8 introduced autospeed, which may have led to the problem.
As a windows user, from the changelogs I read that 1.2.7 might actually be a good choice, as long as you check the preview and get around the use of broken models or non-ascii-characters.

NoNaym, 2015-07-19:
I pretty much immediately stopped using 1.2.9 and reverted to 1.2.7 shortly after 1.2.9 was released (because 1.2.7 beta was already working pretty damn well for me). Below is a comparison of the results from 1.2.7 and 1.2.9 side by side after printing a custom 40mm fan shroud I slapped together (sorry it isn't the sharpest image, but you can definitely see the difference). The settings for these prints were exactly the same between the two versions, only difference was 1.2.7 beta vs. 1.2.9 "stable".

[image]

oysteinkrog, 2015-07-19:
Does anyone know if there is a mitigate this issue?

Hexman64, 2015-07-19:
teppix said the issue doesn't appear as much when using higher layer heights, but from a calculation point of view it looks like the problem is the same.
Look at the gcode I get when I calculate the above wall (0.7mm perimeter and a 0.72mm gap fill):

**0.2mm layer height**
G1 E1.60000 F1620.00000 ;(after retraction)
G1 X80.350 Y120.710 E2.63731 F747.126 ;**(50x0.7mm perimeter, E increases by ~1)**
G1 X80.350 Y119.290 E2.66718
G1 X129.650 Y119.290 E3.70449 ;(another 50mm line, again about E+1)
G1 X129.650 Y120.635 E3.73279
G1 X129.217 Y120.460 F15000.000
G1 X129.321 Y120.000 F15000.000
G1 X80.679 Y120.000 E5.84660 F600.000 ;**(50x0.72mm gap fill line, but E increased by 2!)**

**0.4mm layer height**
G1 E1.60000 F1620.00000 ;(after retraction)
G1 X80.350 Y120.710 E3.53910 F747.402 ;(here the E value grew by 2)
G1 X80.350 Y119.290 E3.59495
G1 X129.650 Y119.290 E5.53405 ;(again E increased by 2)
G1 X129.650 Y120.635 E5.58695
G1 X129.217 Y120.460 F15000.000
G1 X129.343 Y120.000 F15000.000
G1 X80.657 Y120.000 E9.68448 F600.000 ;**(when gap filling, E grows by 4)**
G1 E8.08448 F1620.00000

So the layer height does not seem to have an influence.
But now I see that 16days ago alexrj has already become aware of the problem, so I'm afraid complaining about it isn't necessary any more^^

teppix, 2015-07-19:
@Hexman64 
You are right that the problem doesn't go away. The with calculation is still done in the same way - it's just a very rough estimation really, but it makes the problems less apparent at least.

Larger layer heights results in wider extrusion widths, if set to automatic (0). So the walls gets thicker, and at least for very thin walls there will be less or none gap fill. Perhaps this just makes the problem appear in different places, but from my experience after doing a lot of test prints, I prefer the result from 0.4 layers over 0.2 layers.

Note that setting gap fill speed to 0 (not very obvious) will disable gap fill, and do ordinary infill instead, and will simply leave an empty space in small gaps. I guess this will work like in earlier versions. I haven't really tried this much.

teppix, 2015-07-19:
I think the actual problem is well defined (read the comment from @alexrj), so it's mostly a matter of implementing a fix. I have kept posting just to bring some attention to it, since i think it's a very important issue.

I'll try to take a shot at it myself, but I'm not very familiar with the code, so don't hold your breath. At least I know where to start digging, if I can only find the time for it...

luisonoff, 2015-07-22:
@teppix [OFFTOPIC] You said you wanted to change default settings, I would like to know what would you change, or what you think are proper default settings, maybe this is not the place, but you could post them somewhere or start a new issue. Thank you

simonkuehling, 2015-07-24:
@alexrj I'd like to vote for priorizing this bug, as it is quite hard to debug for the average user and not quite easy to predict as well. For thin walled parts (think hollow boxes, example below), the issue is not just a cosmetic one, but can also reliably result in failures mid-print - due to gap-fill excess material piling up and sealing the nozzle during the next layer (nozzle backpressure getting too high).

While the workaround to disable gap-fill by setting *Gap fill* speed to 0 serves as an option, the speed improvement of gap fill (and the general functionality where it is working correctly) is quite substantial for many complex models. I **really** appreciate that!

My **example** for reference:
[box.stl](https://www.dropbox.com/s/hm4rgxroozjfg9a/box.stl?dl=1)
[box.gcode](https://www.dropbox.com/s/jlq6625xwe7xrdc/box.gcode?dl=1)
[config.ini](https://www.dropbox.com/s/k0r723n7uw8i5se/config.ini?dl=1)

Preview:
[image]

While this box looks alright, the extrusion width of of the gap fill is way to wide - the gap that needs to be filled is actually smaller than one perimeter width.

luisonoff, 2015-07-26:
+1 on this request, gap filling is broken on this version. Also would love to see it compatible with autospeed. #2976

jsatan, 2015-08-09:
I'll add my 2p to the mix, I've come across this issue (I'm new to 3D printing) and my fix was to turn off 'detect thin walls'. In my prints the walls were fine but the infill was building up and creating bumps which added a few layers to the infill.

teppix, 2015-08-10:
@jsatan thanks for the input, but I believe that might not really be related to this particular problem, even though it is similar. In the example I posted att the top, there is no noticable difference in the result. I.e. the gap fill is still there. and the perimeters look the same.

By the way, I'm laying low on this issue now. I don't know exactly what @alexrj is up to, but it looks like a lot of the code related to this is being ported from perl to C++ right now, so I assume that he will adress the issue when that is nearing completion.

nicksears, 2015-08-14:
I thought I had already chimed in on this, but I must have only commented on the reprap forum. This is a KILLER issue. Does anyone ever get any good prints from 1.2.9? How? It's terrible unless you manually select the extrusion width (I've only tried setting them all to 0.5). 1.2.7 is actually not perfect, either.

I did a comparison of a few parameters 1.2.7 vs 1.2.9 (defaults, detect thin walls, gap fill to 0, extrusion width to 0.5), and the only ones that were good were 1.2.7 and 1.2.9 with 0.5 EW (which was best).

CapnBry, 2015-08-15:
I ran into a serious gap fill issue on 1.2.9 as well, trying to print [this 3D printer test](https://www.thingiverse.com/thing:704409). The interesting bit happens a layer height 6.24 when Slic3r tries to gap fill a 0.779mm gap with my 0.4mm extruder. It ends up with this
```
G1 X105.611 Y103.513 F7800.000 ; move to first infill point
G1 E2.00000 F2400.00000 ; unretract
M106 S255 ; enable fan
G1 X105.611 Y97.576 E3.02770 F3600.000 ; infill
G1 X106.390 Y97.576 E3.16241 ; infill
G1 X106.390 Y103.513 E4.19011 ; infill
```
The move from Y103.513 to Y97.75 = 5.763mm and extrudes 1.0277mm of filament. This is 1.75mm filament so at 0.24mm layer height that's a 1.6mm wide bead _on both sides_ or 4.1x the width of the gap. This might go unnoticed but I have pressure advance on, so to prepare for this giant extrusion move, it *pressure advances over 18mm* of filament at 30mm/s which doesn't really work all that well, then it tries to pressure release the same amount which immediately jams the E3DV6 extruder.

The interaction between the gap fill and the pressure advance is really catastrophic, as the error in the gap fill is squared by the pressure advance algorithm. Just wanted to add another test case to the ticket and would be happy to experiment further.

[config.ini](http://www.pastefile.com/f36Ev2)

nopick, 2015-08-19:
I am having problems with this as well. It seems to have broken after 1.2.6. I print thin walled parts with design thicknesses of 0.04 and 0.02 inches with 0.4mm E3D V6 and AO Hexagon hot ends. This problem results in the wall thickness being nearly 2X the design thickness.

I would be happy to test any potential fixes. I am currently running 1.2.6 but it crashes very frequently.

nicksears, 2015-08-19:
Agreed, I tried switching back to 1.2.7 but even it has issues (although not as bad as 1.2.9).
1.2.6 works well but has no 3D preview and crashes constantly. 1.2.6 it is for now.

wfcook, 2015-08-29:
I ran an interesting test this morning which I believe is relevant. I slic3d a rather large and complex thin-wall model with gap fill on in slic3r 1.1.7, 1.2.6, and 1.2.9 using the same settings exported from 1.1.7 in all three. Then I looked at the total extrusion length. The results for 1.75mm filament:

Slic3r 1.1.7: 64627.14mm
Slic3r 1.2.6: 57371.95mm
Slic3r 1.2.9: 77450.74mm

Additionally, with similar settings:

Simplify3D 3.0: 65326.95
Cura: 

The raw volume of the part is 153951.51 mm^3, which would imply a total required extrusion length of 64005.60 mm

This would imply that something is a little off with the Slic3r 1.2 family's extrusion rate calculators.

simonkuehling, 2015-08-29:
@wfcook, just for sake of completeness: i assume you sliced the model with solid 100% infill...?

Clever approach for a quick validation!

wfcook, 2015-08-29:
Yes, I used 100% infill. Sorry, I should have made that clear.

lonelymyp, 2015-09-16:
I got the exact same problem with thin walls when using pressure advance that appeared in 1.2.9

disable this feature by setting all the parameters manually, version 1.2.9 began to print thin walls as well as the old version.

wfcook, 2015-09-16:
I have pressure advance turned off.

As soon as I get some time (maybe next week...) I'll try to clean up, organize, and present my further tests. I created some simple rectangular extrusions, 100mm L x 10mm H at various widths and ran them with the simplest settings I could muster, using 100% infill and with gap fill on. Versions 1.2.9 and 1.2.6 produced extrusion rates that frankly didn't make much sense, while 1.1.7, Cura, and Simplify3D produced much better results.

Quite simply put, if your 100% infill part doesn't extrude the correct volume of filament (and particularly if it over-extrudes by 20+%) you are going to get a print that doesn't look very good.

nicksears, 2015-09-16:
Some, I'd suggest testing 1.2.6 as well though, from what ive seen (and
confirmed with a single test comparison) 1.2.6 extrudes accurately do I've
been using that for all my research. It crashes constantly though.

nopick, 2015-09-16:
I agree with nicksears. I do not see this problem on 1.2.6 but it does
crash a lot.

nicksears, 2015-09-16:
In fact, I'd say that these major flowrate changes should be part of a beta/fork,  not a major "stable" release. 

Someone could add the changes from 1.2.7 that fixed the crashing and call it 1.2.6.1 and that should be the current stable release

nopick, 2015-09-16:
I would LOVE a stable 1.2.6! It does exactly what I need except for the
crashes.

wfcook, 2015-09-16:
@nicksears & @nopick I am testing 1.2.6 also, and as you can see from above, 1.2.6 doesn't extrude correctly either, though in my tests it tends to under-extrude dramatically while 1.2.9 over-extrudes dramatically. The under-extrusion does not produce prints that look as bad as the over-extrusion does.

For rectangular extrusions of varying thickness, I am getting volumes that look like this:
[image]

Which graphs like this:
[image]

You can see the step functions where a particular slicer is leaving empty space (!) until the infill kicks in.

nopick, 2015-09-16:
Wow. That is good info. I might give the old 1.1.7 a shot and see what it
does to my thin wall parts.

Thanks.

lordofhyphens, 2015-09-16:
@wtfcook, could you write down your experimental steps (or link them if
you've put them somewhere else)? I'd like to try to replicate your
experiments with my hotend/nozzle/filament combination. Might be
illuminating, especially if I can work out where the volume is in in the
code and how the math changed from 1.1.7 through the 1.2.x experimentals to
1.2.9.

That kind of information should be able to be extracted and worked out with
enough digging.

Since I have volumetric extrusion configured, I can probably cook something
up to look at the Gcode for the requested E axis moves (on a volume
standpoint), and I think it would be useful to recreate your print tests
with my setup and compare to the ideal "requested", and probably work up
multiple line/nozzle width combinations.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

wfcook, 2015-09-16:
@lordofhyphens I constructed some models of boxes of various thicknesses, always 100mm long and 10mm high and at various thicknesses from 1.0mm to 2.0mm. I then developed a print configuration using a 0.5mm nozzle, 100% infill, no skirt/raft/support/etc., single perimeter, zero top/bottom layers, and otherwise as simple a setup as I could imagine. The one weird variable is the infill overlap%, which cannot be specified in 1.1.7 and which I ran at both 0% and 15% in the other two versions of Slic3r (I don't remember which are in the table above).

I sliced each one, measure the total length of extruded filament using gcode.ws (one could also use the number at the end of the gcode file from Slic3r), and calculated the volume based on the 1.75mm filament I used when slicing (I tried to use a setup I could actually attempt to print, although I haven't found time in the queue to do that yet).

I also tried some sensitivity testing, i.e. thicknesses of 1.9mm 1.99mm, 2.01mm, and 2.1mm, but I haven't had the time to collate those results, and I did it a few weeks ago and frankly forget what I learned then. My recollection is that as you go more than some margin beyond a factor of extrusion width things get mighty weird. I made one of the slicers (Cura, I think) go completely linear by using zero perimeters, but I haven't tested the others rigorously.

I am operating on the assumption that the actual extruded volume should probably be slightly less than the exact volume of the solid because of small voids. The results were enough to make me choose to use a different slicer for the time being.

This procedure could also be used to test infill percentage volumes to make sure we're getting what is desired. There have been a couple of threads on that as well.

I hope this is all useful to somebody.

lordofhyphens, 2015-09-16:
@wfcook I was planning to set it up for volumetric extrusion (which is
something I can do with all of the versions of Slic3r and I think Cura),
and that should let me total up the actual E axis in mm^3 and compare them.

Thank you for writing down your experimental setup.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

wfcook, 2015-09-16:
@lordofhyphens please share your results with us!

lordofhyphens, 2015-09-22:
Well, I have the numbers churning now, but I thought I'd put my
experimental code online:
https://github.com/lordofhyphens/slic3r-comparison-test

It's a bash shell script. I guess if someone wanted to make this more...
flexible python would be a way to go (although this should work with a
couple minor changes and msys on windows).

What I did was download every version of slic3r back to 0.9.10b for Linux
(x86) and extracted each to its own directory. The script itself just
changes the infill and slic3r version and collates the total E steps for
perimeters, infill and total used (according to slic3r). Because older
version of slic3r don't have volumetric, I've set the filament diameter to
1.128379 (from
http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide#Optional:_Switch_to_volumetric_E_units),
and I used relative E units so I could add everything together manually
from the gcode.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

imqqmi, 2015-09-26:
I got the same problems and seen similar issues reported on reprap.org forum. I'm using 1.2.7 as that prints well enough for me. I stick with that version until this issue is resolved. Keep up the good work!

legarasta, 2015-10-06:
Similar problems here. Rather new to 3D printing and i must agree, 1.2.9 should not be "advertised" as stable - and such reference should be removed imediatly as to not affect any more Slic3r's reputation, specially to newcomers who are not so informed.

Even though, cheers on the developers and thank you for this great tool they have developed!

nicksears, 2015-10-22:
I;m just going to bump this hoping for a fix. This is ridiculous. 1.1.7 has the most appropriate fill density but doesn't do multimaterial very well (and doesn't have the 3d visualization). 1.2.6-1.2.8 crash all the time, and 1.2.9 overfills like crazy. I've tried all of these ideas and none can tweak 1.2.9 so it doesn't overfill. I'm about publish a biomedical engineering research paper and I even have slic3r in my methods, but I'm at my wits end and may have to switch. Has anyone gotten 1.2.9 to work well?

curiouspl2, 2015-10-22:
i've noticed this bug aswell, and i agree it is confusing - it took me pretty long time until i've figured out gap fill is the problem and got many failed prints during the process. 

It is especially annoying after one spends lot of time tuning print/printer and then just tries to move to lower layer height , then gets everything messed up. 
for layer heights 0.3 and above it does not seem to be so apparent.

also it would be nice if gap fill density could be somehow regulated - on many delicate, dimension-critical parts it's enough there is at least something stuffed inbetween walls so they do not collapse, that means just few 'dots' per mm would suffice and would be much quicker/no risk of re-melting adjacent walls. 100% gap fill often pushes such walls on the sides or deforms them , especially as somehow fill algorithms do not follow reality ideally when complex shapes are involved (for nice test just try to print single layer object like gear and notice there are many places where perimeters do not join, while at other places they are overfilled)

a4jp-com, 2015-10-22:
I hope Alex comes back to help again soon.

hakimio, 2015-10-22:
There were no commits to slic3r repository since 6th of August. He seems to be working on other projects like alexrj/DBIx-Lite .

Tinchus2009, 2015-10-23:
Well.. Im new in 3d printing. I started 2 monthhs ago. I tested a couple of slicers. At the begginig I used the slic3rs version that cames with repetier host. It also uses cura. Both of tehm worked for me.
Then I wanted more... I tried complete versions of slic3r: 1.2.9... My prints suddenly were HORRIBLE... I was confused. I tried to manually tweak it... Not great results with that. I finally quit and move to the new cura, wich is really grat for me. The same prints were great in cura... so slic3r was left away.
Now I come back decided to discover why I could not print a simple calibration test (my printer IS calibrated, so...)
And my problem is this issue being discussed here. The gaop filling is way overextruded. The plastic acummulates in a couple of layers and you end up with a mess.
Now I have implemented your advices: turning to 0 gap speed and manually setting extrusion . And this solution works somehow great. I have just finished a print test and looks great.
This is an issue that should be fixed really quick... many beginner like me could just go away, and the reputation to slic3r is not going to be good. If you just try cura and then slic3r and see the results (without manually setting anything) you will just escape from sli3r for sure.

a4jp-com, 2015-10-24:
I tried Cura and it was really good but the guy making it didn't want to have a 4 point calibration for people with hotbeds. Just 3. If you don't have a hot bed I think it does the 4 point calibration though. That is why I like Slic3r better. Alex used to alway make small updates to fix bugs as well. Slic3r is so close to being fixed. I hope it gets a few more tweaks though.

ghuser2, 2015-10-25:
I've only used 1.2.9, for about 3 months. I print some very narrow traces sometimes, and only with gap fill turned on it will print them.

moncholo, 2015-10-25:
HI. There's a problem with infill in thin walls in Slicer 1.2.9. I've solved the problem using slicer 1.1.7 stable. A bug was reported when slicer 1.1.7 was released. The name of the bug is "Some extra spots were added in thin walls" and I think that this bug appears again in 1.2.9.

The image shows the spots in a thin wall using 1.2.9

[image]

lordofhyphens, 2015-10-31:
Seems that one of the situations is where the perimeter generation (in libslic3r) decides to gapfill a small area that is already infilled.
No gapfill:
http://imgur.com/QcPwvw0
Gapfill + infill in the same layer.
http://imgur.com/n5PBtMs

Since infill is still being handled in Perl and perimeter/gapfill in XS, I'm wondering if it'd be a good idea to either try to subtract areas that were detected to be infilled from gapfill consideration or the other way around.

[95 more replies]

closed 2016-03-26 by alexrj