slic3r/Slic3r #2963: Switch to 3D view should retain same layer/zoom after slicing changes

peter1960, 2015-06-30:
Currently when you have zoomed to a size in 3d view at a layer, if you change any slice settings and return to the 3d view the zoom is reset and all the layers shown.
It would be good if the previous view was retained so tweaks can be easily checked in a problem area for example.

teppix, 2015-07-14:
I agree.

The only time a view reset might be desirable is when objects are added/removed, and perhaps not even then.

It's easy to get lost in the 3D viewport though, and may be hard finding your way back, so a home-button would be nice, instead of resetting the view for every change. I know that clicking in the object list resets the view to the selected object, but it's not very intuitive. Took me a while to figure that out.

alexrj, 2017-03-20:
This was done a long time ago. Thank you for suggesting :)

closed 2017-03-20 by alexrj