slic3r/Slic3r #2965: tests fail on current master

wolfmanjm, 2015-07-01:
Test Summary Report
-------------------
t/multi.t                    (Wstat: 139 Tests: 0 Failed: 0)
  Non-zero wait status: 139
  Parse errors: Bad plan.  You planned 13 tests but ran 0.

bullding on linux. worked before git pull this morning

lordofhyphens, 2015-07-02:
Need more information about your setup. The Travis server is reporting that
all tests pass, so the problem is more likely related to your particular
system.

wolfmanjm, 2015-07-02:
Ubuntu 14.04 32bit system. doing a clean sudo perl BUILD.PL.
Using stock perl.

wolfmanjm, 2015-07-02:
It got fixed, thanks ;)

closed 2015-07-02 by wolfmanjm

lordofhyphens, 2015-07-02:
You probably should use my fork anyway; make packages and do less damage to
system perl.

alexrj, 2015-07-02:
Yeah thank you, the master branch was a bit unstable these days because of heavy refactoring...