slic3r/Slic3r #2966: Slice issue

ming2008cn, 2015-07-02:
Please refer to: http://forums.reprap.org/read.php?263,527768  for details.

Thank you,

a4jp-com, 2015-07-02:
Can you please post the 3D file and also config file here?

ming2008cn, 2015-07-03:
Unfortunately, I do not know how to attach the .amf file here. But it is a short .xml file I modified from Slic3r output. I attached it below.

I am using the original Slic3r configuration, no change.

Thank you, 

-----------------

<?xml version="1.0" encoding="UTF-8"?>
-<amf unit="millimeter"> <metadata type="cad">Slic3r 1.2.9</metadata> -<object id="0"> <metadata type="name">cubassembly.stl</metadata> -<mesh> -<vertices> -<vertex> -<coordinates> <x>7.54288578033447</x> <y>-17.876522064209</y> <z>50</z> </coordinates> </vertex> -<vertex> -<coordinates> <x>-12.4571142196655</x> <y>-17.876522064209</y> <z>50</z> </coordinates> </vertex> -<vertex> -<coordinates> <x>-12.4571142196655</x> <y>2.12347793579102</y> <z>50</z> </coordinates> </vertex> -<vertex> -<coordinates> <x>-12.4571142196655</x> <y>2.12347793579102</y> <z>60</z> </coordinates> </vertex> -<vertex> -<coordinates> <x>7.54288578033447</x> <y>2.12347793579102</y> <z>50</z> </coordinates> </vertex> -<vertex> -<coordinates> <x>7.54288578033447</x> <y>2.12347793579102</y> <z>60</z> </coordinates> </vertex> -<vertex> -<coordinates> <x>-12.4571142196655</x> <y>-17.876522064209</y> <z>60</z> </coordinates> </vertex> -<vertex> -<coordinates> <x>7.54288578033447</x> <y>-17.876522064209</y> <z>60</z> </coordinates> </vertex> </vertices> -<volume> <metadata type="name">cubassembly.stl</metadata> -<triangle> <v1>0</v1> <v2>1</v2> <v3>2</v3> </triangle> -<triangle> <v1>3</v1> <v2>2</v2> <v3>1</v3> </triangle> -<triangle> <v1>4</v1> <v2>0</v2> <v3>2</v3> </triangle> -<triangle> <v1>5</v1> <v2>4</v2> <v3>2</v3> </triangle> -<triangle> <v1>5</v1> <v2>2</v2> <v3>3</v3> </triangle> -<triangle> <v1>6</v1> <v2>1</v2> <v3>0</v3> </triangle> -<triangle> <v1>6</v1> <v2>3</v2> <v3>1</v3> </triangle> -<triangle> <v1>7</v1> <v2>0</v2> <v3>4</v3> </triangle> -<triangle> <v1>6</v1> <v2>0</v2> <v3>7</v3> </triangle> -<triangle> <v1>7</v1> <v2>4</v2> <v3>5</v3> </triangle> -<triangle> <v1>6</v1> <v2>5</v2> <v3>3</v3> </triangle> -<triangle> <v1>6</v1> <v2>7</v2> <v3>5</v3> </triangle> </volume> </mesh> </object> -<object id="1"> <metadata type="name">cubassembly2.stl</metadata> -<mesh> -<vertices> -<vertex> -<coordinates> <x>25</x> <y>-25</y> <z>0</z> </coordinates> </vertex> -<vertex> -<coordinates> <x>-25</x> <y>-25</y> <z>0</z> </coordinates> </vertex> -<vertex> -<coordinates> <x>-25</x> <y>25</y> <z>0</z> </coordinates> </vertex> -<vertex> -<coordinates> <x>-25</x> <y>-25</y> <z>50</z> </coordinates> </vertex> -<vertex> -<coordinates> <x>25</x> <y>25</y> <z>0</z> </coordinates> </vertex> -<vertex> -<coordinates> <x>-25</x> <y>25</y> <z>50</z> </coordinates> </vertex> -<vertex> -<coordinates> <x>25</x> <y>-25</y> <z>50</z> </coordinates> </vertex> -<vertex> -<coordinates> <x>25</x> <y>25</y> <z>50</z> </coordinates> </vertex> </vertices> -<volume> <metadata type="name">cubassembly2.stl</metadata> -<triangle> <v1>0</v1> <v2>1</v2> <v3>2</v3> </triangle> -<triangle> <v1>3</v1> <v2>2</v2> <v3>1</v3> </triangle> -<triangle> <v1>4</v1> <v2>0</v2> <v3>2</v3> </triangle> -<triangle> <v1>5</v1> <v2>4</v2> <v3>2</v3> </triangle> -<triangle> <v1>3</v1> <v2>5</v2> <v3>2</v3> </triangle> -<triangle> <v1>6</v1> <v2>1</v2> <v3>0</v3> </triangle> -<triangle> <v1>3</v1> <v2>1</v2> <v3>6</v3> </triangle> -<triangle> <v1>7</v1> <v2>0</v2> <v3>4</v3> </triangle> -<triangle> <v1>6</v1> <v2>0</v2> <v3>7</v3> </triangle> -<triangle> <v1>7</v1> <v2>4</v2> <v3>5</v3> </triangle> -<triangle> <v1>3</v1> <v2>7</v2> <v3>5</v3> </triangle> -<triangle> <v1>3</v1> <v2>6</v2> <v3>7</v3> </triangle> </volume> </mesh> </object> -<constellation id="1"> -<instance objectid="0"> <deltax>250.000007450836</deltax> <deltay>249.999988741436</deltay> <rz>0</rz> </instance> -<instance objectid="1"> <deltax>250.000007450836</deltax> <deltay>249.999988741436</deltay> <rz>0</rz> </instance> </constellation> </amf>

alexrj, 2015-07-04:
Thank you, duplicate of #2970

closed 2015-07-04 by alexrj