slic3r/Slic3r #2967: 1.2.9 / Ubuntu 14.04 : GUI won't start

GlennSmith81, 2015-07-02:
Hi,
I've been using Slic3r for a year or so.  Just downloaded V1.2.9 and can't get it to work.  If I try launching from a terminal, I get 
Can't load '/home/glenn/Slic3r/lib/vrt/9e109cfe461eb9a764d58156ba7bc844/OpenGL.so' for module OpenGL: libglut.so.3: cannot open shared object file: No such file or directory at <content>/DynaLoader.pm line 157.
 at /home/glenn/Slic3r/lib/std/Slic3r/GUI/Plater/2DToolpaths.pm line 120
Compilation failed in require at /home/glenn/Slic3r/lib/std/Slic3r/GUI/Plater/2DToolpaths.pm line 120.
BEGIN failed--compilation aborted at /home/glenn/Slic3r/lib/std/Slic3r/GUI/Plater/2DToolpaths.pm line 120.
Compilation failed in require at /home/glenn/Slic3r/lib/std/Slic3r/GUI.pm line 16.
BEGIN failed--compilation aborted at /home/glenn/Slic3r/lib/std/Slic3r/GUI.pm line 16.
Compilation failed in require at (eval 71) line 1.

What's gone wrong ?  I have saved the 1.2.7 directories, but that version wont run either.
Frustrating, as I use my 3D printer professionally...
Glenn

xoan, 2015-07-02:
Please, use the search box before open a new issue.

https://github.com/alexrj/Slic3r/issues?utf8=%E2%9C%93&q=is%3Aissue+libglut.so.3

Duplicated: #2953

GlennSmith81, 2015-07-02:
I did use the search box - but not with exactly that phrase...   I'm not a Linux expert : my computers are tools for my trade.   Thanks, anyway, for pointing me in the right direction.  I hadn't seen the notice about freeglut : perhaps it needs to be written in **bold**.

closed 2015-07-02 by GlennSmith81

xoan, 2015-07-02:
It is a system error message, so Slic3r can't do anything to format the output. Even searching for OpenGL (easy to read in the error message) you've been reached almost the same search results :)

PS. It is allways a good practice to know (even a litle bit) about the tools you use for your trade.

a4jp-com, 2015-07-02:
Xoan, I think the PS message is a bit cutting...

feiming, 2015-09-22:
Since this is the first few result from google search. I'm going to put the solution in here.

sudo apt-get install freeglut3

iviarki, 2015-11-30:
Thanks feiming! One less click to get the solution!

lordofhyphens, 2015-12-01:
Use hyperair's Ubuntu package.

EktoHunter, 2017-09-02:
I still have this problem when i run ./slic3r --gui
This is my Output: 
Can't load '/home/ekto/Sonstiges/Slic3r/lib/std/auto/Wx/Wx.so' for module Wx: libgtk-x11-2.0.so.0: Kann die Shared-Object-Datei nicht öffnen: Datei oder Verzeichnis nicht gefunden at <content>/DynaLoader.pm line 157.
 at /home/ekto/Sonstiges/Slic3r/lib/std/Slic3r/GUI/AboutDialog.pm line 6
Compilation failed in require at /home/ekto/Sonstiges/Slic3r/lib/std/Slic3r/GUI/AboutDialog.pm line 6.
BEGIN failed--compilation aborted at /home/ekto/Sonstiges/Slic3r/lib/std/Slic3r/GUI/AboutDialog.pm line 6.
Compilation failed in require at /home/ekto/Sonstiges/Slic3r/lib/std/Slic3r/GUI.pm line 8.
BEGIN failed--compilation aborted at /home/ekto/Sonstiges/Slic3r/lib/std/Slic3r/GUI.pm line 8.
Compilation failed in require at (eval 71) line 1

PS: Thats what i got after installing freeglut3

lordofhyphens, 2017-09-03:
> oder Verzeichnis nicht gefunden at /DynaLoader.pm line 157.

Can you try the AppImage build?

EktoHunter, 2017-09-03:
Appimage? Do you mean the Tarball archive?

lordofhyphens, 2017-09-03:
https://dl.slic3r.org/dev/linux

EktoHunter, 2017-09-03:
it just gives me the commands again when I do ./Slic3r --gui not appearing with a gui

lordofhyphens, 2017-09-03:
Something else is probably going on. Which is weird as 14.04 is the build
system.

BETLOG, 2018-05-02:
Is 14 still the build system?
I've been trying to tolerate the new kubuntu releases and 18 is almost bearable, the only catch is that almost none of my preferred tools support it. Slic3r being the current point of interest.
I'm happy to keep using 14.04, but next year it's LTS runs out, and updates from this point onwards are historically likely to start breaking things.

lordofhyphens, 2018-05-02:
The build system is what Travis gives us, with some library modifications
(newer gcc, etc). Currently that is Trusty Tahr (14 LTS).

I build for myself on Debian Sid.

You can check the .travis.yml and what it calls to see what gets used.

lordofhyphens, 2018-05-02:
The Appimage that gets made on 14 runs just fine on Debian Sid.

EktoHunter, 2018-05-02:
I've switched to Mac due to incompability of  Linux and recurring problems. The app image works just fine on newer versions of Linux. The program install also started working, as soon as you use systems like Linux Mint 18.01 as they are built on Ubuntu 17.04.

lordofhyphens, 2018-05-02:
You should see the mess I had to go through to get the OSX build working.