slic3r/Slic3r #2970: 1.2.9 Multiple objects (for dual extrusion) - Z position incorrect when generating layers

sreid55, 2015-07-02:
When slicing objects with multiple parts (e.g., from an AMF file--sample attached, or multiple STL files),
the parts appear correctly positioned in the 3d preview, but when sliced, all parts are positioned with their first layer at z=0.  
The output G-code can result in two materials (from two extruders) occupying the same space.

This is a problem for dual extruder prints when some parts (materials) sit on top of other material.

Pictures (Screenshots); http://imgur.com/a/OMFXD
Sample file attached to this post:
http://forums.reprap.org/read.php?263,528202

teppix, 2015-07-02:
I think this is a duplicate of https://github.com/alexrj/Slic3r/issues/2966

sreid55, 2015-07-02:
You're correct.   I saw you put a reference to this one in that bug report, and, modesty aside, I think the description on mine is more descriptive so I'm going to leave it open ;)   
Thanks

teppix, 2015-07-02:
@sreid55 your description was definitely better. I don't blame you for not finding the other one :)

alexrj, 2016-03-19:
Technically speaking, that AMF file doesn't represent an object with multiple parts. It represents multiple distinct objects. And Slic3r lays each object on the print bed (yes, this is not shown in the 3D plater but can be seen in the 3D preview). Your CAD should export the AMF file in a more correct way: one single object with multiple parts (volumes). Slic3r would recognize it as you expect and, besides the correct position, it would provide you with smarter and better toolpath.

Maybe we should detect AMF files with more than one object, having different Z positions, and ask the user whether it's supposed to be a single object with multiple parts instead.

alexrj, 2016-03-19:
I implemented this.

closed 2016-03-19 by alexrj