slic3r/Slic3r #2971: New ASCII STL parser crashes with UPPERCASE STL files

hroncok, 2015-07-02:
I was about to accept the new ASCII STL parser you've introduced in 6095427926bce69548b25a58abbd29e318232817 (and corrected in 911bed827c9e3b8f88dbbf313a02f2dffd8be34c and be635c69e7d8d9a33ccea4a0dae230fead9f8af9) into admesh. Reading the code made me think this cannot work with UPPERCASE files, such as:

```
SOLID  Untitled1
  FACET NORMAL  0.00000000E+00  0.00000000E+00  1.00000000E+00
    OUTER LOOP
      VERTEX -1.96850394E+00  1.96850394E+00  1.96850394E+00
      VERTEX -1.96850394E+00 -1.96850394E+00  1.96850394E+00
      VERTEX  1.96850394E+00 -1.96850394E+00  1.96850394E+00
    ENDLOOP
  ENDFACET
...
```

And yes in deed, using such file ([block.stl](https://github.com/admesh/admesh/blob/master/block.stl)) crashes Slic3r 1.2.9. As you can see, Github recognizes the file, so let's assume UPERCASE in STL is a valid thing and might happen.

alexrj, 2015-07-03:
I had never seen those, actually. They look quite wrong...
Have you ever found such a file? What software does generate them? Do you have, say, a Thingiverse link to such a file? Thank you...

hroncok, 2015-07-03:
So far I found only this block.stl that comes form the original admesh sources from  http://www.varlog.com/admesh-htm

Might be, this is the only case. I think finding such file on the internets might be a bit hard thing to do.

Some resources say the keywords [must appear in lower case](http://www.fabbers.com/tech/STL_Format). However the [thesis that comes with admesh](http://www.varlog.com/admesh-htm/ADMeshThesis.zip?attredirects=0&d=1) says:

> Although the STL specification shows all keywords in lowercase, many ASCII STL files have uppercase keywords.  Therefore, any tool which reads STL files should accept either uppercase or lowercase keywords.

(Without nay link to source.)

Netfabb loads the file OK.