slic3r/Slic3r #2973: ./xs failed

a4jp-com, 2015-07-03:
What can I do to compile? Everything has been working perfectly and compiling up to now.

[image]

a4jp-com, 2015-07-03:
[image]
[image]

alexrj, 2015-07-03:
Hm, I'll need to use `_putenv_s()` on Windows instead of `setenv()`.
https://msdn.microsoft.com/en-us/library/vstudio/eyw7eyfw(v=vs.110).aspx

a4jp-com, 2015-07-03:
Okay. Thank you. 

I bricked my development version so I don't think I'll be able to do anything until it gets tweaked.

Is there anything I can do or try in the meantime? Is it difficult to switch from `_putenv_s()` to `setenv()`?

alexrj, 2015-07-03:
You can use the stable branch instead of master.

a4jp-com, 2015-07-03:
I tried but that was the version before the infill bug or infill rendering bug got fixed. I'll date the files I download next time. Can't wait for a compilable version. Everything is starting to work really well in Slic3r and a bug free version of Marlin is almost ready too ^^

alexrj, 2015-07-04:
Infill bug?
The stable branch should be stable. If it's not, tell me.

a4jp-com, 2015-07-06:
The stable version has the problem where the infill lines aren't being drawn properly. You fixed this in the development version before and everything was really good until the porting of Slic3r::GCode to XS (speed boost) was done. 

If you could update the stable version to the point just before the port, or fix the code that doesn't work in Windows in the development version it would be fantastic.

[image]

Is there anything I can do to help?

alexrj, 2015-07-06:
Can you post a complete report in a separate issue?

a4jp-com, 2015-07-07:
Okay I'll post the config file and a 3D file in a separate post for the stable version. I'm looking forward to the _putenv_s() though. Everything was working really well before.

a4jp-com, 2015-07-08:
I don't know if this will help much but it shows where the 'setenv()' values are.

[image]

a4jp-com, 2015-07-08:
I figured out a way to compile the files in Windows.

Config.cpp
Line 115: '_putenv_s(envname.c_str(), this->serialize(*it).c_str());'

config.hpp
Line 82: '_putenv_s( name.begin(), value.begin());'

[image]

[image]

alexrj, 2015-07-10:
Can you confirm this commit works?

a4jp-com, 2015-07-10:
I'll check now ^^

a4jp-com, 2015-07-10:
./xs is working now and everything compiles. 

Beautiful coding ^^ 

Thank you so much.

closed 2015-07-10 by a4jp-com

llluis, 2015-07-30:
Just to let you know that this fix doesn't work with latest versions of CitrusPerl 5.16.3.18 x64 and MiniGW gcc (gcc-4.6.4 release with patches [build 20130519 by perlmingw.sf.net]) 4.6.4.

    src\libslic3r\Config.cpp: In member function 'void Slic3r::ConfigBase::setenv_()':
    src\libslic3r\Config.cpp:117:9: error: '_putenv_s' was not declared in this scope
    error building dll file from 'src\libslic3r\Config.cpp' at     F:/Software/CitrusPerl/lib/ExtUtils/CBuilder/Platform/Windows.pm line 130.
    FAIL
    ! Installing ./xs failed. See D:\DOCUME~1\xy0334\.cpanm\work\1438269404.1384\build.log for  details. Retry with --force to force install it.
    The XS/C++ code failed to compile, aborting

Maybe CitrusPerl should be dropped as supported.

(I was trying to compile on a "restricted" machine with a portable perl installation. Don't know if I'll be able to install Strawberry Perl here)

lordofhyphens, 2015-07-30:
It is 5.16 that is the problem.

llluis, 2015-07-30:
RTFM for me!

Trying to find CitrusPerl 5.14 for download. 
strawberryperl.com is offline. 
:(

lordofhyphens, 2015-07-30:
Probably just want to use the other Windows Perl distribution I think. Or
linux :D

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

Hurricane31337, 2015-08-08:
For me, Strawberry Perl v5.22.0 with MinGW gcc 4.6.3 didn't work, either (with or without the changes a4jp-com suggested), but gcc 4.9.5 x64 did (without changes). I guess, it should work with CitrusPerl, as long as a newer gcc version is used.

c-born, 2015-09-22:
CitrusPerl 5.14 and 5.16, using the default MinGW gcc they install, fail for me on the _putenv_r, which is declared in sec_api/stdlib_r.h and isn't being included.
I tried installing the latest MinGW gcc package and connecting CitrusPerl to it, but that threw up a bunch of new errors.
What versions and flavours of Perl and MinGW are being used for current Windows development testing, so I can get my system in sync?