slic3r/Slic3r #2976: Gap fill at constant volumetric extrusion rate (autospeed)

Hexman64, 2015-07-04:
(Moved from issue #2810 as it was off topic there):

I set all the extrusion widths to 0, and as long as the speed settings are above 0, I get a nice gap fill which I call "the new style":
[image]

When I then change the speed settings, especially the gap fill speed setting to 0, gap filling is done the legacy way. I guess printers cannot keep up a good (extrusion) speed when filling gaps the zig-zag way:
[image]
(by the way, for this model you see a gap fill error in one of the lower layers)

The gap fill looks good for speed settings above 0, so it should be done the same way when using autospeed.
Constant volumetric speed is a good improvement when printing flexible material.
Slic3r is Windows 1.2.9 x64

luisonoff, 2015-07-18:
Are those two pictures of the same STL?? Because they look quite different, one looks taller or the other one wider...

Hexman64, 2015-07-18:
Just to make sure, in the below test I only changed the gap speed from 30 to 0.

[image]

luisonoff, 2015-07-18:
If you set "gap filling speed" to 0, gap filling is disabled, as the GUI says when you stay over that option.
So if I understand correctly you are asking for an improvement that enables gap filling to work with autospeed in order to keep volumetric extrusion rate constant also while gap filling? Yes?

Hexman64, 2015-07-18:
DAMN! Your suspicion is right. What I thought is the legacy way of gap-filling in fact is a solid infill because the area is smaller than the value in the "solid infill threshold area" setting.
**So it was just my confusion, Slic3r is working as designed**, and yes, then I'd 

**Need an Improvement please: gap filling with constant volumetric extrusion rate.**

I'll change the title fo "Feature request". Thank you for finding the bug in my thoughts.

luisonoff, 2015-07-26:
+1 on this request. Gap filling is a very nice feature and it "looks like" should not be too hard to make it compatible with autospeed. Since widths of gap filling are very variable, that's an extra argument to make it work with autospeed instead of manually setting a speed.

teppix, 2015-07-29:
+1
Have also encountered this before. Definitely would expect 0 to be autospeed for gapfill. Also, gapfill is one of the cases where volumetric speed would be most useful, due to the wide extrusions.

Hexman64, 2015-07-29:
Yes... when printing flexible material, I could set all the widths and speeds to the same value, except for gap filling, which has variable widths. But there autospeed is not implemented. Hopefully the implementation and the fix of #2960 will be done in one step (gap filling is broken since 1.2.8).

lordofhyphens, 2015-07-29:
Probably could use a negative # as indication for disable gap fill instead
of 0.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

teppix, 2015-07-30:
@lordofhyphens 
That would work, but personally I'm against hiding features like that. It would just add yet another thing to explain in the already long tooltip. A UI should be self explanatory as much as possible, and coding things into integers is just counter intuitive.

I have no problem with using the integer behind the scenes, but definitely not in the GUI.

I think it would be better if all speed fields had an "auto" checkbox that would disable the numeric input.

One could also question if gap fill should be enabled/disabled in the speed section. That also is counter intuitive IMHO.  Better to place that option in the infill tab, where you would expect to find it.

lordofhyphens, 2015-07-30:
Works for me, I am mostly thinking about the ini interface.

alexrj, 2017-03-05:
I fixed this: gap_fill_speed = 0 now enabled autospeed for gaps instead of disabling gap fill. A new fill_gaps option was added for that purpose.

closed 2017-03-05 by alexrj