slic3r/Slic3r #2977: Support material interface pokes through wall

hyperair, 2015-07-04:
STL: http://www.thingiverse.com/thing:699323
Gcode: https://gist.github.com/57aeadd7596bc77ebe14
Screenshot of gcode preview: 
[image]
[image]

hyperair, 2015-07-04:
Same issue doesn't seem to happen with rectilinear support.

Zuborg, 2015-07-05:
The pillar support mode is too "raw". I never use it. A result of a pillar support is unexpected...

hyperair, 2015-07-13:
Well, with that model, rectilinear support is almost impossible to remove without breaking it. That's a 1mm wall there with non-filleted corners. It'll snap the moment you try to pry the bulky rectilinear support out.

lonelymyp, 2015-09-16:
This type of support has caused a lot of questions. there are already several reports of problems. waiting when they fix :)