slic3r/Slic3r #2978: Round layer height to nearest submultiple or object height

alexrj, 2015-07-04:
Layer height should be adjusted automatically so that total object height is not altered.

For example, if user sets 0.3mm layer height and slices a 20mm object, Slic3r will now produce a 20.1mm object in order to have an integer number of layers. It should round layer height to 0.2985mm so that object height will be 19.9995mm which is much more accurate (plus, last layer should be forcefully aligned to real height, thus 20mm).

In order to do this, we should increase precision of Z coordinates in G-code to 5 decimals (one microstep at 1/16 with a M8 leadscrew is ΔZ = 0.00039mm).

This should also take microstepping into account as per #1827

xoan, 2015-07-04:
Total height is not the only affected dimension:

[image]

If layer height is rounded to achieve the total height, the first top solid shell at 10.25mm has no an integer number of layers:

    >>> 10.25/0.2985
    34.33835845896147
    >>> 0.2985*34
    10.149
    >>> 0.2985*35
    10.4475

whosawhatsis, 2015-07-04:
If you set the layer height to values like those, you're going to get rounding errors as I describe here: https://www.evernote.com/l/ANNwHDbE3dVGaaSClT2JJMcdHvmSmIKVSHw

This also won't account for shapes that do not have only one flat top surface, as @xoan mentions. A better solution is to do the whole print at the layer height specified by the user, then adjust the thickness of only the last layer to make it dimensionally accurate, as I suggested in https://github.com/alexrj/Slic3r/issues/2782

Mikk36, 2015-07-04:
There should also be a lower limit (user-settable) to these thinner top layers, when you don't want a layer thinner than 0.05 mm, for example.

alexrj, 2015-07-05:
@xoan, good point.
@whosawhatsis, are the rounding errors you mention related to Z steps? (i.e. #1827?)

pyrophreek, 2015-07-05:
This sounds like a good idea, and reminds me of an idea I had before where you can set a max/min layer height (or even extrusion width), then the slicer could adjust within those parameters to best meet the design, ie. if last layer would need to be 0.05 to be dimensionally accurate, and layer height is 0.3, then the second to last layer (which would now be the final layer) could be expanded to 0.35 to compensate.  

I realize the logic behind this might get crazy, and really slow down complex objects so it might not be worth it, but it might be worth trying.

lordofhyphens, 2015-07-06:
Well it is an integer linear programming problem to determine how many
layers or how thick. Maybe just a plain LP problem (which is better).
Although I am not at present quite sure what the objective function would
be (maybe minimizing the standard deviation from the mean).

Might take forever though. Those ILP formulations can take exponential (to
the number of constraints) time to solve. Additionally setting up the
problem and solving it definitely would add a dependency on a lgpl library
(lp_solve). Might be worth a look though.

http://web.mit.edu/lpsolve/doc/FAQ.htm

>
> This sounds like a good idea, and reminds me of an idea I had before
where you can set a max/min layer height (or even extrusion width), then
the slicer could adjust within those parameters to best meet the design,
ie. if last layer would need to be 0.05 to be dimensionally accurate, and
layer height is 0.3, then the second to last layer (which would now be the
final layer) could be expanded to 0.35 to compensate.
>
> I realize the logic behind this might get crazy, and really slow down
complex objects so it might not be worth it, but it might be worth trying.

starno, 2015-07-06:
@whosawhatsis that was a really interesting read, deserves to live in a more accessible place though

whosawhatsis, 2015-07-06:
@alexrj Yes.
@starno It was included in the first issue of Reprap Magazine, but I use the evernote link because I've made a few updates since then.

alexrj, 2017-03-14:
I implemented this as per @whosawhatsis's suggestion: the top layer thickness is adjusted in order to match the input object height.
This might be reimplemented using a smooth layer height transition when we merge the variable layer height code.

closed 2017-03-14 by alexrj