slic3r/Slic3r #2979: Feature Request: Force Full Reslice

the-edge01, 2015-07-05:
It would be useful to have a button to force a full reslice of an object if something goes wrong and you want to restart the entire slicing process.

pyrophreek, 2015-07-05:
I agree, and really what I would like to have is a button to press to view the sliced gcode before exporting it, but without having to turn on background processing.  I love the idea of background processing, but if you are planning to adjust a few parameters, sometimes it slows everything down so much as it tries to reslice each time you change any parameters or load a part.  I would prefer to be able to set everything up, then hit slice.

alexrj, 2015-07-05:
@the-edge01, what do you mean by *"if something goes wrong"*? Can you elaborate more on your scenario?

the-edge01, 2015-07-05:
I was referring to something such as #2980 or #2958 where a forced reslice would probably have fixed the problem.

MartinMajewski, 2015-07-12:
I support that request and want to add, that it would be good for slic3r if the slicing action would be more controllable by the user, i.e. if one could deactivate the automatic invocation and have a "perform slicing" button in a convenient place.

Especially when I have a high detailed, large scale model and just want to tune a few options slic3r tries to immediately perform the slicing on every single changed option, leaving me with an unresponsive application and unnecessary computation.

gege2b, 2015-07-13:
it's (almost) already the case.

go to prefs, disable background processing, and then slicing will occurs only when "export gcode" is clicked

MartinMajewski, 2015-07-13:
I use slic3r with the direct uplink to OctoPrint and don't want to send my object every time I slice my object for review within slic3r. So no, it is not "already the case", it's a workaround and not convenient from the usability point of view.

lordofhyphens, 2015-07-13:
Wouldn't background processing already slice it and give you the update
preview? Else you could just export gcode normally--not like it HAS to go
to octoprint.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

MartinMajewski, 2015-07-13:
Please read my previous post. I want an option to trigger the processing on demand, because background processing is not always the optimal behavior.

kefir-, 2015-07-13:
Perhaps some hotkeys to control behaviour, like:
- Enable/disable autoslice
- Slice manually
- Save gcode
- Jump to various config/preview screens

I'm a big fan of keyboard shortcuts, and I think this could solve the issues without stuffing more stuff into the gui.

a4jp-com, 2015-07-13:
Just save a new preset in the printer area and when you switch between the 2 files you will always get a reslice. It's better than adding a button as it does the exact same job without cluttering the interface. No programming needed ^^

alexrj, 2015-07-13:
@the-edge01, the proper solution for the issues you mentioned is to fix those issues rather than adding a button.

@MartinMajewski, as others said you can turn off background processing. We should probably still trigger it automatically whenever user switches to the preview tabs. That means that user wants to see a preview, and we have to compute that preview.

lordofhyphens, 2015-07-13:
@alexjr, I generally would prefer to produce a non-blocking notice
(probably a ! icon on the tab or similar) if background reslicing is off
and the preview tab is requested. I tend to make several changes with the
preview tab still open, or mis-click the preview tab when I want 2D.
Trigging reslicing then (especially if it is a complex model) would be
annoying at best.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

pyrophreek, 2015-07-13:
I understand and agree with the desire to keep the interface uncluttered, so I suggest the following:

when clicking on any of the preview tabs (3D, layers), if background processing is off, and a current preview is not available, place a button in the center of the preview window to generate a preview (slice the model without exporting it).  

Bonus, after clicking the generate preview button, add a progress bar to show the status of the slicing and change the generate preview button into a cancel slicing button.

gege2b, 2015-07-14:
Well, I agree with @alexrj 's logic , triggering background processing only when switching tabs would be a good solution (if background processing is disabled in prefs of course), as this would mean that user want to see the result of some fine tuning.

In that case, I don't see the point of adding a new button to force a re-slice

In case of a mis-click, maybe a "cancel" button could be added next to the progress bar already present at the bottom of the window (or respond to the ESC key, or both)

MartinMajewski, 2015-07-14:
@alexrj With "preview-tabs" you mean the "Plater" tab on the top? Okay, so you suggest that the slicing process gets suspended as long as I am inside the "Print Settings", "Filament Settings" or "Printer Settings" tab and gets triggered when I switch back to the "Platter" tab?

If this is correct I would support that!!!

I also support the idea of having a "cancel" button next to the progress bar.

And I would generalize it a bit further:

Suspend the slicing progress as long as the 2D, Preview and Layers View ARE NOT in focus / visible. Even inside the 3D View it is not necessary - or even desirable - to have the slicing process active. Why? Because inside the 3D view you a) don't see a thing of the slicing result and b) you want to perform object positioning or arranging and every time you change something the slicer would go nuts - what he does on complex models.

And: Apply the cancel-operation to every background process, even uploading the gcode to OctoPrint or exporting as file. Sometimes it takes a very long time and you notice a mistake in the meantime, or OctoPrint is not reachable and you have to wait for the timeout or something, something. Spare us the unnecessary processing time, please!

pyrophreek, 2015-07-14:
@MartinMajewski 
This sounds like a good way to handle it.  I think having a progress bar and cancel slicing button appear in the layers/preview window to let the user know that the result of clicking on the tab is that the slicing has begun would be a good idea, as it might not be clear to newer users that the preview is being generated.  

One issue I see might be that if the slicing is cancelled while the preview window is open, and the user then wants to slice it again, it might not be clear that to reinitialize the slicing they would have to switch away from the tab and back.  Perhaps then it might be a good idea to have the button appear to manually activate the slicing.

MartinMajewski, 2015-07-15:
There could also be an overlay message on top of the preview area (or 2D or Layer View) that appears when a slicing process was cancelled instead of the actual sliced object. This message could be clickable and saying something like "You have cancelled the slicing process, therefor no preview is available. Click here to start slicing again"

This way no additional buttons have to be added, because the GUI shows this additional interaction field by context inside the area that is most significant for the user's attention then having the Plater tab open.

I think it would be also enough to have a "X" -cancel button right to the progress bar with the right tooltip label. The progress bar only appears when something is going on in the background and the user looks intuitively at the progress bar. This way any background progress would be cancelled with just this one button. To reactivate the slicing, see the paragraph above, to reactivate the OctoPrint / file -export the corresponding buttons are already there.

slic3r also has a message area an the lower left so it can be also used to tell the user that he has cancelled a job.

a4jp-com, 2015-10-02:
Has this been fixed?

MartinMajewski, 2015-10-02:
As I can see the latest commit to the slicer codebase is two months old and there have been no new releases since. So I guess no...

lordofhyphens, 2015-10-09:
@MartinMajewski  work goes in fits and spurts. @alexrj does this in his spare time! If it's that important to you fix it yourself and submit a pull request.

MartinMajewski, 2015-10-09:
@lordofhyphens My reply wasn't a complain, it was just an observation! :-)

alexrj, 2016-12-18:
I'm waiting to merge the "Reslice" button coded by @lordofhyphens because I wanted to try the minimal approach described here. If background processing is disabled, reslice only then selecting the preview tabs (a "Cancel" button is there). Whenever user makes changes that need reslicing, the plater view is automatically selected again.

This is pushed in master now. Any thoughts?

lordofhyphens, 2016-12-18:
@bubnikv had an extended version of it. I don't remember if my pr was
updated along with it.

lordofhyphens, 2016-12-18:
Mostly because I just woke up. Should be able to attach the reslice code to
an event handler for the tab button,  either on click or maybe on focus for
the window frame it belongs to?

alexrj, 2016-12-18:
@lordofhyphens, I already implemented the minimal approach, see current master and let me know your thoughts...

lordofhyphens, 2016-12-22:
Looks fine to me, although some people would probably feel better with a dedicated UI button.