slic3r/Slic3r #2981: Brim not attached to part when support is enabled

pyrophreek, 2015-07-05:
I was trying to print a part with a small width lip, followed by a large thickness wall
[image]
and I noticed that if I enabled brim, it didn't attach to the part when support was enabled, so it started to peel up.  Would it be possible to make the brim attach directly to the part rather than to the support material?  Any support material that coincided with the brim would then print on top of the brim rather than in place of it.

lordofhyphens, 2017-03-27:
@pyrophreek could you dig out a test case for this?

alexrj, 2017-04-02:
I fixed this. Thank you for reporting it @pyrophreek

closed 2017-04-02 by alexrj