slic3r/Slic3r #2982: 3D and 2D view are different than Preview and Layers view

efa, 2015-07-06:
I reported this on reprap forum before, but no answer there:
http://forums.reprap.org/read.php?263,528286,529568#msg-529568

The trouble is, using Slic3r 1.2.9 with this STL file:
http://forums.reprap.org/file.php?263,file=57628,filename=revers_3.stl,download=1

the 3D and 2D view are different than Preview and Layers view, the complete know is missing.
See shots:

[image]
[image]
[image]
[image]

lordofhyphens, 2015-07-06:
For starters it looks like your stl is broken. Second, 2D view shows the
shadow of the part on the z plane.

Could you post your slic3r settings and OS as well?

efa, 2015-07-06:
which configuration settings ini files do you need:
filament, print or printer?
(seems that the only accepted attachments are images, neither .stl files can be loaded on this issue tracker)

alexrj, 2015-07-06:
Your STL file has wrong normals. Fix it.

[image]
[image]
[image]

closed 2015-07-06 by alexrj

efa, 2015-07-06:
please tell me which software do you use to see the normals error.
(try loading Slic3r_config_bundle.ini say:
Unfortunately, we don't support that file type. Try again with a PNG, GIF, JPG)

efa, 2015-07-06:
ok using Blender I fixed the normals and now the part is sliced correctly, see
http://forums.reprap.org/read.php?263,528286,529762#msg-529762

lordofhyphens, 2015-07-07:
efa in the future upload configs/stls to a hosting site and post a link
here.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

efa, 2015-07-07:
reprap forum is OK for you for upload and link?

lordofhyphens, 2015-07-07:
@alexjr is the final authority on what's acceptable, as it's his project :D

Issue's been closed, though, so nothing else need be posted here.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

efa, 2015-07-10:
next time (future issues) I had to upload configs/stls, it is OK to use the reprap forum and link them here?