slic3r/Slic3r #2986: 1.2.9: Bad choice of layer start point

CCS86, 2015-07-08:
Slic3r seems to choose a very poor place to start/end layers/perimeter seams here. Unfortunately, repeatedly parking the nozzle, while waiting for a retract move, on the most overhung portion of the print, dumps a lot of extra heat into it and causes curling.

Is it possible for Slic3r to avoid overhung areas for start positions?

https://www.youtube.com/watch?v=EDwi3B2S3Nw

Hexman64, 2015-07-08:
Try to turn your object in Slic3r (by 180 degree or something). If you had "aligned" as starting point, try "nearest"

CCS86, 2015-07-08:
I actually did have it set to nearest, so this behavior seems odd

CCS86, 2015-07-15:
No thoughts?

xoan, 2015-07-15:
Not sure, but maybe related to #2909