slic3r/Slic3r #2987: Error building Git version on Windows

Mikk36, 2015-07-08:
Not sure if I'm doing something wrong, but this started to happen some time ago. Same thing happens on both Strawberry perl 5.20 and 5.22, both 32bit and portable.

    gcc -c -s -O2 -DWIN32  -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -fwrapv -fno-strict-aliasing-mms-bitfields -xc++ -DPERL_IMPLICIT_SYS -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -Isrc -Ibuildtmp -s -O2 -I"C:\strawberryperl\5.22.0.1_32\perl\lib\CORE" -I"C:\strawberryperl\5.22.0.1_32\c\include" -o "src\libslic3r\Config.o" "src\libslic3r\Config.cpp"
    src\libslic3r\Config.cpp: In member function 'void Slic3r::ConfigBase::setenv_()':
    src\libslic3r\Config.cpp:115:64: error: 'setenv' was not declared in this scope
             setenv(envname.c_str(), this->serialize(*it).c_str(), 1);
                                                                    ^
    error building xs.dll file from 'src\libslic3r\Config.cpp' at C:/strawberryperl/5.22.0.1_32/perl/lib/ExtUtils/CBuilder/Platform/Windows.pm line 129.
    FAIL

lordofhyphens, 2015-07-08:
This is a duplicate of #2973, I am pretty sure.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

Mikk36, 2015-07-08:
Indeed, you're right.

closed 2015-07-08 by Mikk36

a4jp-com, 2015-07-08:
Yes、I think this might be the same problem as the current Slic3r program was ported and as it uses setenv() instead of _putenv_s(). That means Windows systems can't compile the developer's version anymore. Well at least until the code gets updated.

a4jp-com, 2015-07-08:
If you look at my thread @Mikk36 there will be better answers there, but I found a way to compile the source if you are interested in directly editing 2 files.
---------------------------

Config.cpp
Line 115: `_putenv_s(envname.c_str(), this->serialize(*it).c_str());`

config.hpp
Line 82: `_putenv_s( name.begin(), value.begin());`

[image]

[image]