slic3r/Slic3r #2989: How to package slic3r on OS X into an executable app

MartinMajewski, 2015-07-08:
Hello,

I am missing the instruction on how to package slic3r into an executable app on OS X after building slic3r from source at your Wiki.

CavaPackager still does not support perl 5.18 - and I don't want to have multiple versions installed on my machine, because Slic3r ist the only application that's written in Perl (plus it was a pain to get all dependencies set up to build slic3r, so I better don't touch a running system).

Running Slic3r by chmod +x and removing the ".pl" extension works, but this way the menu bar is missing / not showing up.

Same for running Slic3r via an AppleScript or Automator's "Run Shell Script" workflow.

How do you package Slic3r for the stable builds?

Bye,
Martin

MartinMajewski, 2015-07-27:
Nobody? How do you guys perform releases then?

strahlex, 2015-07-31:
I managed to integrate Slic3r on my build server: http://buildbot.roessler.systems/files/slic3r/
I can send you the instructions for deploying Slic3r if you want

MartinMajewski, 2015-07-31:
Slic3r for OSX is a hard nut, indeed... After reinstalling Yosemite from scratch I haven't figured out how to get WX back to work (i.e. compiling and installing) again, because it requires some strange dependencies from the 10.9 SDK's version (at least a few libs at this specific location). So, currently I have no possibility to compile slic3r anyway. But your instructions would interest me nevertheless. Maybe you could provide them for us all? Thank you.

strahlex, 2015-07-31:
These are my private notes so they are messy, but follow the Mac/OSX steps and you should get a working and deployable Slic3r on OSX. I tested it on Yosemite and Mountain Lion:
https://github.com/strahlex/asciidoc-sandbox/blob/master/Slic3r_deploy.org

Basically this here should give you working perl:
https://github.com/strahlex/asciidoc-sandbox/blob/master/Slic3r_deploy.org#mac

You will need perlbrew and the command line developer tools installed see http://railsapps.github.io/xcode-command-line-tools.html

MartinMajewski, 2015-08-19:
Thank you very much.

Sadly that doesn't work for me on OSX 10.10.5.

Following that instruction I have managed to install Wx for the first time, but:

1. There is no freeglut (as well as no apt-get) for OSX. I use homebrew as my favorite package manager but freeglut has been pulled from there for unknown reason. I tried to build and install freeglut manually but it relies on X11 and I don't want to have this on my machine because it caused trouble in the past. There is a reason why Apple discontinued shipping X11...

2. With Perl 5.16.3 I've had no luck to build slic3r (even I've managed to install XS manually from the xs directory), because it failed to compile several things like Class::XSAccessor, Time-HiRes-1.9726, XML-SAX-ExpatXS-1.33, Scalar-List-Utils-1.42, etc.
Even trying to force install these separately failed. They build however well with Perl 5.23.0 but I only tested it with the x86_64 perl installation so far. So I think they are either not compatible with Perl 5.16.x or with Perl for i386.

3. Interestingly I could compile slic3r without gui with Perl 5.23.0 for x86_64. The --gui build failed because of Wx. Alien::WxWidgets cannot be compiled and so Wx.

I used your build server to get the newest OSX build in the past, but I have stupidly ignored the questionmark symbol today and overwritten my working copy with the broken build. It seems that your build server had also no luck lately? ^^ Unfortunately your build history is too short to list a last good build. :-(

lordofhyphens, 2015-08-19:
Perl 5.16 is *KNOWN* to not work with Slic3r. As far as I know, it's just
5.16.x, 5.18 and 5.20 (and later) versions are not incompatible.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

MartinMajewski, 2015-08-19:
I tried it with a couple of versions above 5.16.x

Currently I'm trying my luck with 5.23 as i386. But I cannot get Wx to install. Alien-wxWidgets compiled without errors, not so Wx.
Wx requires OSX SDK 10.9 and this is my main problem, because I have a clean 10.10 install with XCode 6.4 which provides only SDK 10.10. So I'm downloading XCode 5.1.1. right now and will try to extract SDK 10.9 from it.

a4jp-com, 2015-08-20:
I'm using

Perl 5, version 22, subversion 0 (v5.22.0) built for MSWin32-x86-multi-thread-64int

The set up was a pain though. Someone helped me and redirect a few of the files. If you look for my last post you should be able to do something similar.

strahlex, 2015-08-20:
@MartinMajewski Follow only the Mac OSX steps. These notes where written for Windows, Linux and OSX. It tested this on a brand new Mac Mini with the latest SDK. You have both SDK 10.9 and 10.10 installed if you have the newest XCode.

I had to force some packages. Makes you read the errors correctly and only force the dependency packages. I think I also needed to force the Wx packages since it failed on some browser stuff.

Make sure you use the correct perl version by checking the output of `which perl`

strahlex, 2015-08-20:
@a4jp-com If the author of this thread really wants to deploy Slic3r he will require *Perl 5.16.x* since it is the only version that can be deployed easily. If he only want to build Slic3r for his own Mac you can try your luck with newer versions.

MartinMajewski, 2015-08-20:
@strahlex I have followed the steps for osx to the letter. I have also forced every single packages I found in the error log, but some of them - like the Class-XSAccessor - won't install nevertheless. I checked the perl version bevor the procedure and it was 5.16.3 - i386.

I don't want to deploy slic3r for other people but myself, so it is not necessary to have a commonly used perl version imho.

Interestingly I have managed to install Wx one single time, but during the build of slic3r it was not recognized. I have still problems to understand the difference between cpan and cpan-minus installs, as they seem to install modules that are partially not accessible / visible for slic3r's building procedure - but are definitely there. I really learned to hate the perl ecosystem.

MartinMajewski, 2015-08-20:
@strahlex Okay, now I understand why I need 5.16.x - because the CavaPackager is so outdated, right?

There must be another path to deploy perl application... maybe through Automator? Not being able to simply get a standalone packaged App is one of my biggest problems with perl... -.-

strahlex, 2015-08-20:
@MartinMajewski Yes, that is the reason. I found no other way that works so far. I deployed quite a few different programming languages to different platforms. Most are a pain in the ass when it comes to deployment but Perl really stands out.

What does not work with the dmg files on my build server? Is there any error message?

MartinMajewski, 2015-08-20:
The dmg files are not the problem.
The app does not start. It quits with an Automator error message where it says (in German on my machine) that I should check the "action properties" and retry to run the work process. No further information provided here. 

Trying to run the perl script from inside the app-package shows the following error on the Terminal

```
/Applications/Slic3r.app/Contents/MacOS/slic3r ; exit;
"use" not allowed in expression at /Applications/Slic3r.app/Contents/Resources/lib/std/Slic3r.pm line 36, at end of line
syntax error at /Applications/Slic3r.app/Contents/Resources/lib/std/Slic3r.pm line 36, near "$var

use Moo 1.003001"
BEGIN not safe after errors--compilation aborted at /Applications/Slic3r.app/Contents/Resources/lib/std/Slic3r.pm line 38.
Compilation failed in require at /Applications/Slic3r.app/Contents/Resources/lib/std/cavaexecscript/slic3r.pl line 15.
BEGIN failed--compilation aborted at /Applications/Slic3r.app/Contents/Resources/lib/std/cavaexecscript/slic3r.pl line 15.

logout

[Prozess beendet]
```

Interestingly there is no cavaexecscript folder, nor a Slic3r.pm file nor a slic3r.pl file at this specific location - not even hidden.

strahlex, 2015-08-24:
Looks like the Slic3r.pm file is corrupted for some reason. Can you please check line 36.

MartinMajewski, 2015-08-24:
If you can tell me, where I can find slic3r.pm, I would be happy to help. But as I mentioned before: I don't see this file inside the App-package. In the plain source bundle line 36 of slic3r.pm is a blank line:

[image]

a4jp-com, 2015-08-24:
slic3r\lib

MartinMajewski, 2015-08-24:
Yeah, this is where I looked inside the source folder and took the screenshot above.

But inside the packaged app there is not such a file:

[image]

strahlex, 2015-08-24:
Because the file packaged into the binary.

Can you please copy and paste the whole line. I suppose you missed the editing the Slicer.pm file step. Slic3r.pm looks for the resources in a different location as when it is deployed by CavaPackager

MartinMajewski, 2015-08-24:
To make things clear: I am talking about @strahlex build server OSX version of slic3r right now. So I have no access to the source code that @strahlex uses, just the packaged app bundle.

My slic3r.pm file, that was pulled from slic3r's gitHub repository looks like shown in the following screenshot.

[image]

strahlex, 2015-08-24:
Then you probably used an outdated file. Try http://buildbot.roessler.systems/files/slic3r/slic3r_v1.2.9_ve-6_mac.dmg

Don't do the official binaries work for you?

MartinMajewski, 2015-08-24:
Okay! Hold on...

I removed every single trace of perlbrew incl. all custom perl installations and switched back to the OSX perl distribution that is

```
This is perl 5, version 18, subversion 2 (v5.18.2) built for darwin-thread-multi-2level
```

And now the most recent @strahlex build server build

```
slic3r_v1.2.9_ve-6_mac.dmg	06-Aug-2015 14:21	14M	
```

runs again!!! No error messages...

Seams that perlbrew messed things up...

I have really enough of Perl and the whole version conflicts and workarounds. My conclusion is that Perl is a real mess that needs too much time get running. I better stick to other less nerdy languages and use @strahlex builds for the further, if I may?! :-)

Nevertheless: Thank you for your patience and support!

MartinMajewski, 2015-08-24:
The problem with the stable builds is, that they are sometimes weeks or month behind the source code state. Some critical issues are solved way before the next stable build is official and since I rely on slic3r at my daily work, I better have the option to use both: the stable and the bleeding edge version...

strahlex, 2015-08-24:
Just to say: the Slic3r binary on my build server is based on the 1.2.9 branch. If there is interest in development binaries I could easily add the master branch to my build server. 

@alexrj  Am I allowed to do this and would you share the link?

MartinMajewski, 2015-08-24:
This would be cool...

jluis, 2015-08-25:
El Dilluns 24 Agost 2015, a les 08:07:31, Martin Majewski va escriure:
> Seams that perlbrew messed things up...
> 
> I have really enough of Perl and the whole version conflicts and workarounds. My conclusion is that Perl is a real mess that needs too much time get running. I better stick to other less nerdy languages and use @strahlex builds for the further, if I may?! 

If you have problems managing multiple perl installations with perlbrew; you may try plenv 
(https://metacpan.org/pod/release/TOKUHIROM/App-plenv-v1.2.0/bin/plenv).

Plenv allows you to have the system perl and one or more user installed perl and lets you to specify  the perl to be run as default *global* (like perlbrew) and override it at the *local* directory; see http://www.dagolden.com/index.php/2390/setting-up-a-perl-development-environment-with-plenv/

lordofhyphens, 2017-02-22:
We're working on getting the build scripts thoughtfully provided by @bubnikv so you can have fresh builds as they become available from a build server.

alexrj, 2017-03-11:
The build script is done (thanks to the hard work of @lordofhyphens), and we have automated builds available at https://bintray.com/lordofhyphens/Slic3r/slic3r_dev/view

Running `package/osx/make_dmg.sh` should work out of the box. We recommend installing a userland perl using perlbrew (5.24.1 tested) and re-running `perl Build.PL` and `perl Build.PL --gui` in order to ensure the dependencies are installed in local-lib. We will document it better but basically that's it.

closed 2017-03-11 by alexrj