slic3r/Slic3r #2991: Feature Request - Angular part arrangement

elkooo, 2015-07-09:
Hey There, 

I use a Delta Printer with a circular printbed since a few weeks (my fourth printer, but the first delta). My printers are often used for production and print large jobs (>10 parts at a time) - often also the same part. Therefore I always try to optimise the part-placement on the printbed in terms of space. 

Since I use the round printbed, it seems to me that «angular part placement» feature would be great (automaticly arrange parts around the middle with a certain angle instead of just placing them in rows). I think this should be possible since the code for turning parts is already implemented… only it would be necessary to measure a good distance from the centre in order that the parts do not collide. 

Would be cool to see this feature in future releases…

best Regards
elkoo

[image]

lordofhyphens, 2016-05-29:
That's all in Perl; take a look at the current algorithm used to lay parts out in the plater.

elkooo, 2016-05-30:
...I have no idea about how it's done. I am just a everyday user of Slic3r and giving feedback what I think might be useful... :)
So, if it's in Perl or any other Language - I have no Idea about that all - but would be nice if someone who has, comes up with a good solution?! At least if I am not the only person who finds such a thingi useful ;)

lordofhyphens, 2016-05-30:
My point is that this will probably languish until someone submits a PR for
it. Just the reality of resource availability.

elkooo, 2016-05-30:
...I see... 
Well... I can well live without it - I just propose stuff I would find helpful. If the developers think the same way, I'll be perfectly happy - aswell as if they don't ;)

Thanks for your comment though!
Perhaps someone likes the idea sometime, if not, I'll go on arranging files in CAD and wait a little longer for the slicing process :)

lordofhyphens, 2016-05-30:
Yeah. I am trying to organize everything at present and categorizing these
into milestones.

The "PR or bust" milestone is for the wishlisting, for the requests that
are submitted but are low priority  (for the 2 or so people actively
working in their spare time on this project). If someone gets an idea for
how to solve the problem then they can pick up items from there. At present
they get flagged low priority because they have been otherwise overlooked.
Scope creep is a different problem and currently not my focus.

I am much more willing to help someone clean up marginal code than to write
new features that I don't use myself.

I will add some of this information to the contribution guidelines and
probably link/summarize it in the main description.

Your idea isn't a bad one, just that it isn't something that I can do
immediately and @alexrj has a precious little time to work on this as it is.

elkooo, 2016-05-30:
No worries! :)
I am happy, you guys invest so much effort in this great program! To bad I am no help with coding (my personal strengths are more in 3D design, product development, graphic and manual tool-shop work). But I appreciate you are dealing with all the inquiries and I look forward to seeing the ongoing process on this really nice piece of software! Thanx for your good effort & keep it up!

bubnikv, 2016-07-18:
The heated print beds have usually a difficulty to keep the temperature at the set value at the edges. Therefore auto-arranging objects near the center of the print bed is useful for the rectangular beds as well. Therefore I don't think the auto-arrangement needs to be "angular" and I think the feature request is quite general: Pull the objects to the center of the bed while auto-arranging them. I believe Cura does something like this already.

lordofhyphens, 2016-07-19:
The current algorithm does try to keep things in the center, but it won't
rotate the pieces for you (which has some printing side effects as well as
being computationally expensive in comparison).

I know of another feature request to implement another packing algorithm
but that one is written in Javascript ;)

The spacing I think can be a bit generous and maybe could stand to have a
tunable attached to it.

mrvn, 2016-07-19:
What would be nice if one could select an placement order. One printer here for example has a strip of metal on the left side of the extruder to trigger the endstop. When printing a piece at a time it is often necessary to arange parts right to left and back to front to maximize the volume where no collision happens.

alexrj, 2016-11-24:
Slicing multiple copies of an object, having the same orientation, is optimized in Slic3r because the object is processed only once and then the toolpaths are repeated. This also ensures copies are identical.
Rotated copies are currently not possible and they would need additional complexity in the code for keeping this information and rotating the toolpaths in the preview and in G-code export. Quite a lof of work.

Also, I'm a bit skeptical about the idea as I can't imagine a solution which is general enough. Radial arrangement is fine for objects with a narrow shape, where one end is smaller than the other.

I think it's better to work on a smart arrange/nesting algorithm which would try to fit copies in a given generic shape, being circular or not.

elkooo, 2016-11-25:
alexrj, 
Right, I can follow your explanation perfectly. 

My thoughts came up because Slic3r is adding parts one after another on a line currently. On most printers, the printing bed has its most consistant temperature (on heated beds) in the middle. Therefore, if I print a lot of parts (which I do oftentimes since my business contains partly of selling 3D printed products), I always have to rearrange the parts in the middle (try to have an even distance to the bed border, since adhesion is lower at borders due to lower temperature and potential bed leveling issues which are most extremely on bed border).

If Slic3r would consider the form of the printing bed and make a new «line» of pieces to keep ratio of part-distribution on the bed similar to the shape of the bed, this would help already alot. (check picture enclosed..)
[image]

This is perhaps a bit more difficult for round beds or custom shapes but I think it might be easier than changing the part orientation?

Best
elkoo