slic3r/Slic3r #2992: 1.2.9 crash when open STL files on Mac osx 10.10.3

yangbo, 2015-07-10:
Apple Crash reports:
Process:               slic3r [65950]
Path:                  /Applications/Slic3r.app/Contents/MacOS/slic3r
Identifier:            slic3r
Version:               Slic3r 1.2.9.99 (1.2.9.99)
Code Type:             X86 (Native)
Parent Process:        ??? [1]
Responsible:           slic3r [65950]
User ID:               501

Date/Time:             2015-07-10 23:11:41.899 +0800
OS Version:            Mac OS X 10.10.3 (14D136)
Report Version:        11
Anonymous UUID:        0CF99EEB-1247-2B73-D0A1-FB8653994449

Sleep/Wake UUID:       C95D1C6B-C3DB-4342-8584-BEB38B7CD9AE

Time Awake Since Boot: 100000 seconds
Time Since Wake:       5900 seconds

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x00000000d746c750

VM Regions Near 0xd746c750:
    Stack                  00000000bf800000-00000000c0000000 [ 8192K] rw-/rwx SM=COW  
--> 
    Submap                 00000000ffff0000-00000000ffff1000 [    4K] r--/r-- SM=PRV  process-only VM submap

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_malloc.dylib        	0x9897d81b small_malloc_from_free_list + 522
1   libsystem_malloc.dylib        	0x9897ba84 szone_malloc_should_clear + 1292
2   libsystem_malloc.dylib        	0x9897b571 szone_malloc + 24
3   libsystem_malloc.dylib        	0x9897b519 malloc_zone_malloc + 75
4   libsystem_malloc.dylib        	0x9897a308 malloc + 52
5   com.apple.CoreGraphics        	0x9400499c aal_create + 228
6   libRIP.A.dylib                	0x91cdbbd9 ripr_Acquire + 129
7   libRIP.A.dylib                	0x91cdbabe RIPRenderPath + 64
8   libRIP.A.dylib                	0x91ccee60 ripc_GetClipState + 2704
9   libRIP.A.dylib                	0x91cce17e ripc_GetRenderingState + 144
10  libRIP.A.dylib                	0x91ccdfa4 ripc_DrawRects + 85
11  com.apple.CoreGraphics        	0x93fd7033 CGContextFillRects + 124
12  com.apple.CoreGraphics        	0x93ff8788 CGContextFillRect + 32
13  libwx_osx_cocoau_core-3.0.0.0.0.dylib	0x046566c6 wxMacCoreGraphicsContext::DrawRectangle(double, double, double, double) + 518
14  libwx_osx_cocoau_core-3.0.0.0.0.dylib	0x04718969 wxGCDCImpl::DoDrawRectangle(int, int, int, int) + 265
15  libwx_osx_cocoau_core-3.0.0.0.0.dylib	0x04718d4d wxGCDCImpl::Clear() + 397
16  libwx_osx_cocoau_core-3.0.0.0.0.dylib	0x045fd7ca wxWindow::MacDoRedraw(long) + 1226
17  libwx_osx_cocoau_core-3.0.0.0.0.dylib	0x046e2ca4 wxWidgetCocoaImpl::drawRect(void*, NSView*, void*) + 676
18  libwx_osx_cocoau_core-3.0.0.0.0.dylib	0x046e3420 wxOSX_drawRect(NSView*, objc_selector*, _NSRect) + 368
19  com.apple.AppKit              	0x9594ddc2 -[NSView _drawRect:clip:] + 4440
20  com.apple.AppKit              	0x9594ad3e -[NSView _recursiveDisplayRectIfNeededIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:topView:] + 3024
21  com.apple.AppKit              	0x9594b794 -[NSView _recursiveDisplayRectIfNeededIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:topView:] + 5670
22  com.apple.AppKit              	0x95949cb4 -[NSThemeFrame _recursiveDisplayRectIfNeededIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:topView:] + 311
23  com.apple.AppKit              	0x95946308 -[NSView _displayRectIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:] + 2942
24  com.apple.AppKit              	0x958f92ce -[NSView displayIfNeeded] + 1629
25  com.apple.AppKit              	0x958f8bbc -[NSWindow displayIfNeeded] + 324
26  com.apple.AppKit              	0x958f87a0 _handleWindowNeedsDisplayOrLayoutOrUpdateConstraints + 1262
27  com.apple.AppKit              	0x960e358b __83-[NSWindow _postWindowNeedsDisplayOrLayoutOrUpdateConstraintsUnlessPostingDisabled]_block_invoke1579 + 58
28  com.apple.CoreFoundation      	0x97060e3d _runLoopObserverWithBlockContext + 29
29  com.apple.CoreFoundation      	0x9706081e __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 30
30  com.apple.CoreFoundation      	0x97060760 __CFRunLoopDoObservers + 400
31  com.apple.CoreFoundation      	0x970520c2 __CFRunLoopRun + 946
32  com.apple.CoreFoundation      	0x97051aa6 CFRunLoopRunSpecific + 390
33  com.apple.CoreFoundation      	0x9705190b CFRunLoopRunInMode + 123
34  com.apple.HIToolbox           	0x99ef38f8 RunCurrentEventLoopInMode + 262
35  com.apple.HIToolbox           	0x99ef3503 ReceiveNextEventCommon + 192
36  com.apple.HIToolbox           	0x99ef342c _BlockUntilNextEventMatchingListInModeWithFilter + 99
37  com.apple.AppKit              	0x958f5721 _DPSNextEvent + 742
38  com.apple.AppKit              	0x958f4dc5 -[NSApplication nextEventMatchingMask:untilDate:inMode:dequeue:] + 350
[... 410 lines cut ...]
__OBJC                             4412K
__TEXT                            216.7M
__UNICODE                           552K
mapped file                       222.2M
shared memory                        68K
===========                      =======
TOTAL                             889.6M
TOTAL, minus reserved VM space    889.6M
 

Model: MacBook7,1, BootROM MB71.0039.B0E, 2 processors, Intel Core 2 Duo, 2.4 GHz, 8 GB, SMC 1.60f6
Graphics: NVIDIA GeForce 320M, NVIDIA GeForce 320M, PCI, 256 MB
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1067 MHz, 0x80CE, 0x4D34373142353237334248312D4346384E4F
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1067 MHz, 0x80CE, 0x4D34373142353237334248312D4346384E4F
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x93), Broadcom BCM43xx 1.0 (5.106.98.100.24)
Bluetooth: Version 4.3.4f4 15601, 3 services, 18 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
Network Service: gvpn na6, PPP (PPTP), ppp0
Serial ATA Device: Samsung SSD 840 EVO 250GB, 250.06 GB
Serial ATA Device: HL-DT-ST DVDRW  GS23N
USB Device: Built-in iSight
USB Device: BRCM2070 Hub
USB Device: Bluetooth USB Host Controller
USB Device: Apple Internal Keyboard / Trackpad
Thunderbolt Bus:

camchaney, 2015-07-19:
Having the same problem. Slic3r has been crashing all of the time with my laptop after the update.

morlac, 2015-07-19:
Same issue here on loading of .stl-files on MacOS X10.9.5 ..

Process:         slic3r [98809]
Path:            /Users/USER/Desktop/Slic3r.app/Contents/MacOS/slic3r
Identifier:      slic3r
Version:         Slic3r 1.2.9.99 (1.2.9.99)
Code Type:       X86 (Native)
Parent Process:  launchd [304]
Responsible:     slic3r [98809]
User ID:         501

Date/Time:       2015-07-19 20:58:57.293 +0200
OS Version:      Mac OS X 10.9.5 (13F1096)
Report Version:  11
Anonymous UUID:  7718AA19-C132-F825-D5D8-6D6FE28C4018

Sleep/Wake UUID: 22321869-C388-4B53-9EBC-FD22D2262C9D

Crashed Thread:  0  Dispatch queue: com.apple.main-thread

Exception Type:  EXC_CRASH (SIGABRT)
Exception Codes: 0x0000000000000000, 0x0000000000000000

Application Specific Information:
abort() called
terminating with uncaught exception of type std::bad_alloc: std::bad_alloc

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x9881e952 __pthread_kill + 10
1   libsystem_pthread.dylib       	0x958c0167 pthread_kill + 101
2   libsystem_c.dylib             	0x92b4a29c abort + 155
3   libc++abi.dylib               	0x96b266c9 abort_message + 169
4   libc++abi.dylib               	0x96b4747d default_terminate_handler() + 264
5   libc++abi.dylib               	0x96b44c30 std::__terminate(void (*)()) + 14
6   libc++abi.dylib               	0x96b4464b __cxa_throw + 116
7   libc++abi.dylib               	0x96b44e74 operator new(unsigned long) + 100
8   XS.bundle                     	0x03229e3a std::vector<float, std::allocator<float> >::_M_insert_aux(__gnu_cxx::__normal_iterator<float*, std::vector<float, std::allocator<float> > >, float const&) + 170
9   XS.bundle                     	0x0322a017 Slic3r::GLVertexArray::push_vert(float, float, float) + 119
10  XS.bundle                     	0x03227487 Slic3r::_3DScene::_extrusionentity_to_verts_do(std::vector<Slic3r::Line, std::allocator<Slic3r::Line> > const&, std::vector<double, std::allocator<double> > const&, std::vector<double, std::allocator<double> > const&, bool, double, Slic3r::Point const&, Slic3r::GLVertexArray*, Slic3r::GLVertexArray*) + 647
11  XS.bundle                     	0x03144af4 XS_Slic3r__GUI___3DScene__extrusionentity_to_verts_do(interpreter*, cv*) + 3956
12  libperl.dylib                 	0x000e74a6 Perl_pp_entersub + 2294
13  libperl.dylib                 	0x000e5af6 Perl_runops_standard + 22
14  libperl.dylib                 	0x00074ca5 Perl_call_sv + 725
15  Wx.bundle                     	0x03fb9e67 wxPliEventCallback::Handler(wxEvent&) + 1111
16  libwx_baseu-3.0.0.0.0.dylib   	0x04a71d9f wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const + 95
17  libwx_baseu-3.0.0.0.0.dylib   	0x04bb97a8 wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) + 104
18  libwx_baseu-3.0.0.0.0.dylib   	0x04bb9f98 wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 88
19  libwx_baseu-3.0.0.0.0.dylib   	0x04bba904 wxEvtHandler::TryHereOnly(wxEvent&) + 52
20  libwx_baseu-3.0.0.0.0.dylib   	0x04bba9ac wxEvtHandler::ProcessEventLocally(wxEvent&) + 60
21  libwx_baseu-3.0.0.0.0.dylib   	0x04bbaa21 wxEvtHandler::ProcessEvent(wxEvent&) + 81
22  libwx_osx_cocoau_core-3.0.0.0.0.dylib	0x04657dd2 wxWindowBase::TryAfter(wxEvent&) + 114
23  libwx_baseu-3.0.0.0.0.dylib   	0x04bbab11 wxEvtHandler::ProcessEvent(wxEvent&) + 321
24  libwx_baseu-3.0.0.0.0.dylib   	0x04bba3d7 wxEvtHandler::SafelyProcessEvent(wxEvent&) + 39
25  libwx_osx_cocoau_core-3.0.0.0.0.dylib	0x04656bfb wxWindowBase::HandleWindowEvent(wxEvent&) const + 27
26  libwx_osx_cocoau_core-3.0.0.0.0.dylib	0x04472471 wxNotebook::OSXHandleClicked(double) + 385
27  com.apple.AppKit              	0x996b5d36 -[NSTabView selectTabViewItem:] + 1193
28  com.apple.AppKit              	0x99aa1fab -[NSTabView mouseDown:] + 166
29  libwx_osx_cocoau_core-3.0.0.0.0.dylib	0x0452d916 wxWidgetCocoaImpl::mouseEvent(NSEvent*, NSView*, void*) + 150
30  libwx_osx_cocoau_core-3.0.0.0.0.dylib	0x045288e5 wxOSX_mouseEvent(NSView*, objc_selector*, NSEvent*) + 53
31  com.apple.AppKit              	0x99599a9d -[NSWindow sendEvent:] + 11953
32  libwx_osx_cocoau_core-3.0.0.0.0.dylib	0x04514e2d -[wxNSWindow sendEvent:] + 157
33  com.apple.AppKit              	0x9953591d -[NSApplication sendEvent:] + 4034
34  libwx_osx_cocoau_core-3.0.0.0.0.dylib	0x0443302a -[wxNSApplication sendEvent:] + 90
35  com.apple.AppKit              	0x9937a1bc -[NSApplication run] + 823
36  libwx_osx_cocoau_core-3.0.0.0.0.dylib	0x04506685 wxGUIEventLoop::OSXDoRun() + 277
37  libwx_baseu-3.0.0.0.0.dylib   	0x04b837c6 wxCFEventLoop::DoRun() + 38
38  libwx_baseu-3.0.0.0.0.dylib   	0x04ab29ba wxEventLoopBase::Run() + 74
39  libwx_baseu-3.0.0.0.0.dylib   	0x04a72a3f wxAppConsoleBase::MainLoop() + 95
40  Wx.bundle                     	0x03ebfbe2 wxPliApp::MainLoop() + 50
41  Wx.bundle                     	0x03e648b1 XS_Wx__App_MainLoop(interpreter*, cv*) + 129
[... 417 lines cut ...]
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1067 MHz, 0x855D, 0x436F6D707573746F63782028435358292020
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1067 MHz, 0x855D, 0x436F6D707573746F63782028435358292020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x93), Broadcom BCM43xx 1.0 (5.106.98.100.22)
Bluetooth: Version 5.2.7f4 15858, 3 services, 23 devices, 2 incoming serial ports
Network Service: Ethernet, Ethernet, en0
Network Service: AirPort, AirPort, en1
Network Service: Bluetooth PAN, Ethernet, en2
[... cut ...]
Serial ATA Device: Samsung SSD 840 EVO 500GB, 500,11 GB
Serial ATA Device: MATSHITADVD-R   UJ-898
USB Device: Hub
USB Device: Internal Memory Card Reader
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM2070 Hub
USB Device: Bluetooth USB Host Controller
USB Device: Hub
USB Device: USB2.0 Hub
USB Device: LaCie Hard Drive USB
USB Device: USB2.0 Hub
USB Device: Keyboard Hub
USB Device: Apple Keyboard
USB Device: USB2.0 Hub
USB Device: IR Receiver
USB Device: Built-in iSight
Thunderbolt Bus:

lobermann, 2015-08-05:
Can confirm this issue. http://pastebin.com/SufC3Jxb
But, I only have this with some STL files. Using Slic3er some time now without issues. Just now I run into the issue. And, it also crashes on windows when trying to load the STLs there. 
Should I provide the STLs to somebody for analysing?

JakeQZ, 2016-03-08:
Possibly due to #3267

alexrj, 2017-03-13:
@yangbo, @lobermann, we now have fresh automated builds available at
https://bintray.com/lordofhyphens/Slic3r/slic3r_dev/view

They are made on a recent OSX machine and I expect them to be very stable. Please reopen this if you have problems with those builds. Thank you for your report.

closed 2017-03-13 by alexrj