slic3r/Slic3r #2994: Manual does not describe how to operate 3D view

realflash, 2015-07-13:
I don't know how to zoom in on the 3D toolpath preview. Possibly the mouse scroll wheel does it but since my trackball doesn't have a scroll wheel I don't know. I tried to find the relevant key but when I read the manual, the only relevant page is 

http://manual.slic3r.org/first-print/working-with-models

which doesn't show the 3D views at all.

lordofhyphens, 2015-07-13:
That's a screenshot of 1.1.7 (which was the previous stable release), which
didn't have a 3D view.
I can confirm that the scroll wheel is one of the ways to zoom in/out.

@alexjr, is there a meta key for scrolling?

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

alexrj, 2015-07-13:
No. Ideas?

lordofhyphens, 2015-07-13:
Probably should be one of the three common buckybits (Ctrl alt or shift).

OpenSCAD apparently has four methods to zoom, but the one they use that we
could adopt for this issue is to use shift along with either a middle or
right-click and a drag.

https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/The_OpenSCAD_User_Interface

Blender uses Ctrl-middle mouse.
https://www.blender.org/manual/getting_started/basics/navigating/3d_view.html#zooming-the-view

Both use the +/- keys as a pure-keyboard (basically the same as a
mousewheel).

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

realflash, 2015-07-13:
Well I'm usually using SLic3r immediately after finishing in OpenSCAD, 
so my brain tells me to hold down Shift and right click together and 
then move the mouse up (zoom in) and down (zoom out). So that would be 
my choice.

GilesBathgate, 2015-09-15:
+1 for Shift and Right click together and then move the mouse up (zoom in) and down (zoom out)